FedeDP/Clight #86: Multi-camera support

SevereOverfl0w, 2019-07-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. EG: I'm always frustrated when [...]

I use a laptop, and plug that laptop into a dock periodically through the day and shut my laptop.  Shutting my laptop masks the camera.  I would like to use the USB WebCam if available in these situations.

**Describe the solution you'd like**

If there was an ordered list of devices to try (e.g. `["video2", "video0"]`) then clight could fall back onto the built-in camera when the USB camera was unplugged.

**Describe alternatives you've considered**

I could use https://github.com/troglobit/xplugd to detect when the camera is plugged in and perform a `sed` on the appropriate line in the config, followed by restarting the clight service.

FedeDP, 2019-07-09:
Hi! Thanks for taking the time to open this feature request!

Clight by default uses first capture device it finds (when "sensor_devname" is empty in config).
>  Shutting my laptop masks the camera

If this means that camera is still visible but it cannot take any capture, a solution would be to create an udev rule that matches internal camera to a superior (less priority) devnode, eg: "/dev/video1" instead of "/dev/video0", this way when you add a new USB camera, it should take "video0" node, thus having an higher priority.
Something like:
`ACTION=="add", SUBSYSTEM=="video4linux", ATTR{name}=="MyInternalCamera", NAME="video2"`
may work.

SevereOverfl0w, 2019-07-09:
Not looking positive:

> NAME="video3" ignored, kernel device nodes can not be renamed; please fix it in /etc/udev/rules.d/50-low-priority-webcam.rules:1

FedeDP, 2019-07-09:
It seems it cannot be done. I was hoping for a quick fix!
You can use a similar udev rule that detects when new USB camera is attached, and then calls [Clight bus API](https://github.com/FedeDP/Clight/wiki/Api#bus-properties) to set "SensorName" clight properly.

```
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="xxxx", ATTR{idProduct}=="yyyy", RUN+="/usr/bin/su YOURUSER -c 'busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf SensorName "s" "/dev/video1"'"
```

You can find "xxxx" and "yyyy" through `lsusb`.

I cannot test it right now, but apart some minor changes, it should work.

SevereOverfl0w, 2019-07-13:
The script I've ended up with is:

```
# /etc/udev/rules.d/50-low-priority-webcam.rules
SUBSYSTEM=="video4linux", RUN+="/usr/bin/su dominic -c '/home/dominic/bin/camera-hook $env{ACTION} %N'"
```

and then ~/bin/camera-hook is

```
#!/bin/sh

action="$1"
device="$2"

if [ "$action" = "add" ] && [ "$device" != "/dev/video0" ]; then
  busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf SensorName "s" "$device"
fi

# TODO: Handle case where multiple cameras have been added, and there's the
# option to fallback before using video0
if [ "$action" = "remove" ] && [ "$device" != "/dev/video0" ]; then
  busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf SensorName "s" "/tmp/video0"
fi
```

that seems to do the trick!

I'm not sure if/how it works if I restart clight after the camera is plugged in, will the SensorName be forgotten?  Also, is there a better way to unset the property?

FedeDP, 2019-07-14:
> that seems to do the trick!

Great to hear, thanks for these scripts!
> I'm not sure if/how it works if I restart clight after the camera is plugged in, will the SensorName be forgotten?

Yes; changes done through BUS API are runtime-only unless you store your new conf, with
`busctl --user call org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf Store`; this will produce a config file for your user, with any change you made (in this case, with new sensor).

> Also, is there a better way to unset the property?

I'd say just call 
`busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf SensorName "s" ""`
should do the trick.

FedeDP, 2019-08-25:
I am closing the issue. Feel free to reopen it for any question!

closed 2019-08-25 by FedeDP