FedeDP/Clight #87: How to start Clight?

ModDM, 2019-07-11:
FIrstly I have to say that for a regular computer user (not a software freak, but a normal computer user) the installation guides are very user-unfriendly. My computer is not my work, I am not an expert like you, but I use my computer just as a tool to do my work, as millions of other people.

I glady followed your instructions, but as mentioned above, I had some trouble with it.

I was able to succefully install git and after that creating a file called install.sh (on my desktop), placed these contents in it:
https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies

Opened the terminal from my desktop and executed the commands:
chmod 755 install.sh
(for the right permissions)

and finally I could doubleclick the file and was able to run it. It took me almost an hour to find out that it works like that. 1 minute of your time to add this information in your Wiki would be a big help for future users who are also struggling for an hour. (because you didn´t provide this information for basic users).

Ok, the install.sh script created four folders on my desktop:

Clight
Clightd
libmodule
ddcutil

I looked in your Wiki how to start Clight. Under the chapture Starting.
It says "it ships a xdg autostart desktop file".

I searched in all four folders for "xdg", but no such file found.

It also says "a desktop file to take a fast screen backlight recalibration, useful to be binded to a keyboard shortcut, is installed too". But there is nothing placed on my desktop! Where can I find this?

It furthermore says: "and it will show up in your applications menu."

I checked that, but the only thing I find in my application menu is:
2 folders and the clight service. When I click on it it opens my text editor with these contents:

[Unit]
Description=Bus service to manage various screen related properties (gamma, dpms, backlight)
Requires=@POLKIT_NAME@.service

[Service]
Type=dbus
BusName=org.clightd.clightd
User=root
ExecStart=@DAEMON_DIR@/clightd
Restart=on-failure
RestartSec=5

[Install]
WantedBy=multi-user.target

Ok, to prevent that I take more time for you, can you please help me out with simple instructions how to simply start your beautiful program? I am very interested to use it, but I and other basic users would be very thankful with some non-expert instructions.

(normally I run a command sudo apt install and after that a shortcut to the program is on my PC and I can run it, just in a minute, but no problem it if your advanced program with a lot of settings required some more steps, I only need to know which steps)

THANK YOU!

FedeDP, 2019-07-11:
> I was able to succefully install git and after that creating a file called install.sh (on my desktop), placed these contents in it:
https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies

That is just a script to deal with clight dependencies.
Normally, you should just download Clight, Clightd, libmodule and ddcutil latest releases and build them.
Unfortunately i have no time to maintain distro packages.

Note that you installed Clight and its dependencies from git master and not their latest deps. The "dealing with dependencies" script was provided by a kind user, and i just added it to wiki pages.
I will modify it to point to latest release, instead of git versions.

I see you're on Ubuntu/Debian; Clight/Clightd -git support creating deb packages for them. 
Thus, i'd proceed this way:
**Libmodule**
* Download latest libmodule from https://github.com/FedeDP/libmodule/releases/latest
* extract the tar.gz, cd into extracted folder
* Follow libmodule [build-how-to](https://github.com/FedeDP/libmodule#build-dep-and-how-to-build) plus installation for Debian

**DDcutil**
* Download latest ddcutil from https://github.com/rockowitz/ddcutil/releases/latest
* extract tar.gz, cd into extracted folder
* Run: 
```
./configure --prefix=/usr --enable-lib
make
sudo make install
```

**Clightd**
* clone Clightd repo: `git clone https://github.com/FedeDP/Clightd.git`
* cd into Clightd folder
* Follow [build instructions](https://github.com/FedeDP/Clightd/wiki/Build#build-instructions) plus installation for Debian

**Clight**
* clone Clight repo: `git clone https://github.com/FedeDP/Clight.git`
* cd into Clight folder
* Follow [build instructions](https://github.com/FedeDP/Clight/wiki/Build#how-to-build) plus installation for Debian

Finally, enable clightd.service and restart your pc.

Let me know if you need any more clarification, i will gladly answer. This will become part of wiki pages :)

FedeDP, 2019-07-11:
Note that ddcutil is present in Ubuntu repositories but unfortunately it does not ship its library.

ModDM, 2019-07-11:
OK, I immediately try it. Does it matter in which directories the several utilities are installed?

FedeDP, 2019-07-11:
> Does it matter in which directories the several utilities are installed?

When you run `sudo make install` (or install deb package build with `cpack -G DEB`) you must run it as root because you're installing it system-wide :)

ModDM, 2019-07-11:
BTW, I'd better delete the install.sh script and the four folders on my desktop, that I created before your instructions that you sent me?

FedeDP, 2019-07-11:
Yes it would be better!

ModDM, 2019-07-11:
Ooh! Step 1 went great. Step 2 gives problems: The DDcutil.

I extracted Mister Tarry Ball: tar -xzvf ddcutil-0.9.5.tar.gz
Went fine.

And then in the directory ddcutil-0.9.5 (temporary extract directory) I ran this command:
./configure --prefix=/usr --enable-lib

But my terminal says:
bash: ./configure: No such file or directory

Did I miss an install step?

ModDM, 2019-07-11:
oh sorry, I have to download the source code I think (this might was the compiled version on the download link)

ModDM, 2019-07-11:
No, same result: bash: ./configure: No such file or directory

ModDM, 2019-07-11:
... or do DDcutil and Libmodule need to be extracted (from tar.gz) into the same folder? (then I misunderstood)

FedeDP, 2019-07-11:
Nono! You're doing it right!
Libmodule is installed fine, great!
Let's move on to ddcutil. Once you downloaded and extracted latest release, you should cd into extracted folder and run
```
./autogen.sh
./configure --prefix=/usr --enable-lib
make
```
I forgot about autogen.sh! Sorry! I'll update above message!

ModDM, 2019-07-11:
oh you unhandy..... no joke, thanks a lot for your quick solution!

ModDM, 2019-07-11:
It says (sorry, again an error)
./autogen.sh: 3: ./autogen.sh: autoreconf: not found
./autogen.sh: 4: ./autogen.sh: ./configure: not found

ModDM, 2019-07-11:
I downloaded ddcutil-0.9.5.tar.gz. Or do I have to download Source code
(tar.gz).

ModDM, 2019-07-11:
SOLVED.
The solution was:
`sudo apt-get install autoconf`

ModDM, 2019-07-11:
Now `./autogen.sh` is solved.
`./configure --prefix=/usr --enable-lib` works now.
`make` returns this error:
`make: *** No targets specified and no makefile found.  Stop.`

ModDM, 2019-07-11:
The only make files I find in the extracted folder are:
Makefile.am
Makefile.in

Something is not working.

FedeDP, 2019-07-11:
I think ./configure is not finding some required packages. See "Required packages" at http://www.ddcutil.com/building/.
I just tried the same on Ubuntu 19.04 and it worked.

ModDM, 2019-07-11:
I will check it and see if any package is missing. The ./configure step also ended with a syntax error.
I don´t know if that is causing this problem.

`./configure --prefix=/usr --enable-lib
configure: debug messages disabled
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking for gcc option to accept ISO C99... (cached) none needed
./configure: line 5851: syntax error near unexpected token `disable-static'
./configure: line 5851: `LT_INIT(disable-static)'
`

FedeDP, 2019-07-11:
This is weird...i just did the same once more, to be sure, and ./configure did not throw any error (and i could `make` ddcutil).

ModDM, 2019-07-11:
I downloaded ddcutil-0.9.5.tar.gz. You run the same?

FedeDP, 2019-07-11:
> I downloaded ddcutil-0.9.5.tar.gz. You run the same?

Yes!

ModDM, 2019-07-11:
These were needed:
sudo apt-get install libglib2.0-dev
sudo apt-get install libx11-dev
sudo apt-get install libxrandr-dev

Then run the configure command again and then it works.

Although the packages were installed, the configure command does not recognize them and still says 'package not found'. In that case you have to put the word 'lib' in front of it and - dev after it, like libxrandr-dev.

Configure went succesful now, make went also successful now.

ModDM, 2019-07-11:
In case of Clightd these were needed:
sudo apt-get install libpolkit-gobject-1-dev
sudo apt-get install liblibelogind-dev
sudo apt-get install libsystemd-dev
sudo apt-get install libdbus-1-dev

However, when I want to install Clightd it says it requires ddcutil and ddcutil is not installed according to the error in the terminal (even when I succesfully installed ddcutil). So, now I look at that error.

Very complicated software.

ModDM, 2019-07-11:
`dpkg -s ddcutil 
dpkg-query: package 'ddcutil' is not installed and no information is available`

ModDM, 2019-07-11:
And I am bot able to solve this. I tried it also via sudo apt-get, but ddcutil can not be installed via that way:

Exact error:
`sudo apt-get install ddcutil`
`Reading package lists... Done`
`Building dependency tree       `
`Reading state information... Done`
`Package ddcutil is not available, but is referred to by another package.`
`This may mean that the package is missing, has been obsoleted, or`
`is only available from another source`

`E: Package 'ddcutil' has no installation candidate`

**I am lost in this jungle. What to do now?**

FedeDP, 2019-07-11:
Did you run `sudo make install` for ddcutil?
Btw dpkg won't find ddcutil as it is not bound to any package, it is installed explicitly (not through a deb package).

ModDM, 2019-07-11:
Yes, **make** and then **sudo make install.** I can see it in my terminal history.

ModDM, 2019-07-11:
Ok, you might see what I do wrong:

sudo dpkg -i  install clightd-4.0-Linux.deb
dpkg: error processing archive install (--install):
 cannot access archive: No such file or directory
(Reading database ... 292724 files and directories currently installed.)
Preparing to unpack clightd-4.0-Linux.deb ...
Unpacking clightd (4.0) over (4.0) ...
dpkg: dependency problems prevent configuration of clightd:
 clightd depends on ddcutil (>= 0.9.5); however:
  Package ddcutil is not installed.

dpkg: error processing package clightd (--install):
 dependency problems - leaving unconfigured
Processing triggers for dbus (1.10.6-1ubuntu3.4) ...
Errors were encountered while processing:
 install
 clightd

FedeDP, 2019-07-11:
Mmmh i may see now.
Clightd package requires ddcutil package, but in your case it was manually installed...
Ok: instead of installing clightd with `cpack -G DEB`, just issue a `sudo make install` and go on to Clight!

FedeDP, 2019-07-11:
This entire process would be much simpler if Ubuntu shipped ddcutil together with its own library... :(

FedeDP, 2019-07-11:
Oh wow, ddcutil itself provides such packages! See: https://www.ddcutil.com/install/.
Please let me know if this works!
If it complains that some files are already present (remember your ddcutil/make install), just dpkg --force ddcutil DEB installation!

ModDM, 2019-07-11:
> Mmmh i see now.
Clightd package requires ddcutil package, but in your case it was manually installed (thus dpkg does not find it).
Ok: instead of installing clightd with cpack -G DEB, just issue a sudo make install and go on to Clight!

That did the trick!

ModDM, 2019-07-11:
> This entire process would be much simpler if Ubuntu shipped ddcutil together with its own library... :(

No problem, you do your best and I appreciate that!

ModDM, 2019-07-11:
> Oh wow, ddcutil itself provides such packages! See: https://www.ddcutil.com/install/.
Please let me know if this works!
If it complains that some files are already present (remember your ddcutil/make install), just dpkg --force ddcutil DEB installation!

Ok, I understand correctly, that is good news. Well, I must confrim that ddcutil is installed successfully and Clightd also, meaning I have to go to Clight now, the last step to make everything work. What would you advise, go to Clight now or make another change to ddcutil, with the fact in mind that that package is installed successfully already?

FedeDP, 2019-07-11:
> > > Oh wow, ddcutil itself provides such packages! See: https://www.ddcutil.com/install/.
Please let me know if this works!
If it complains that some files are already present (remember your ddcutil/make install), just dpkg --force ddcutil DEB installation!

I quickly removed that comment because it seems the package is not working.
Go on with Clight then!!!

ModDM, 2019-07-11:
When I try to install a package required by Clight, I still get the error about Clightd that requires ddcutil and that ddcutil is not installed:

sudo apt-get install libpopt-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 **clightd : Depends: ddcutil (>= 0.9.5) but it is not installable**
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

FedeDP, 2019-07-11:
Remove Clightd (sudo apt remove clightd), and re-run `sudo make install` for Clightd once more.
Then go back at Clight and its deps!

ModDM, 2019-07-11:
Ok, let´s do that.

ModDM, 2019-07-11:
done, but I am no step further with Clight.

ModDM, 2019-07-11:
Question about Clight:

cpack -G DEB
or
sudo make install?

Which do I choose?

ModDM, 2019-07-11:
I ask this, because for the other program you said "instead of installing with cpack -G DEB, just issue a sudo make install"

ModDM, 2019-07-11:
I tried sudo make install

This resulted in these files, hope it´s right:
-- /usr/bin/clight
-- /usr/share/dbus-1/services/org.clight.clight.service
-- /etc/default/clight.conf
-- /etc/xdg/autostart/clight.desktop
-- /usr/share/applications/clightc.desktop
-- /usr/share/bash-completion/completions/clight
-- /usr/share/icons/hicolor/scalable/apps/clight.svg

Further I´d like to update, just for your info and for other users that t**hese packages were needed for Clight:**

**sudo apt-get install libpopt-dev
sudo apt-get install libgsl-dev
sudo apt-get install libconfig-dev**

Notice the -dev at the end.

ModDM, 2019-07-11:
**EVERYTHING INSTALLED. RESULT:**

After restarting the computer (twice) Clight is installed and runs automatically. (I saw the webcam LED twinkling once).

It has made a bit chaos on my computer, because for big health issues I need radiation to be OFF always. Therefore I have blacklisted Wifi and Bluetooth and since I installed Clight Wifi and Bluetooth modules are loaded on Linux startup and appear in the tray area. I can feel that my health issues come back again and I need to run modprobe for both Wifi and Bluetooth each time I boot the computer. The blacklisted modules in /etc/modprobe.d/blacklist are totally ignored since Clight is installed.

Since Clight is installed I get a warning on my bar that says:

> An error occured. Please run Package Manager from the right click menu or apt-get in a terminal to see what is wrong. The error message was 'Error BrokenCount > 0'. This usually means that your installed packages have unmet dependencies.

More details:
`sudo apt-get check`
`Reading package lists... Done`
`Building dependency tree`       
`Reading state information... Done`
`You might want to run 'apt-get -f install' to correct these.`
`The following packages have unmet dependencies:`
 `clight : Depends: libgsl23 but it is not installable`
         ` Depends: clightd (>= 4.0) but it is not installable`
`E: Unmet dependencies. Try using -f.`

In the meantime I'm interested to know if Clight is working.
Brightness: works

But I have the idea that I now have Calise. Clight should be able to do more, like gamma. So I checked this:
`xgamma`
`-> Red  1.000, Green  1.000, Blue  1.000`

This is the default (unchanged) gamma setting, so it seems nothing has changed.

How can I check if Clight actually influenced the color temperature and gamma for example? Are there any values I can verify? Otherwise I even can install Calise, which should work better than this Clight with all dependencies problems.

I am interested in your advise and help.

FedeDP, 2019-07-12:
> It has made a bit chaos on my computer, because for big health issues I need radiation to be OFF always. Therefore I have blacklisted Wifi and Bluetooth and since I installed Clight Wifi and [...]

There is nothing in Clight that forces such a behaviour, and it does not depend wifi nor bluetooth. 

> clight : Depends: libgsl23 but it is not installable
Depends: clightd (>= 4.0) but it is not installable
E: Unmet dependencies. Try using -f.

I can see libgsl23 on [ubuntu packages](https://packages.ubuntu.com/disco/libgsl23), may be you're on an older version?
Btw you should install Clight with `sudo make install` (or, remove any previously installed Clight package and re-run `sudo make install`).

> How can I check if Clight actually influenced the color temperature and gamma for example? Are there any values I can verify?

You should check $HOME/.clight.log file.

ModDM, 2019-07-12:
> There is nothing in Clight that forces such a behaviour, and it does not depend upon wifi nor bluetooth.

Since all time all icons / modules concerning connections etc. (except wired connection), that are found in the tray (and tray menu) were not there, because they are always blacklisted explicitely on my computer since the beginning (because I need that for my health > no radiation > I can´t live with 5 minutes of radiation). But I am aware of the fact that other modules / processes easily start blacklisted modules when they need them. If I want to blacklist modules permanently, I need to turn them off in BIOS. But at least strange that since always the tray icons (those modules) were not present, because of the blacklist, and that exactly after finishing this installation process, the tray icons are back again, and not just 1 icon, even multiple at the same time! (Wifi, Bluetooth and Location) They were never there! There should be a relating factor... No, it does not should be, it´s 100% sure it is!

> > How can I check if Clight actually influenced the color temperature and gamma for example? Are there any values I can verify?

> You should check $HOME/.clight.log file.

Thank you, works!
But, I meant a direct check outside Clight, to check if Clight is doing its thing. The brightness is working great, but can I check the gamma in terminal once in a time and see it it really changed?

Finally, thank you for all your assistance. It took me more than 1 day to install this program, but I appreciate your help.

ModDM, 2019-07-12:
For example: I checked this:
`xgamma`
`-> Red 1.000, Green 1.000, Blue 1.000`

This is the default (unchanged) gamma setting, so it seems nothing has changed.

ModDM, 2019-07-12:
> I can see libgsl23 on [ubuntu packages](https://packages.ubuntu.com/disco/libgsl23), may be you're on an older ubuntu version?

Ubuntu 16.04 LTS.
I try to install libgsl23, but it has a problem, guess what: a dependency problem with another (non-installable) package / module.

ModDM, 2019-07-12:
quickest way to restart Clight? Restart control (?) and by the way, any way also to turn Clight off (for the case I don´t know, just to know...).

FedeDP, 2019-07-12:
> quickest way to restart Clight?

Just kill it and restart it!

>  by the way, any way also to turn Clight off (for the case I don´t know, just to know...).

If you wish to turn anything in clight off, kill it. Btw you can turn off "dimmer" and or "automatic calibration" with a right click on clight desktop file.

ModDM, 2019-07-12:
Ok, I will look at it. Is it both a process and a service I need to kill or just 1 process?

ModDM, 2019-07-12:
for some reason my comments are posted in 2 issues (#87 and #88), but I assume you did that trick in a setting...

FedeDP, 2019-07-12:
Just `killall clight` is enough. Clightd is a system service but won't do anything on its own.

FedeDP, 2019-07-12:
> for some reason my comments are posted in 2 issues (#87 and #88), but I assume you did that trick in a setting...

No, i did not anything ...moreover i see my comments answering to you before yours O.O i think github is getting crazy...

ModDM, 2019-07-12:
HAHA.

ModDM, 2019-07-12:
YES: * Ambient gamma:                Enabled

ModDM, 2019-07-12:
Is this also normal way to start it? `clight`

FedeDP, 2019-07-12:
> Is this also normal way to start it? clight

Yes! or just leave and log in into your session!

ModDM, 2019-07-12:
Is this normal when running command `clight`:
`Config file /home/sysuser/.config/clight.conf not found.`
I understand. Config file is in /etc/default/ but why is this error?

FedeDP, 2019-07-12:
Yes, it means it cannot find your per-user conf file. It is just a warning :)

[8 more replies]

closed 2019-07-13 by ModDM