FedeDP/Clight #89: Any idea how to disable LED indicator from webcam during small Clight time intervals?

ModDM, 2019-07-12:
When you are a hard core Clight user with a time interval of 2 seconds, it can be (very) annoying when the webcam LED indicator twinkles every 2 seconds. Of course I know the old hint to put some tape on it, but in case the developer (or someone else) has a workaround that doesn't require modification of your laptop design, I mean a software way would be prefered, any advise would be welcome.

Thanks in advance.

FedeDP, 2019-07-13:
You can try something like this: https://www.togaware.com/linux/survivor/Video_Camera.html.
I am not sure though; it depends upon your webcam driver to expose such a capability.
Or, may be, your driver exposes led's informations in /sys/class/leds/ ? I don't think any webcam driver does that though (it would be great otherwise!)

ModDM, 2019-07-14:
I think it´s the best answer possible. Unfortunately not working in my case, maybe in the case of other people. However, I also did a good study about it, Googled a lot, and it seems that laptop manufacturers nowadays disable the control of the LED, as that was possible with laptops in the past. Buuying an old laptop would be the solution! (bad solution)

FedeDP, 2019-07-14:
> Buying an old laptop would be the solution! (bad solution)

!!! :D 
Very unfortunate, but I think that from a privacy pov, it seems quite reasonable to avoid having webcam LED exposed to an OS level.

closed 2019-07-22 by FedeDP