FedeDP/Clight #90: Backlight keep changing back and forth between two values

ModDM, 2019-07-15:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.0

**Used distribution:**
> Ubuntu

**Describe the bug**
Backlight keep changing back and forth between two values. For some reason capture 1, 3, 5, 7, 9 etc. capture an ambient brightness of 0.2 and capture 2,4,6,8,10 etc. capture an ambient brightness of 0.3. Result: A flickering screen: Lighter, darker, lighter, darker, lighter, darker (endlessly!).

**Expected behavior**
Backlight change is logical and expected. But it should be lighter / the same / the same / maybe one time a bit darker / the same / the same / lighter (because sun goes up further in case of a sunrise) etc. Normal behavior. But undesired is: lighter, darker, ligher, darker, lighter, darker, lighter, darker (as a robot, very unnatural).

**To Reproduce**
Steps to reproduce the behavior: Turn on Clight in terminal and check the changing values. To get the best view of the problem (same setting as I) set a time delay of 2 seconds between each capture.

Please attach a Clight verbose log too!
If needed, how do I do that.

ModDM, 2019-07-15:
I could not edit my post (maybe an error on Github).
Extra reproduce info:
ac_capture_timeouts = [ 2, 2, 2 ];

FedeDP, 2019-07-15:
I think you are getting hit by a subtle issue in some camera's firmware: it seems like it takes a couple of seconds for your webcam to auto set correct exposure; it means that capture 1 is "low", while capture 2 is brighter, as it now has correct exposure.
Again, after a couple of seconds it "cools down", restoring previous situation.
I don't think it is a firmware bug; your webcam is just slow to adapt to ambient brightness.
Mine is too :)
You can try to set a longer timeout (eg: 10s?) and check if it works better. 

Note that this issue is being discussed (and possibly fixed) in https://github.com/FedeDP/Clightd/pull/26 .

ModDM, 2019-07-15:
The thing is, btw thanks for your very! quick reply, it does not matter if I set a delay of 2 seconds, 1 minute or 5 minutes. It keeps toggling between brighter, darker, brighter, darker. My experience is that it is annoying regardless of how long or short your delay is. (when you are at work, 5 minutes is nothing, it feels like a few seconds) Of course, to solve this problem, I need to set a delay of 24 hours, but that is not desired! (I am just making fun, you understand what I mean). The thing discussed in the other link is something too technical for me to understand, but do what you can do with bugs like these, I appreciate if you learn from these cases to further tune your software. Or if it´s indeed my firmware (I don't think so, see above comment of mine), then I´d like to know it of course.

FedeDP, 2019-07-16:
> if I set a delay of 2 seconds, 1 minute or 5 minutes. It keeps toggling between brighter, darker, brighter, darker.

That is weird then!
Hopefully the Pull Request being reviewed/proposed for Clightd will fix this. 
Unfortunately I cannot reproduce the same behaviour on my system!
Btw to quickly test, you can call:
`busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "si" "/dev/videoX" 5`

as many times as you wish (in sequence). You can paste here the results.
This command is telling Clightd to do 5 captures from /dev/videoX and returns an array of 5 average brightness.

ModDM, 2019-07-16:
Hello FedeDP, I appreciate you do your best to help.

**Good news**

I applied the most old-fashioned solution possible: Just restarting my computer. And it is solved. Every capture Clight captures almost the same value! No flickering / jumping anymore between values back and forth.

However, let me just check if this thing you suggested is in order:
`busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "si" "/dev/videoX" 5`
Result:
`No such device`

ModDM, 2019-07-16:
By the way, Clight always starts with this message:
`PowerManagement inhibition appears to be unsupported. Adding bus match anyway.`

FedeDP, 2019-07-16:
> However, let me just check if this thing you suggested is in order: 

Your results seem to agree with you: the issue is not there :)

 > By the way, Clight always starts with this message:

This is just a warning to tell you that when clight starts, `org.freedesktop.PowerManagement` dbus interface is not exhistent.  
Note that it adds a match anyway, in case it is created later.
Clight uses that dbus interface to pause dimmer while eg: watching a movie.

ModDM, 2019-07-16:
The same happens again. Around the same time as yesterday (19:30). Restarting the computer was no solution anymore. I restarted 3 times, 3 times without any solution.

Conform your earlier advise I set the delay on 10 seconds, but the results are the same for each delay, even a delay of 1 minute or 5 minutes.

What can I do?

Config file /home/sysuser/.config/clight.conf not found.
SIGNAL module started.
BUS module started.
USERBUS module being started as submodule of BUS...
USERBUS module started.
INTERFACE module being started as submodule of USERBUS...
INTERFACE module started.
Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
Trying to start GAMMA module as its INTERFACE dependency was loaded...
Trying to start DIMMER module as its INTERFACE dependency was loaded...
Trying to start DPMS module as its INTERFACE dependency was loaded...
Trying to start LOCATION module as its BUS dependency was loaded...
geoclue_client_start(): No such property 'TimeThreshold'
Set timeout of 3s 0ns on fd 7.
LOCATION module started.
Trying to start UPOWER module as its BUS dependency was loaded...
UPOWER module started.
Trying to start BACKLIGHT module as its UPOWER dependency was loaded...
Trying to start DIMMER module as its UPOWER dependency was loaded...
Trying to start DPMS module as its UPOWER dependency was loaded...
Trying to start CLIGHTD module as its BUS dependency was loaded...
Clightd found, version: 4.0.
CLIGHTD module started.
Trying to start BACKLIGHT module as its CLIGHTD dependency was loaded...
Trying to start GAMMA module as its CLIGHTD dependency was loaded...
Trying to start DIMMER module as its CLIGHTD dependency was loaded...
Trying to start DPMS module as its CLIGHTD dependency was loaded...
Trying to start INHIBIT module as its USERBUS dependency was loaded...
inhibit_check(): The name org.freedesktop.PowerManagement.Inhibit was not provided by any .service files
PowerManagement inhibition appears to be unsupported. Adding bus match anyway.
INHIBIT module started.
Trying to start DIMMER module as its INHIBIT dependency was loaded...
Trying to start DPMS module as its INHIBIT dependency was loaded...
DPMS module started.
Callback added for module DPMS on module UPOWER bus match.
Callback added for module DPMS on module INHIBIT bus match.
Callback added for module DPMS on module INTERFACE bus match.
Callback added for module DPMS on module INTERFACE bus match.
Location 52.66 6.75 loaded from cache file!
Trying to start GAMMA module as its LOCATION dependency was loaded...
Set timeout of 0s 1ns on fd 8.
GAMMA module started.
Callback added for module GAMMA on module LOCATION bus match.
Callback added for module GAMMA on module INTERFACE bus match.
Next gamma alarm due to: Tue Jul 16 21:17:00 2019
Set timeout of 3276s 0ns on fd 8.
Trying to start BACKLIGHT module as its GAMMA dependency was loaded...
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
init_kbd_backlight(): No such interface 'org.freedesktop.UPower.KbdBacklight' on object at path /org/freedesktop/UPower/KbdBacklight
Keyboard backlight calibration unsupported.
Set timeout of 0s 1ns on fd 9.
BACKLIGHT module started.
Callback added for module BACKLIGHT on module UPOWER bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Callback added for module BACKLIGHT on module INTERFACE bus match.
Average frames brightness: 0.503420.
Normal transition to 5784 gamma temp started.
Ambient brightness: 0.503 -> Backlight pct: 0.714
Set timeout of 10s 0ns on fd 9.
Trying to start DIMMER module as its BACKLIGHT dependency was loaded...
DIMMER module started.
[... 49 lines cut ...]
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.577405.
Normal transition to 5960 gamma temp started.
Ambient brightness: 0.577 -> Backlight pct: 0.784
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.520939.
Normal transition to 5828 gamma temp started.
Ambient brightness: 0.521 -> Backlight pct: 0.731
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.580359.
Normal transition to 5966 gamma temp started.
Ambient brightness: 0.580 -> Backlight pct: 0.787
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.523201.
Normal transition to 5833 gamma temp started.
Ambient brightness: 0.523 -> Backlight pct: 0.733
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.580285.
Normal transition to 5966 gamma temp started.
Ambient brightness: 0.580 -> Backlight pct: 0.787
Set timeout of 10s 0ns on fd 9.
Average frames brightness: 0.523804.
Normal transition to 5835 gamma temp started.
Ambient brightness: 0.524 -> Backlight pct: 0.734
Set timeout of 10s 0ns on fd 9.

ModDM, 2019-07-16:
And for your info (better too much than too less) here another capturing period, but now from the log file (instead of the clight process terminal output).

Clight
* Software version:		4.0
* Starting time:		Tue Jul 16 20:22:21 2019

Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Automatic calibration:		Enabled
* Keyboard backlight:		Enabled
* Number of captures:		5
* Webcam device:		Unset
* Backlight path:		Unset
* Shutter threshold:		0.00
* Ambient gamma:		Enabled
* User position:		Unset
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 10	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900	BATT 300

### Modules ###
* BACKLIGHT:		Enabled
* GAMMA:		Enabled
* DIMMER:		Enabled
* DPMS:		Enabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		ENTER: 30, EXIT: 30
* Gamma long transition:		Disabled

(D)[20:22:21]{modules.c:62}	SIGNAL module started.
(D)[20:22:21]{modules.c:62}	BUS module started.
(D)[20:22:21]{modules.c:99}	USERBUS module being started as submodule of BUS...
(D)[20:22:21]{modules.c:62}	USERBUS module started.
(D)[20:22:21]{modules.c:99}	INTERFACE module being started as submodule of USERBUS...
(D)[20:22:21]{modules.c:62}	INTERFACE module started.
(D)[20:22:21]{modules.c:150}	Trying to start BACKLIGHT module as its INTERFACE dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start GAMMA module as its INTERFACE dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start DIMMER module as its INTERFACE dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start DPMS module as its INTERFACE dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start LOCATION module as its BUS dependency was loaded...
(D)[20:22:21]{bus.c:313}	geoclue_client_start(): No such property 'TimeThreshold'
(D)[20:22:21]{timer.c:41}	Set timeout of 3s 0ns on fd 7.
(D)[20:22:21]{modules.c:62}	LOCATION module started.
(D)[20:22:21]{modules.c:150}	Trying to start UPOWER module as its BUS dependency was loaded...
(D)[20:22:21]{modules.c:62}	UPOWER module started.
(D)[20:22:21]{modules.c:150}	Trying to start BACKLIGHT module as its UPOWER dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start DIMMER module as its UPOWER dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start DPMS module as its UPOWER dependency was loaded...
(D)[20:22:21]{modules.c:150}	Trying to start CLIGHTD module as its BUS dependency was loaded...
[... 108 lines cut ...]
(D)[20:24:51]{timer.c:41}	Set timeout of 10s 0ns on fd 9.
(D)[20:25:02]{my_math.c:29}	Average frames brightness: 0.580285.
(I)[20:25:02]{gamma.c:265}	Normal transition to 5966 gamma temp started.
(I)[20:25:02]{backlight.c:125}	Ambient brightness: 0.580 -> Backlight pct: 0.787
(D)[20:25:02]{timer.c:41}	Set timeout of 10s 0ns on fd 9.
(D)[20:25:12]{my_math.c:29}	Average frames brightness: 0.523804.
(I)[20:25:12]{gamma.c:265}	Normal transition to 5835 gamma temp started.
(I)[20:25:12]{backlight.c:125}	Ambient brightness: 0.524 -> Backlight pct: 0.734
(D)[20:25:12]{timer.c:41}	Set timeout of 10s 0ns on fd 9.
(I)[20:25:15]{signal.c:46}	received signal 2. Leaving.
(W)[20:25:15]{bus.c:144}	Wrong signature in bus call: o.
(D)[20:25:15]{modules.c:270}	DIMMER module destroyed.
(D)[20:25:15]{modules.c:270}	BACKLIGHT module destroyed.
(D)[20:25:15]{modules.c:270}	GAMMA module destroyed.
(W)[20:25:15]{bus.c:144}	Wrong signature in bus call: o.
(D)[20:25:15]{modules.c:270}	DPMS module destroyed.
(D)[20:25:15]{modules.c:270}	INHIBIT module destroyed.
(D)[20:25:15]{modules.c:270}	CLIGHTD module destroyed.
(D)[20:25:15]{modules.c:270}	UPOWER module destroyed.
(D)[20:25:15]{location.c:209}	Latest location stored in cache file!
(D)[20:25:15]{modules.c:270}	LOCATION module destroyed.
(D)[20:25:15]{modules.c:270}	INTERFACE module destroyed.
(D)[20:25:15]{modules.c:270}	USERBUS module destroyed.
(D)[20:25:15]{modules.c:270}	BUS module destroyed.
(D)[20:25:15]{modules.c:270}	SIGNAL module destroyed.

ModDM, 2019-07-16:
I found a relation between ambient brightness and this issue.

During daylight (0.55 brightness and more) --> nothing happens, no issue at all.
(that´s why the next time, when I restarted the PC and when you asked me to run `busctl call org.clightd.clightd /org/clightd/clightd/Sensor org.clightd.clightd.Sensor Capture "si" "/dev/videoX" 5` the issue was not there anymore: Because it was during daylight with values of 0.55 brightness!

Now it is the same brightness as the time I reported this bug (evening time: 19:30 and later, toggling between 0.51 and 0.58 which is an avarage value of 0.545 brightness and lower: at this time it is 0.4, it´s a while later). So, the difference in values is bigger now **(0.4 instead of 0.55)**.

To doublecheck this relation, I just tried to shine light with my flashlight on the webcam. The brightness values increased to 0.6, but they remained stable (no changing values back and forth). When I hold the flashlight still on the same place, the values stayed the same.

So, on higher brightness values the process does it job accurate and fine (brightness of 0.55 and higher) and on lower brightness the process gets troubles (brightness of 0.54 and lower).

Or, you could say: During daylight, no trouble, but in the dusk / evening and also the dawn / morning (from 11 o'clock in the morning there is problem anymore).

I experienced this issue on 2 evenings and 1 morning! (all times with low brightnesses!).

Does this also have something to do (the evening / morning issue) with the other link you sent me (can there be seen a relation?).

ModDM, 2019-07-16:
**Good news update:**

When I do this setting I get a more gradual course.

Config file /home/sysuser/.config/clight.conf not found.
Clightd found, version: 4.0.
PowerManagement inhibition appears to be unsupported. Adding bus match anyway.
Location 52.66 6.75 loaded from cache file!
Next gamma alarm due to: Tue Jul 16 21:47:00 2019
Keyboard backlight calibration unsupported.
Normal transition to 5131 gamma temp started.
Ambient brightness: 0.288 -> Backlight pct: 0.453
Normal transition to 5135 gamma temp started.
Ambient brightness: 0.289 -> Backlight pct: 0.454
Normal transition to 5076 gamma temp started.
Ambient brightness: 0.272 -> Backlight pct: 0.431
Normal transition to 4976 gamma temp started.
Ambient brightness: 0.245 -> Backlight pct: 0.391
Normal transition to 4981 gamma temp started.
Ambient brightness: 0.246 -> Backlight pct: 0.393
Normal transition to 4891 gamma temp started.
Ambient brightness: 0.222 -> Backlight pct: 0.357
Normal transition to 5130 gamma temp started.
Ambient brightness: 0.287 -> Backlight pct: 0.452
Normal transition to 5140 gamma temp started.
Ambient brightness: 0.290 -> Backlight pct: 0.456
Normal transition to 5064 gamma temp started.
Ambient brightness: 0.269 -> Backlight pct: 0.426
Normal transition to 4978 gamma temp started.
Ambient brightness: 0.245 -> Backlight pct: 0.391
Normal transition to 4984 gamma temp started.
Ambient brightness: 0.247 -> Backlight pct: 0.394
Normal transition to 4885 gamma temp started.
Ambient brightness: 0.220 -> Backlight pct: 0.354

ModDM, 2019-07-16:
SOLUTION FOR NOW:

I just had to try out each value for this setting:

## Number of frames or ALS device pollings to be captured
captures = 11;

And capture = 11 is the best value for me. This makes my screen super stable.

Here an example with captures = 11:

Ambient brightness: 0.101 -> Backlight pct: 0.160
Normal transition to 4413 gamma temp started.
Ambient brightness: 0.104 -> Backlight pct: 0.165
Normal transition to 4394 gamma temp started.
Ambient brightness: 0.100 -> Backlight pct: 0.158
Normal transition to 4400 gamma temp started.
Ambient brightness: 0.101 -> Backlight pct: 0.160
Normal transition to 4407 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.163
Normal transition to 4399 gamma temp started.
Ambient brightness: 0.101 -> Backlight pct: 0.160
Normal transition to 4395 gamma temp started.
Ambient brightness: 0.100 -> Backlight pct: 0.158
Normal transition to 4402 gamma temp started.
Ambient brightness: 0.102 -> Backlight pct: 0.161
Normal transition to 4428 gamma temp started.
Ambient brightness: 0.108 -> Backlight pct: 0.171
Normal transition to 4417 gamma temp started.
Ambient brightness: 0.105 -> Backlight pct: 0.167
Normal transition to 4409 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4411 gamma temp started.
Ambient brightness: 0.104 -> Backlight pct: 0.165
Normal transition to 4425 gamma temp started.
Ambient brightness: 0.107 -> Backlight pct: 0.170
Normal transition to 4409 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4404 gamma temp started.
Ambient brightness: 0.102 -> Backlight pct: 0.162
Normal transition to 4409 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4409 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.163
Normal transition to 4403 gamma temp started.
Ambient brightness: 0.102 -> Backlight pct: 0.161
Normal transition to 4419 gamma temp started.
Ambient brightness: 0.106 -> Backlight pct: 0.168
Normal transition to 4418 gamma temp started.
Ambient brightness: 0.106 -> Backlight pct: 0.167
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4401 gamma temp started.
Ambient brightness: 0.102 -> Backlight pct: 0.161
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4410 gamma temp started.
Ambient brightness: 0.104 -> Backlight pct: 0.164
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.163
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4418 gamma temp started.
Ambient brightness: 0.105 -> Backlight pct: 0.167
Normal transition to 4432 gamma temp started.
Ambient brightness: 0.109 -> Backlight pct: 0.173
Normal transition to 4433 gamma temp started.
Ambient brightness: 0.109 -> Backlight pct: 0.173
Normal transition to 4410 gamma temp started.
Ambient brightness: 0.104 -> Backlight pct: 0.164
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.163
Normal transition to 4418 gamma temp started.
Ambient brightness: 0.105 -> Backlight pct: 0.167
Normal transition to 4408 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.163
Normal transition to 4409 gamma temp started.
Ambient brightness: 0.103 -> Backlight pct: 0.164
Normal transition to 4424 gamma temp started.
Ambient brightness: 0.107 -> Backlight pct: 0.170
Normal transition to 4380 gamma temp started.
Ambient brightness: 0.097 -> Backlight pct: 0.152
Normal transition to 4388 gamma temp started.
Ambient brightness: 0.099 -> Backlight pct: 0.155
Normal transition to 4380 gamma temp started.
Ambient brightness: 0.097 -> Backlight pct: 0.152
Normal transition to 4363 gamma temp started.
Ambient brightness: 0.093 -> Backlight pct: 0.146
Normal transition to 4371 gamma temp started.
Ambient brightness: 0.095 -> Backlight pct: 0.149
Normal transition to 4363 gamma temp started.
Ambient brightness: 0.093 -> Backlight pct: 0.146
Normal transition to 4366 gamma temp started.
Ambient brightness: 0.094 -> Backlight pct: 0.147
Normal transition to 4375 gamma temp started.
Ambient brightness: 0.096 -> Backlight pct: 0.150
Normal transition to 4374 gamma temp started.
Ambient brightness: 0.095 -> Backlight pct: 0.150

> ## Question for you @FedeDP:

What actually does this setting (captures) do? What would be the technical explanation (as far as this is possible for you) that captures = 11 solved my issue and that alle other values for capture leaded to this issue, except for capture = 9 (almost good, but 11 is good). Thank you in advance for your information!

demoy, 2019-07-17:
Have you tried capture 1? (I believe capture tells how many frames to capture with camera)

FedeDP, 2019-07-17:
> What actually does this setting (captures) do

@demoy is right! Clight computes an average between N captures; the higher that value, the more accurate the average (as more frames are captured, leading to a better dataset).

ModDM, 2019-07-17:
At the moment the problems don't get back, so it seems captures of 11 did the trick in my case.

> Have you tried capture 1? (I believe capture tells how many frames to capture with camera)

Yes, see my second last comment (captures = 1 with results). Indeed, not bad.

> > What actually does this setting (captures) do
> 
> @demoy is right! Clight computes an average between N captures; the higher that value, the more accurate the average (as more frames are captured, leading to a better dataset).

Good to know. As in most things in live, you can also try to set the setting too good, as it seems. When I set the value of captures other than 11 (not only lower, but also higher!) then the issue starts again or the screen is at least unstable (many changing values).

demoy, 2019-07-18:
@FedeDP in my case the more frames the less accurate because the camera's flashlight skews the results at night.

FedeDP, 2019-07-18:
> in my case the more frames the less accurate because the camera's flashlight skews the results at night.

Very unfortunate; hopefully this will be addressed by new clightd capture algorithm.

> As in most things in life, you can also try to set the setting too good, as it seems.

Yes, that is the reason that value is not hardcoded. Diffirent webcams behave differently in similar situations...

closed 2019-07-18 by ModDM