FedeDP/Clight #91: Allow ambient Gamma to work without setting timezones

demoy, 2019-07-17:
**Is your feature request related to a problem? Please describe.**
When I as offline for an extended period I noticed the ambient Gamma support stopped until I used Red Shift to set a timezone, then Clight started using this timezone from cache.

**Describe the solution you'd like**
Would it be doable to have ambient gamma work when offline and timezone is unset?

**Describe alternatives you've considered**
I haven't.

Thank you for making such a great software BTW, if only I could replace my phone's default with Clight.

FedeDP, 2019-07-17:
> Thank you for making such a great software BTW, if only I could replace my phone's default with Clight.

Come on, you make me blush :)
>  When I as offline for an extended period I noticed the ambient Gamma support stopped until I used Red Shift to set a timezone, then Clight started using this timezone from cache.

This is weird as clight does cache your latest position when leaving, and uses that while geoclue2 does not give any location.
Moreover, i can't see how redshift could affect clight! O.O

You can set a fixed location in conf file though!
We should understand why location caching is not working though.

ModDM, 2019-07-17:
Have you considered this setting? I copy it from the configuration file. Maybe I say something weird, but just in case you missed it....

```

  ## Let screen temperature match ambient brightness, like monitor backlight.
  ## When enabled, screen temperature won't be changed time-based.
  ## Note that it uses same curve points as backlight.
  ## Finally, it requires BACKLIGHT module to be enabled, otherwise GAMMA gets disabled.

ambient_gamma = true;
```
Then Clight will not use timezone, but your ambient brightness to set gamma. I also use this setting, because it changes gamma realtime based on ambient light.

demoy, 2019-07-18:
Thank you @ModDM I had my config file from before the feature was added been calling the feature from the command line this seems so obvious in retrospect. @FedeDP, I cause it just started being able to access the cache again coincidentally when I started using redshift again and my mind made up a connection. Thank you both for the suggestions.

closed 2019-07-18 by demoy

FedeDP, 2019-07-18:
Thanks both! It is great to see other users suggesting solutions :)