FedeDP/Clight #92: Minimum brightness not truly minimum

sveinungss, 2019-07-20:
I've been using Clight for a few weeks now, and am really pleased with how well it works. One small niggle, though: Minimum brightness (0.0) is significantly brighter than the actual minimal brightness of my laptop monitor. That said, what Clight interprets as minimal brightness seems to correspond to the lowest standard brightness level using whatever interface/command XFCE uses as standard. But using xbacklight with -1, I can lower monitor brightness three levels (-3) below what Clight interprets as 0. Would it be possible to set xbacklight -1/+1 as the command to lower and increase brightness in Clight?

FedeDP, 2019-07-21:
Hi, thanks for opening this bug report/feature request!

> Minimum brightness (0.0) is significantly brighter than the actual minimal brightness of my laptop monitor.

That is kind of weird as Clight(d) uses data from /sys/class/backlight/ that is kernel-exposed interface on backlight.
May be you have more than 1 interface in /sys/class/backlight/, thus you should change default used one; care to check?

Btw i suggest you to give https://wiki.archlinux.org/index.php/Backlight#ACPI a read, it is interesting and maybe it has some fixes (eg: [Kernel command-line options](https://wiki.archlinux.org/index.php/Backlight#Kernel_command-line_options)!

sveinungss, 2019-07-21:
Thanks for replying so quickly!

> May be you have more than 1 interface in /sys/class/backlight/, thus you should change default used one; care to check?

I've checked, and there is only 1: intel_backlight. Using `cat /sys/class/backlight/intel_backlight/max_brightness`, max brightness is reported to be 937, and testing reveals that min brightness before screen blacks out is 1.

If I physically block the webcam and trigger Clight, the resulting backlight level is revealed to be 38 using  `cat /sys/class/backlight/intel_backlight/brightness'.

Adding `acpi_backlight=video` to the kernel parameters breaks backlight control completely, while the two other parameters (vendor and native) have no visible effect.

However, the fact that the minimum backlight level set by Clight is 38 out of 937, or 0.04, got me thinking that the problem might be that the default backlight transition step in Clight appears to be 5% or 0.05. I therefore tried changing the Clight config file line `backlight_trans_step` to 0.002 (i.e. approximately 2/937). As I am sure you would have told me, that did not work, but only slowed the transition down. Minimum brightness is still at 38.

FedeDP, 2019-07-21:
> If I physically block the webcam and trigger Clight, the resulting backlight level is revealed to be 38 using `cat /sys/class/backlight/intel_backlight/brightness'.

Can you try to set backlight to 0.0 directly from clightd:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight SetAll "d(bdu)s" 0.0 0 0 0 ""`

and then check if it does match minimum level as xbacklight tool? (and through `cat /sys/class/backlight/intel_backlight/brightness`)

sveinungss, 2019-07-22:
> Can you try to set backlight to 0.0 directly from clightd:
> `busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight SetAll "d(bdu)s" 0.0 0 0 0 ""`

This sets the backlight to absolute 0, i.e. blacking out the screen. `cat /sys/class/backlight/intel_backlight/brightness` supports this, returning the value 0.

FedeDP, 2019-07-22:
> This sets the backlight to absolute 0, i.e. blacking out the screen. cat /sys/class/backlight/intel_backlight/brightness supports this, returning the value 0.

Ok, so, what about:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight SetAll "d(bdu)s" 0.015 0 0 0 ""`

sveinungss, 2019-07-22:
`busctl call org.clightd.clightd /org/clightd/clightd/Backlight org.clightd.clightd.Backlight SetAll "d(bdu)s" 0.015 0 0 0 ""`resulted in the brightness value 14, and with a little testing, `0.002 0 0 0 ""`results in brightness value 1!

sveinungss, 2019-07-22:
So I tried editing the Clight conf again, and playing around with the numbers I finally achieved the desired effect. The solution turned out to be to lower the second to last of the Y values in the calibration profile. By lowering this, Clight also lowers the backlight level down to single digits.

Hence, it seems that my camera never returns a value that Clight interprets as perfect darkness, and thus the lower end of the calibration curve (X=0) never came into play.

FedeDP, 2019-07-22:
> It seems that my camera never returns a value that Clight interprets as perfect darkness, and thus the lower end of the calibration curve (X=0) never came into play.

Yes, that is the reason i made those value configurable. Each webcam returns different values and each user should be able to refine them :)

sveinungss, 2019-07-23:
:) Yeah, works great! Maybe add a line in the readme to highlight the possibility of the extremes of the calibration curve not coming into effect?

closed 2019-07-23 by sveinungss