FedeDP/Clight #93: Failed to connect to user bus

yakthungba, 2019-08-10:
Hello,

I am trying to run clight on void-linux with wayland and sway. When I run clight from command line I get this message
```
$ clight --verbose
SIGNAL module started.
BUS module started.
USERBUS module being started as submodule of BUS...
Failed to connect to user bus: No such file or directory
BUS module destroyed.
SIGNAL module destroyed.

```
Also I wanted to create a clihgt service for runit but I can't see anything under /etc/sv dir.

FedeDP, 2019-08-10:
Hi! Thanks for opening this issue. 
I am currently on vacation thus I am not able to help much.
I think that your login manager/desktop manager is not properly starting a systemd user session.

A quick search on Google revealed lots of similar issues, can you please try some of the proposed solutions?
Thank you very much!

yakthungba, 2019-08-13:
Thank you for taking out your time.

Actually I'm trying to run clight with runit. Does it support runit?

FedeDP, 2019-08-14:
You're welcome :)

Clight should support runit as it should not depend upon systemd at runtime (except for its sdbus library).
If, as it seems, it is not the case, that is a bug.
I'll give this a proper look next week!
In the meantime, I found this: https://askubuntu.com/questions/207936/what-program-starts-dbus-daemon
So, it seems your desktop manager does not start a local(per-user) dbus session.

closed 2019-08-20 by yakthungba