FedeDP/Clight #95: More points in curve for AC / battery states in config file - Feature request about backlight-to-ambient-brightness

ModDM, 2019-08-15:
In the current installation of Clight I experience a good performance the light adjustments based on ambient light, with the default curve points (as found in the config file).
But when you change those values, I experience too less steps (curve points).
The less curve points you have, the less transition you see between points.
When you theoretiocally would have 2 curve points in total, your screen would do FLOOP screen is maximum bright in the afternoon and FLOOP screen is maximum dark when it´s later in the evening or early in the morning (speaking about sunlight). You would have no transition!

The current amount of curve points support a transition (make it getting darker smoothly or make it getting brighter smoothly), but are not well enough suitable for for example a more steep subcurve in the darker (left) curve values, to let the screen become faster darker when it gets dark in the room (because my eyes are more sensitive for screen light in the dark than in the light). In my case - I tried it - this doesn't work out well, because the screen becomes 'flickery' (if that is an English word) within this dark (left) curve range, or let's say in the part from the middle to the left, because the steps between the curves are too big (screen needs to switch between too big difference between each point).

The default curve (demonstrated in a chart):
**Click on the picture for a bigger view.**

[image]

The adjusted curve:
**Click on the picture for a bigger view.**

[image]

Within the range of 4, 5, 6 and 7 the flickering starts, because the difference between those values are the biggest. When we had for example a double amount of curve points I aspect this issue would be solved. The more steps (points) the better the transition.

I think this stands apart from the amount of frames to be captured. In my case captures = 11 is ideal and better than for example 4, but that might be another subject, which also helps a lot if the user changes that in the config file (see captures function).

FedeDP, 2019-08-17:
Hi @ModDM !
Thanks for your feedback, much appreciated!

> The less curve points you have, the less transition you see between points.

Can you explain what does 'transition' mean in this context?

> Within the range of 4, 5, 6 and 7 the flickering starts

Again, what do you mean with 'flickering' here?

ModDM, 2019-08-19:
> Can you explain what does 'transition' mean in this context?

The time between captures can be set in the delay settings of the config file. Say you have a delay of 1 minute (I have a much shorter delay), and you are in a time period in which the light changes quick (later in the evening and early in the morning the light level changes according to the sunrise or sunset). And: To add another factor to this interaction process: You have set a curve that also makes an extra big change between the two curve points in which between that current time period falls (not based on time of the day, but ambient light in my case). Then:
- 1) The ambient light stimulates screen backlight change (because of the sun)
- 2) The curve stimulates screen backlight change, because the calculation of the ambient-to-backlight-system translates any ambient light change into a big change of the screen backlight (because the difference between the two curve points are big in this part of the curve)
- 3) The low delay settings stimulates a lot of changes of screen light. If the delay would be set to a bigger delay, the screen will change less often (but in some cases, you prefer a small delay).

i + ii + iii stimulate more change. On later evenings when it gets darker, the ambient light is more sensitive. During daylight, when you stand up, no problem. But in the evening, even when you move your head 3 centimeters more in front of a lamp behind you, the screen already changes. And now back to the clue of my suggestion:
Of course nothing would change when the curve was 1 straight line of all the same values from left to right, lets say 1, 1, 1, 1, 1, 1 etc. A darker ambient light or a lighter ambient light would have no change. But if you have (I exaggerate just for this demonstration) for example:
1, 1, 0, 3, 1, 1, 1 and based on the current ambient light level, the system needs to calculate between 0 and 3 then the current capture will capture a value of 0.5 and the next one 2.5 and the next one 0.8 and the next one 2 (all between 0 and 3). But I really think that when you would have a curve with two times more curve points, the screen would change (I call that flicker, especially when you have a delay of 2 seconds), as I believe less extreme, and more stable, because I think when the curve settings value is not differing between two points to big, then the screen light change would also be less.
A curve with 2 times more curve points would look like this:
1, 1, 1, 0.5, 0, 1.5, 3, 2, 1, 1, 1, 1, 1, 1
So, as you can see, a little lighter capture will result in a change of no more than 1.5 (between 0, 1,5 and 3), where it would result in a change of 3 (!) in the shorter curve (between 0 and 3).

I hope I made more clear what I meant.

> Again, what do you mean with 'flickering' here?

Flickering is a quick change in strong light and then less light, as is the way I describe the bigger changes with the current curve and daylight situation and delay setting.

> Do you mean that small ambient light changes bring to large screen backlight drop?
I think we mean the same indeed.

Last note: Of course these curves I wrote are just nonsence and no real curves, but they are just to demonstrate what I mean. I had to give a readable example. That's why I wrote a curve with a lot of '1's. And by the way, maybe my argumentation is not completely correct, because you know your program better than I do, but fact is that the shorter the curve is, the less details you can input, and the curve would be 'rawer', where the curve would be precizer perhaps when it would be separated into more points, no matter if it is 200% more points or 150% or 300%... That's something not up to me to know what is wise.

FedeDP, 2019-08-19:
Ok i now fully understand your suggestion!

Thanks for the detailed explanation. I cannot see any real downside and Clight has not any inherent limitation about that. 
I may increase the limit to eg: 50 points, giving much more granularity; obviously by default it will remain identical to now, with 11 points.
I don't think one would ever need more than 50 points :)

ModDM, 2019-08-19:
Well, I thought, I'd like to start with a modest small amount of just 5000 points... After that I'll see if I want to use more.

...kidding.

Thank you for your quick and helpful answer! Really nice you are willing to improve it based on suggestions from your users! 50 points is great and gives a lot of opportunity to finetune!

FedeDP, 2019-08-20:
> Thank you for your quick and helpful answer! Really nice you are willing to improve it based on suggestions from your users!

I think it is really nice that users are willing to spend time asking me to improve clight :) I am really proud of it!

Note that i am currently working on [libmodule](https://github.com/FedeDP/Clight/tree/libmodule) branch; thus i'll fix this feature request there as backporting the fix can be much troublesome given the number of changes.
It is working fine right now and it has a much better code; unfortunately it relies upon an unreleased libmodule version, thus it is not easy to try it out.

I'll merge it as soon as i feel ready to release both libmodule, clightd and clight (all together!).
Thanks for your patience!

ModDM, 2019-08-20:
Nice, I can see you are doing useful modifications in the background... Happy to hear users makes you glad and proud. We're a team!

FedeDP, 2019-08-20:
This is now fixed in libmodule branch: 651b7b04ad03cee128fb8a6dfac7f325be9beb33 !

ModDM, 2019-08-20:
Wow, you are quicker than the light! I will check it out!!

FedeDP, 2019-08-20:
> Wow, you are quicker than the light! I will check it out!!

I'm on vacation, so i have plenty of time :)
Note, as i said, that you need libmodule from master, clight from libmodule branch and clightd from libmodule5 branch. It is kinda hard to test it, sorry!

ModDM, 2019-08-20:
I see it's quite clear what the changes are that you did. Those changes (branch 651b7b0) are all changes you did? Can I also change it on my PC manually by following all edits you did, as displayed on that page? Then I don't need to use several download links / clone links.

FedeDP, 2019-08-20:
> Can I also change it on my PC manually by following all edits you did, as displayed on that page?

Yes you can manually port them into your local clight source.
You should change:
* src/commons.h
* src/conf/config.c 
* src/conf/opts.c
* src/utils/my_math.c
You can avoid porting src/modules/interface.c if you are not going to test it with dbus interface :)

ModDM, 2019-08-26:
Dear FedeDP,

Thank you for all advise and help. I tested this out carefully. First I thought I forgot to do some manual changes, as you did. But to prevent these mistakes I just replaced the files completely. It seems you forgot something to mention, because it seems you have done more. To be sure I have literally the same files as you, I replaced these 4 files with these 4 files as found in the libmodule branch. Everything works, but as soon as I even add 1 curve point to the curve as mentioned in this issue ticket, then clight says `"Wrong number of ac_backlight_regression_points array elements."`. I first thought that Clight is working, but as soon as I change something in the curve points of the config file I get this error and my changes are totally ignored by clight, even if I try to do an extreme change in the curve (all very low values), no reaction can be seen and clight ignores it.
So, there is changed something else.
In this issue you earlier wrote: 

> "unfortunately it relies upon an unreleased libmodule version, thus it is not easy to try it out"

I did not change anything in Libmodule on my computer. Is that the problem? Did I forget something there?

Further it could be useful to know where the message `"Wrong number of ac_backlight_regression_points array elements."`. comes from and which function / problem triggers this message, so it could be traced where this error comes from and what is missing.

Thank you very much in advance for your help,

With kind regards,

ModDM

ModDM, 2019-08-26:
For your info:
I did `sudo make install` and `sudo systemctl enable clightd.service`. The last step returned `[2]+  Killed                  clight` in my terminal, but I think this is the way to re-enable the service, just run it over the older service?

FedeDP, 2019-08-27:
> For your info:
I did sudo make install and sudo systemctl enable clightd.service. The last step returned [2]+ Killed clight in my terminal, but I think this is the way to re-enable the service, just run it over the older service?

Yes, no issues there!

> To be sure I have literally the same files as you, I replaced these 4 files with these 4 files as found in the libmodule branch. 

This is wrong! There have been so many changes there!
You should start from master branch and only port the aforementioned commit (manually).

> Everything works, but as soon as I even add 1 curve point to the curve as mentioned in this issue ticket, then clight says "Wrong number of ac_backlight_regression_points array elements."

This is generated by src/opts.c: https://github.com/FedeDP/Clight/blob/master/src/conf/opts.c#L275.

I tried on my pc (libmodule branch) and everything is working as expected; i can see different linear regression parameters set (i tried with 15 points).
No "Wrong number of..." log is present.

FedeDP, 2019-09-07:
The work is now in master. It does require libmodule 5.0.0 though (released but unavailable yet on AUR to avoid breaking current clight/clightd installations).

ModDM, 2019-09-08:
Dear FedeDP,

Thank you. I installed Libmodule 5.0 succesfully.
When I now try to install the new Clight I get this error at this install
step:

cmake \
>     -G "Unix Makefiles" \
>     -DCMAKE_INSTALL_PREFIX=/usr \
>     -DCMAKE_INSTALL_LIBDIR=lib \
>     -DCMAKE_BUILD_TYPE="Release" \
>     ..
-- Checking for modules 'popt;gsl;libconfig;libmodule>=5.0.0'
--
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367
(message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532
(_pkg_check_modules_internal)
  CMakeLists.txt:37 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/sysuser/Downloads/Clight/build/CMakeFiles/CMakeOutput.log".

What to do now?

With kind regards,

ModDM

ᐧ

FedeDP, 2019-09-08:
Weird! It does not mention which package is missing...have you installed pkg-config? On which OS are you on?

> See also "/home/sysuser/Downloads/Clight/build/CMakeFiles/CMakeOutput.log".

Can you share this file?

ModDM, 2019-09-08:
>
> have you installed pkg-config?
>

My terminal says "pkg-config/xenial,now 0.29.1-0ubuntu1 amd64 [installed]"

On which OS are you on?

Ubuntu 16.04 LTS

Attached you find my CMakeOutput.log file.

Yes, it is weird. It looks like something more has changed between the
previous Clight and this one, except Libmodule version and the Clight
files, but if you are sure nothing else has changed, then it's a weird
error.

Hopefully the attached file says something more (I can't find it). I
currently have the old Clight installed. I made sure Clight is killed
before I install the new one.

With kind regards,

ModDM

ᐧ

FedeDP, 2019-09-08:
> Attached you find my CMakeOutput.log file.

Mmmh i cannot see it :)

ModDM, 2019-09-08:
You can not see the error in my log or you can not see the attached log file? Have you tried to open the message via email? (I also sent my previous message via email).

FedeDP, 2019-09-08:
I cannot see attached file (even with gmail!)

ModDM, 2019-09-08:
Ok, no problem, I will send you directly outside Github. 1 moment...

FedeDP, 2019-09-08:
Great thanks!
Unfortunately it does not adds many informations...
Can you try to change [CMakeLists.txt line 37](https://github.com/FedeDP/Clight/blob/master/CMakeLists.txt#L37) removing ">=5.0.0" from?

ModDM, 2019-09-08:
Hey it WORKS for the cmake step!! Thank you.

Now I get errors on the make step. Just to keep it nice and happy...

Here are the errors:

`Scanning dependencies of target clight
[  4%] Building C object CMakeFiles/clight.dir/src/pubsub/topics.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/pubsub/validations.c.o
[ 13%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 18%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[ 22%] Building C object CMakeFiles/clight.dir/src/main.c.o
/home/sysuser/Downloads/Clight/src/main.c: In function ‘load_user_modules’:
/home/sysuser/Downloads/Clight/src/main.c:163:17: **warning**: implicit declaration of function ‘m_load’ [-Wimplicit-function-declaration]
             if (m_load(gl.gl_pathv[i]) == MOD_OK) {
                 ^
[ 27%] Building C object CMakeFiles/clight.dir/src/utils/timer.c.o
/home/sysuser/Downloads/Clight/src/utils/timer.c: In function ‘read_timer’:
/home/sysuser/Downloads/Clight/src/utils/timer.c:76:5: **warning**: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
     read(fd, &t, sizeof(uint64_t));
     ^
[ 31%] Building C object CMakeFiles/clight.dir/src/utils/my_math.c.o
[ 36%] Building C object CMakeFiles/clight.dir/src/utils/log.c.o
/home/sysuser/Downloads/Clight/src/utils/log.c: In function ‘open_log’:
/home/sysuser/Downloads/Clight/src/utils/log.c:20:13: **warning**: ignoring return value of ‘ftruncate’, declared with attribute warn_unused_result [-Wunused-result]
             ftruncate(fd, 0);
             ^
[ 40%] Building C object CMakeFiles/clight.dir/src/utils/idler.c.o
[ 45%] Building C object CMakeFiles/clight.dir/src/modules/dpms.c.o
/home/sysuser/Downloads/Clight/src/modules/dpms.c: In function ‘init’:
/home/sysuser/Downloads/Clight/src/modules/dpms.c:22:9: warning: implicit declaration of function ‘m_poisonpill’ [-Wimplicit-function-declaration]
         m_poisonpill(self());
         ^
/home/sysuser/Downloads/Clight/src/modules/dpms.c:22:22: **warning**: implicit declaration of function ‘self’ [-Wimplicit-function-declaration]
         m_poisonpill(self());
                      ^
In file included from /home/sysuser/Downloads/Clight/src/commons.h:9:0,
                 from /home/sysuser/Downloads/Clight/src/utils/timer.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/bus.h:4,
                 from /home/sysuser/Downloads/Clight/src/utils/idler.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/dpms.c:1:
/home/sysuser/Downloads/Clight/src/modules/dpms.c: In function ‘receive’:
/home/sysuser/Downloads/Clight/src/public.h:19:58: **error**: ‘msg_t {aka const struct <anonymous>}’ has no member named ‘ps_msg’
 #define MSG_TYPE()                  msg->is_pubsub ? (msg->ps_msg->type == USER ? ((message_t *)msg->ps_msg->message)->type : SYSTEM_UPD) : FD_UPD
                                                          ^
/home/sysuser/Downloads/Clight/src/modules/dpms.c:35:13: **note**: in expansion of macro ‘MSG_TYPE’
     switch (MSG_TYPE()) {
             ^
/home/sysuser/Downloads/Clight/src/public.h:19:100: **error**: ‘msg_t {aka const struct <anonymous>}’ has no member named ‘ps_msg’
 #define MSG_TYPE()                  msg->is_pubsub ? (msg->ps_msg->type == USER ? ((message_t *)msg->ps_msg->message)->type : SYSTEM_UPD) : FD_UPD
                                                                                                    ^
/home/sysuser/Downloads/Clight/src/modules/dpms.c:35:13: **note**: in expansion of macro ‘MSG_TYPE’
     switch (MSG_TYPE()) {
             ^
In file included from /home/sysuser/Downloads/Clight/src/commons.h:9:0,
                 from /home/sysuser/Downloads/Clight/src/utils/timer.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/bus.h:4,
                 from /home/sysuser/Downloads/Clight/src/utils/idler.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/dpms.c:1:
/home/sysuser/Downloads/Clight/src/public.h:20:52: **error**: ‘msg_t {aka const struct <anonymous>}’ has no member named ‘ps_msg’
 #define MSG_DATA()                  ((uint8_t *)msg->ps_msg->message + offsetof(message_t, loc)) // offsetof any of the internal data structure to actually account for padding
                                                    ^
/home/sysuser/Downloads/Clight/src/modules/dpms.c:43:46: note: in expansion of macro ‘MSG_DATA’
             timeout_upd *up = (timeout_upd *)MSG_DATA();
                                              ^
/home/sysuser/Downloads/Clight/src/modules/dpms.c: In function ‘on_new_idle’:
/home/sysuser/Downloads/Clight/src/modules/dpms.c:72:27: error: macro "m_publish" passed 4 arguments, but takes just 3
         M_PUB(&display_req);
                           ^
In file included from /home/sysuser/Downloads/Clight/src/commons.h:9:0,
                 from /home/sysuser/Downloads/Clight/src/utils/timer.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/bus.h:4,
                 from /home/sysuser/Downloads/Clight/src/utils/idler.h:3,
                 from /home/sysuser/Downloads/Clight/src/modules/dpms.c:1:
/home/sysuser/Downloads/Clight/src/public.h:24:37: **error**: ‘m_publish’ undeclared (first use in this function)
 #define M_PUB(ptr)                  m_publish(topics[(ptr)->type], ptr, sizeof(message_t), false);
                                     ^
[... cut ...]
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/clight.dir/all' failed
make[1]: *** [CMakeFiles/clight.dir/all] **Error 2**
Makefile:149: recipe for target 'all' failed
make: *** [all] **Error 2**
`

By the way, glad you could put everything in master! For me it is a great help to be able to use the master instead of doing all commits, for which I planned to do some study in that, to know how it works (follow a course), but this is a lot easier. Now the installation error as final challenge...

FedeDP, 2019-09-08:
Mmh...ok it seems libmodule 5.0.0 is not installed :)
How did you install it?

ModDM, 2019-09-08:
OK, good to know.

> How did you install it?

With my computer!
Ha,. better answer:
I used the same install instructions you gave me earlier. I'll copy them for you here, so you can see what I did:

# step 1: install Libmodule
# Download latest libmodule from https://github.com/FedeDP/libmodule/releases/latest
# extract the tar.gz, cd into extracted folder (any folder, just a temp folder, such as Downloads)
# Follow libmodule build-how-to plus installation for Debian:
# https://github.com/FedeDP/libmodule#build-dep-and-how-to-build

Actually that went very fine and quick. But you are right. My terminal says:

`libmodule/now 4.2.0 amd64 [installed,local]`

Which step do I need to do different?

ModDM, 2019-09-08:
SOLVED!

I forgot to run:
`sudo dpkg -i libmodule-5.0.0-Linux.deb`

Sorry for wasting your time! An accident. I now try to install Clight, which should be go fine.

FedeDP, 2019-09-08:
Great! Ahah

ModDM, 2019-09-08:
Yes, looks better now. Not totally without errors. Hopefully I may ask it, you might know what I do wrong.

Here is the error:

`~/Downloads/Clight/build$ cmake \
>     -G "Unix Makefiles" \
>     -DCMAKE_INSTALL_PREFIX=/usr \
>     -DCMAKE_INSTALL_LIBDIR=lib \
>     -DCMAKE_BUILD_TYPE="Release" \
>     ..
-- Found libsystemd version 229
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sysuser/Downloads/Clight/build
sysuser@sysuser-HP-250-G6-Notebook-PC:~/Downloads/Clight/build$ make
[  4%] Building C object CMakeFiles/clight.dir/src/modules/dpms.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/modules/upower.c.o
[ 13%] Building C object CMakeFiles/clight.dir/src/modules/gamma.c.o
[ 18%] Building C object CMakeFiles/clight.dir/src/modules/interface.c.o
[ 22%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 27%] Building C object CMakeFiles/clight.dir/src/modules/dimmer.c.o
[ 31%] Building C object CMakeFiles/clight.dir/src/modules/signal.c.o
[ 36%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
[ 40%] Building C object CMakeFiles/clight.dir/src/modules/display.c.o
[ 45%] Building C object CMakeFiles/clight.dir/src/modules/location.c.o
[ 50%] Building C object CMakeFiles/clight.dir/src/modules/inhibit.c.o
[ 54%] Building C object CMakeFiles/clight.dir/src/modules/screen.c.o
[ 59%] Linking C executable clight
CMakeFiles/clight.dir/src/main.c.o: In function `load_user_modules':
main.c:(.text+0xd1): undefined reference to `m_load'
collect2: error: ld returned 1 exit status
CMakeFiles/clight.dir/build.make:614: recipe for target 'clight' failed
make[2]: *** [clight] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/clight.dir/all' failed
make[1]: *** [CMakeFiles/clight.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
`

FedeDP, 2019-09-08:
Wow! This is surprising! It does not find a symbol from libmodule 5.0.0.
Can you attach here your `/usr/include/module/module_easy.h` file?

ModDM, 2019-09-08:
Sure, here he is. I changed the extension .h to .txt, because .h is not supported here on Github, when you try to add a file.
[module_easy.txt](https://github.com/FedeDP/Clight/files/3587978/module_easy.txt)

FedeDP, 2019-09-08:
> #define m_load(path)                            module_load(path, MODULES_DEFAULT_CTX)
> #define m_unload(path)                          module_unload(path) // just an alias

Ok, the header is correct and the symbols are there.  
Can you paste output of: 
`find /usr/lib -name "libmodu*"`
and
`ls -l /usr/lib/libmodule.so`
?

ModDM, 2019-09-08:
`find /usr/lib -name "libmodu*"
/usr/lib/libmodule.so.5
/usr/lib/libmodule.so.5.0.0
/usr/lib/libmodule.so
`

`ls -l /usr/lib/libmodule.so
lrwxrwxrwx 1 root root 14 Sep  8 21:19 /usr/lib/libmodule.so -> libmodule.so.5`

FedeDP, 2019-09-08:
Ok everything is correct.
What's the output of: `nm -D /usr/lib/libmodule.so` instead?

ModDM, 2019-09-08:
```
nm -D /usr/lib/libmodule.so
00000000002081d0 B __bss_start
                 U calloc
                 U close
                 w __cxa_finalize
                 U dlclose
                 U dlerror
                 U dlopen
                 U dup
00000000002081d0 D _edata
0000000000208240 B _end
                 U epoll_create1
                 U epoll_ctl
                 U epoll_wait
                 U __errno_location
                 U fcntl
0000000000005b34 T _fini
                 U __fprintf_chk
                 U free
                 w __gmon_start__
0000000000001dc0 T _init
                 w _ITM_deregisterTMCloneTable
                 w _ITM_registerTMCloneTable
                 w _Jv_RegisterClasses
00000000000021b0 W main
                 U malloc
00000000000031b0 T map_clear
00000000000031f0 T map_free
00000000000032a0 T map_get
00000000000033d0 T map_has_key
00000000000033f0 T map_iterate
0000000000002e40 T map_itr_get_data
0000000000002e20 T map_itr_get_key
0000000000003170 T map_itr_new
0000000000002d50 T map_itr_next
0000000000002df0 T map_itr_remove
0000000000002e60 T map_itr_set_data
0000000000003150 T map_length
0000000000003230 T map_new
0000000000002e80 T map_put
00000000000034d0 T map_remove
                 U memcpy
0000000000004560 T module_become
0000000000005700 T module_broadcast
0000000000004340 T module_deregister
0000000000004890 T module_deregister_fd
0000000000004a40 T module_dump
00000000000049e0 T module_get_context
00000000000049a0 T module_get_name
0000000000004a20 T module_is
0000000000004470 T module_load
0000000000004620 T module_log
0000000000004bc0 T module_pause
0000000000005860 T module_poisonpill
00000000000055b0 T module_publish
00000000000051c0 T module_ref
0000000000004090 T module_register
0000000000004760 T module_register_fd
0000000000004c10 T module_resume
00000000000028b0 T modules_ctx_dispatch
0000000000002860 T modules_ctx_get_fd
00000000000026e0 T modules_ctx_loop_events
0000000000002810 T modules_ctx_quit
0000000000002690 T modules_ctx_set_logger
0000000000004720 T module_set_userdata
0000000000002180 W modules_pre_start
00000000000025e0 T modules_set_memhook
0000000000004b70 T module_start
0000000000004c60 T module_stop
0000000000005240 T module_subscribe
0000000000005440 T module_tell
00000000000045c0 T module_unbecome
0000000000004510 T module_unload
00000000000053c0 T module_unsubscribe
                 U pipe
                 U __printf_chk
                 U pthread_create
                 U pthread_join
0000000000003880 T queue_clear
0000000000003820 T queue_dequeue
00000000000036d0 T queue_enqueue
00000000000038d0 T queue_free
00000000000037b0 T queue_iterate
0000000000003690 T queue_itr_get_data
0000000000003770 T queue_itr_new
0000000000003650 T queue_itr_next
00000000000036b0 T queue_itr_set_data
0000000000003750 T queue_length
0000000000003620 T queue_new
0000000000003900 T queue_peek
                 U read
                 U realloc
                 U regcomp
                 U regexec
                 U regfree
                 U __stack_chk_fail
0000000000003b60 T stack_clear
0000000000003bb0 T stack_free
0000000000003aa0 T stack_iterate
00000000000039a0 T stack_itr_get_data
0000000000003a60 T stack_itr_new
0000000000003960 T stack_itr_next
00000000000039c0 T stack_itr_set_data
0000000000003a40 T stack_length
0000000000003930 T stack_new
0000000000003be0 T stack_peek
0000000000003b10 T stack_pop
00000000000039e0 T stack_push
                 U stderr
                 U stdout
                 U strcmp
                 U strerror
                 U strlen
                 U __vfprintf_chk
                 U write

```

FedeDP, 2019-09-08:
> 0000000000004470 T module_load

Ok, clight is linking a wrong shared object i guess. 
First rule for any PC issue: can you reboot? :D

ModDM, 2019-09-08:
Haha, I was afraid it was that simple... Maybe that is a good idea!

BTW here also the output of the other command:

```
readelf -Ws /usr/lib/libmodule.so

Symbol table '.dynsym' contains 116 entries:
   Num:    Value          Size Type    Bind   Vis      Ndx Name
     0: 0000000000000000     0 NOTYPE  LOCAL  DEFAULT  UND 
     1: 0000000000001dc0     0 SECTION LOCAL  DEFAULT    9 
     2: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND free@GLIBC_2.2.5 (2)
     3: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND pthread_create@GLIBC_2.2.5 (3)
     4: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND __vfprintf_chk@GLIBC_2.3.4 (4)
     5: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND __errno_location@GLIBC_2.2.5 (3)
     6: 0000000000000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_deregisterTMCloneTable
     7: 0000000000000000     0 OBJECT  GLOBAL DEFAULT  UND stdout@GLIBC_2.2.5 (2)
     8: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND fcntl@GLIBC_2.2.5 (3)
     9: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND write@GLIBC_2.2.5 (3)
    10: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND regfree@GLIBC_2.2.5 (2)
    11: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND strlen@GLIBC_2.2.5 (2)
    12: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND __stack_chk_fail@GLIBC_2.4 (5)
    13: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND dup@GLIBC_2.2.5 (2)
    14: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND close@GLIBC_2.2.5 (3)
    15: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND pipe@GLIBC_2.2.5 (2)
    16: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND regcomp@GLIBC_2.2.5 (2)
    17: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND epoll_ctl@GLIBC_2.3.2 (6)
    18: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND read@GLIBC_2.2.5 (3)
    19: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND calloc@GLIBC_2.2.5 (2)
    20: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND strcmp@GLIBC_2.2.5 (2)
    21: 0000000000000000     0 NOTYPE  WEAK   DEFAULT  UND __gmon_start__
    22: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND memcpy@GLIBC_2.14 (7)
    23: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND dlopen@GLIBC_2.2.5 (8)
    24: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND malloc@GLIBC_2.2.5 (2)
    25: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND regexec@GLIBC_2.3.4 (4)
    26: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND dlclose@GLIBC_2.2.5 (8)
    27: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND realloc@GLIBC_2.2.5 (2)
    28: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND __printf_chk@GLIBC_2.3.4 (4)
    29: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND epoll_wait@GLIBC_2.3.2 (6)
    30: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND pthread_join@GLIBC_2.2.5 (3)
    31: 0000000000000000     0 NOTYPE  WEAK   DEFAULT  UND _Jv_RegisterClasses
    32: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND __fprintf_chk@GLIBC_2.3.4 (4)
    33: 0000000000000000     0 NOTYPE  WEAK   DEFAULT  UND _ITM_registerTMCloneTable
    34: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND strerror@GLIBC_2.2.5 (2)
    35: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND epoll_create1@GLIBC_2.9 (9)
    36: 0000000000000000     0 FUNC    WEAK   DEFAULT  UND __cxa_finalize@GLIBC_2.2.5 (2)
    37: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND dlerror@GLIBC_2.2.5 (8)
    38: 0000000000000000     0 OBJECT  GLOBAL DEFAULT  UND stderr@GLIBC_2.2.5 (2)
    39: 00000000000039e0    85 FUNC    GLOBAL DEFAULT   12 stack_push
    40: 0000000000002180     2 FUNC    WEAK   DEFAULT   12 modules_pre_start
    41: 0000000000004340   291 FUNC    GLOBAL DEFAULT   12 module_deregister
    42: 0000000000002e20    27 FUNC    GLOBAL DEFAULT   12 map_itr_get_key
    43: 0000000000003be0    36 FUNC    GLOBAL DEFAULT   12 stack_peek
    44: 0000000000004bc0    66 FUNC    GLOBAL DEFAULT   12 module_pause
    45: 0000000000004090   679 FUNC    GLOBAL DEFAULT   12 module_register
    46: 0000000000005700   352 FUNC    GLOBAL DEFAULT   12 module_broadcast
    47: 00000000000051c0   118 FUNC    GLOBAL DEFAULT   12 module_ref
    48: 00000000000025e0   166 FUNC    GLOBAL DEFAULT   12 modules_set_memhook
    49: 0000000000003150    18 FUNC    GLOBAL DEFAULT   12 map_length
    50: 00000000000039c0    25 FUNC    GLOBAL DEFAULT   12 stack_itr_set_data
    51: 0000000000004760   295 FUNC    GLOBAL DEFAULT   12 module_register_fd
    52: 0000000000002690    67 FUNC    GLOBAL DEFAULT   12 modules_ctx_set_logger
    53: 0000000000004890   271 FUNC    GLOBAL DEFAULT   12 module_deregister_fd
    54: 00000000000055b0   323 FUNC    GLOBAL DEFAULT   12 module_publish
    55: 00000000000036d0   125 FUNC    GLOBAL DEFAULT   12 queue_enqueue
    56: 00000000000049e0    60 FUNC    GLOBAL DEFAULT   12 module_get_context
    57: 0000000000005b34     0 FUNC    GLOBAL DEFAULT   13 _fini
    58: 0000000000003b10    68 FUNC    GLOBAL DEFAULT   12 stack_pop
    59: 0000000000003230   107 FUNC    GLOBAL DEFAULT   12 map_new
    60: 0000000000002df0    47 FUNC    GLOBAL DEFAULT   12 map_itr_remove
[... 231 lines cut ...]
   173: 0000000000003a40    17 FUNC    GLOBAL DEFAULT   12 stack_length
   174: 00000000000039e0    85 FUNC    GLOBAL DEFAULT   12 stack_push
   175: 0000000000004620   250 FUNC    GLOBAL DEFAULT   12 module_log
   176: 0000000000003a60    52 FUNC    GLOBAL DEFAULT   12 stack_itr_new
   177: 0000000000003820    84 FUNC    GLOBAL DEFAULT   12 queue_dequeue
[... cut ...]
   189: 0000000000004890   271 FUNC    GLOBAL DEFAULT   12 module_deregister_fd
   190: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND epoll_create1@@GLIBC_2.9
   191: 0000000000000000     0 FUNC    WEAK   DEFAULT  UND __cxa_finalize@@GLIBC_2.2.5
   192: 0000000000001dc0     0 FUNC    GLOBAL DEFAULT    9 _init
   193: 0000000000003aa0   100 FUNC    GLOBAL DEFAULT   12 stack_iterate
   194: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND dlerror@@GLIBC_2.2.5
   195: 0000000000000000     0 OBJECT  GLOBAL DEFAULT  UND stderr@@GLIBC_2.2.5

```

ModDM, 2019-09-08:
After reboot, what would be wise to do first? Run some particular command or something else?

FedeDP, 2019-09-08:
>    114: 0000000000004470   152 FUNC    GLOBAL DEFAULT   12 module_load

Great...it is global thus it is visible!

> After reboot, what would be wise to do first? Run some particular command or something else?

I'd retry a build; and if it doesn't work, clean up clight build/ folder and run again cmake!

ModDM, 2019-09-08:
Ok, clear. I will do that now.

ModDM, 2019-09-08:
Voila!

`cmake \
>     -G "Unix Makefiles" \
>     -DCMAKE_INSTALL_PREFIX=/usr \
>     -DCMAKE_INSTALL_LIBDIR=lib \
>     -DCMAKE_BUILD_TYPE="Release" \
>     ..
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for modules 'popt;gsl;libconfig;libmodule>=5.0.0'
--   Found popt, version 1.16
--   Found gsl, version 2.1
--   Found libconfig, version 1.5
--   Found libmodule, version 5.0.0
-- Checking for one of the modules 'libelogind;libsystemd>=221'
-- Found libsystemd version 229
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sysuser/Downloads/Clight/build
sysuser@sysuser-HP-250-G6-Notebook-PC:~/Downloads/Clight/build$ make
Scanning dependencies of target clight
[  4%] Building C object CMakeFiles/clight.dir/src/pubsub/topics.c.o
[  9%] Building C object CMakeFiles/clight.dir/src/pubsub/validations.c.o
[ 13%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 18%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[ 22%] Building C object CMakeFiles/clight.dir/src/main.c.o
[ 27%] Building C object CMakeFiles/clight.dir/src/utils/timer.c.o
/home/sysuser/Downloads/Clight/src/utils/timer.c: In function ‘read_timer’:
/home/sysuser/Downloads/Clight/src/utils/timer.c:76:5: warning: ignoring return value of ‘read’, declared with attribute warn_unused_result [-Wunused-result]
     read(fd, &t, sizeof(uint64_t));
     ^
[ 31%] Building C object CMakeFiles/clight.dir/src/utils/my_math.c.o
[ 36%] Building C object CMakeFiles/clight.dir/src/utils/log.c.o
/home/sysuser/Downloads/Clight/src/utils/log.c: In function ‘open_log’:
/home/sysuser/Downloads/Clight/src/utils/log.c:20:13: warning: ignoring return value of ‘ftruncate’, declared with attribute warn_unused_result [-Wunused-result]
             ftruncate(fd, 0);
             ^
[ 40%] Building C object CMakeFiles/clight.dir/src/utils/idler.c.o
[ 45%] Building C object CMakeFiles/clight.dir/src/modules/dpms.c.o
[ 50%] Building C object CMakeFiles/clight.dir/src/modules/upower.c.o
[ 54%] Building C object CMakeFiles/clight.dir/src/modules/gamma.c.o
[ 59%] Building C object CMakeFiles/clight.dir/src/modules/interface.c.o
[ 63%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 68%] Building C object CMakeFiles/clight.dir/src/modules/dimmer.c.o
[ 72%] Building C object CMakeFiles/clight.dir/src/modules/signal.c.o
[ 77%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
[ 81%] Building C object CMakeFiles/clight.dir/src/modules/display.c.o
[ 86%] Building C object CMakeFiles/clight.dir/src/modules/location.c.o
[ 90%] Building C object CMakeFiles/clight.dir/src/modules/inhibit.c.o
[ 95%] Building C object CMakeFiles/clight.dir/src/modules/screen.c.o
[100%] Linking C executable clight
[100%] Built target clight
sysuser@sysuser-HP-250-G6-Notebook-PC:~/Downloads/Clight/build$ sudo make install
[100%] Built target clight
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/bin/clight
-- Installing: /usr/include/clight/public.h
-- Installing: /usr/share/dbus-1/services/org.clight.clight.service
-- Installing: /etc/default/clight.conf
-- Installing: /etc/xdg/autostart/clight.desktop
-- Installing: /usr/share/applications/clightc.desktop
-- Installing: /usr/share/icons/hicolor/scalable/apps/clight.svg
-- Installing: /usr/share/clight/daytime.skel
-- Installing: /usr/share/clight/inhibit_bl.skel
-- Installing: /usr/share/clight/modules.d/
-- Installing: /usr/share/bash-completion/completions/clight
sysuser@sysuser-HP-250-G6-Notebook-PC:~/Downloads/Clight/build$ 
`

FedeDP, 2019-09-08:
I'd like to know what the hell happened before :D 
Great though, have fun testing it!
(and thanks for your time!)

ModDM, 2019-09-08:
Restarting helped. Everything installed succesful. Thank you VERY much for all your time. You know a lot about your program, and sometimes you know there is just such a solution like this. You are welcome. YOU thanks for your time!

closed 2019-09-10 by ModDM