FedeDP/Clight #96: Clight stopped working after using Skype?

sveinungss, 2019-08-30:
I'm on Antergos (with Xfce 4.14), and Clight has been working perfectly for months. After using Skype the other day, Clight seemed to stop working. I am not 100% confident that Skype is to blame, but the two events coincided. Now, I am unable to make Clight work. The service is enabled and running, but nothing happens when I run Clight or trigger a reading. No light next to the webcam to indicate that it is active, and no change in screen brightness. I've tried editing the config to make sure that Clightd is addressing the right adapter (Video0), to no avail. I would really appreciate help as to how I might further troubleshoot the issue.

FedeDP, 2019-08-31:
Hi!
Thanks for using Clight and opening this issue!

This is quite weird. Can you attach a $HOME/.clight.log and a `journalctl -b --unit=clight`?

sveinungss, 2019-09-01:
Of course, thanks for responding so quickly.

`$ journalctl -b --unit=clight
-- Logs begin at Tue 2019-05-14 13:52:10 CEST, end at Sun 2019-09-01 14:29:24 C>
-- No entries --
`

[clight.log](https://github.com/FedeDP/Clight/files/3563464/clight.log)

FedeDP, 2019-09-01:
Clight log is clear!
What do you mean with:
>  Clight seemed to stop working

? Do you mean everything or backlight autocalibration only?

sveinungss, 2019-09-01:
Everything. It is as if it is not installed at all.e

FedeDP, 2019-09-02:
Your Clight log does indeed stop at 29 August; does it mean clight is not starting anymore, or you uploaded an old log?  

In the former case, can you manually run clight in verbose mode, ie:
`clight --verbose`
and share here the output?
Thanks!

sveinungss, 2019-09-02:
Thanks for your patience! `clight --verbose` returns:
`clight: error while loading shared libraries: libgsl.so.23: cannot open shared object file: No such file or directory`

I think we're getting somewhere :) Although you'll have to tell me what to do about it!

FedeDP, 2019-09-02:
Ah-ah! It seems libgsl was uninstalled someway! Clight requires libgsl shared object.  
Ubuntu version? Can you try to install libgsl23 and run `clight --verbose` again?

sveinungss, 2019-09-02:
Question for clarification: gsl  2.6-1 is installed, is libgsl23 not part of/related to this? I can't find libgsl23 searching AUR.

FedeDP, 2019-09-02:
Oh i see, you are using Arch!
It seems there was a gsl bump recently (26 august) and gsl 2.6-1 now installs a shiny new library version: `usr/lib/libgsl.so.25`.
You should rebuild clight from aur and it should be fine!

sveinungss, 2019-09-04:
Excellent, thank you! Installing from source (clight-git and clightd-git) does indeed seem to have solved the problem.

closed 2019-09-04 by sveinungss