FedeDP/Clight #98: Failed to create org.freedesktop.ScreenSaver

ModDM, 2019-09-08:
I installed Clight from the latest download files (from master) and I get this error at clight startup. Because of this error, the program does not start.

```
clight
Config file /home/sysuser/.config/clight.conf not found.
Clightd found, version: 4.0.
Initial AC state: connected.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Error loading from cache file.

```

NICHOLAS85, 2019-09-08:
I had the same problem, I simply issued the command `systemctl restart clightd` which seemed to fix this error for me.

ModDM, 2019-09-08:
That would be excellent. I just tried it and my terminal returns this:

`Failed to restart clightd.service: Unit polkit.service not found.`

ModDM, 2019-09-08:
I also restarted my PC several times.

NICHOLAS85, 2019-09-08:
Ah, then I'm unsure why your clight fails to start. Upon closer inspection, I seem to have the same `org.freedesktop.ScreenSaver` error but it doesn't seem to cause my clight to fail to function. Maybe it's something related to the `Error loading from cache file message`?

FedeDP, 2019-09-09:
Hi! About org.freedesktop.ScreenSaver error, have a read here: https://github.com/FedeDP/Clight/issues/76#issuecomment-524611752 .

Unfortunately that is the best we could achieve. I have the same issue (i use kde plasma).

> That would be excellent. I just tried it and my terminal returns this:
Failed to restart clightd.service: Unit polkit.service not found.

This is weird; polkit systemd unit is called "polkitd.service" in ubuntu 16.04 (there is even a comment in Clightd CMakeLists.txt: https://github.com/FedeDP/Clightd/blob/master/CMakeLists.txt#L104).
The dependency should be automatically set for you through clightd cmake. If that's not the case, please open an issue on Clightd!

> Because of this error, the program does not start.

Program does not start because of polkitd issue :)

closed 2019-09-09 by FedeDP