FedeDP/Clight #100: Clight doesn't run in terminal due to library error

Tasty213, 2019-09-16:
When I try and run clight from a terminal (using `clight`) the following error is produced. 
```
clight: error while loading shared libraries: libgsl.so.23: cannot open shared object file: No such file or directory
```
I am using Manjaro Linux KDE edition latest stable update.

FedeDP, 2019-09-16:
Hi!
Thanks for opening this issue. Special thanks for opening issue n 100!!! :)
You probably have testing repos enabled, are you?
I see in Extra/gsl, gsl is at 2.5-1, while in testing it is at 2.6-1 version. You probably updated gsl in between (and clightd stopped working as gsl bumped its soversion).
Just rebuilding clight from scrach should fix the issue for you.

FedeDP, 2019-09-19:
Closing as fixed in #101 (that duplicated this issue).

closed 2019-09-19 by FedeDP