FedeDP/Clight #101: [BUG]  clightd service does not start :(

RezaSh13, 2019-09-18:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> ...

**Used distribution:**
> ...

**Describe the bug**
A clear and concise description of what the bug is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior.

Please attach a Clight verbose log too!

```

  plasma  ~  systemctl status clightd.service
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
   Loaded: loaded (/usr/lib/systemd/system/clightd.service; enabled; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-09-18 13:18:04 +0430; 2s ago
  Process: 877 ExecStart=/usr/lib/clightd/clightd (code=exited, status=127)
 Main PID: 877 (code=exited, status=127)

 plasma  ~  systemctl status clightd.service 
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
   Loaded: loaded (/usr/lib/systemd/system/clightd.service; enabled; vendor preset: disabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-09-18 13:18:04 +0430; 2s ago
  Process: 877 ExecStart=/usr/lib/clightd/clightd (code=exited, status=127)
 Main PID: 877 (code=exited, status=127)

Sep 18 13:18:04 plasma systemd[1]: clightd.service: Service RestartSec=5s expired, scheduling restart.
Sep 18 13:18:04 plasma systemd[1]: clightd.service: Scheduled restart job, restart counter is at 25.
Sep 18 13:18:04 plasma clightd[877]: /usr/lib/clightd/clightd: error while loading shared libraries: libddcutil.so.0: cannot open shared object file:>
Sep 18 13:18:04 plasma systemd[1]: Stopped Bus service to manage various screen related properties (gamma, dpms, backlight).
Sep 18 13:18:04 plasma systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Sep 18 13:18:04 plasma systemd[1]: clightd.service: Main process exited, code=exited, status=127/n/a
Sep 18 13:18:04 plasma systemd[1]: clightd.service: Failed with result 'exit-code'.
Sep 18 13:18:04 plasma systemd[1]: Failed to start Bus service to manage various screen related properties (gamma, dpms, backlight).

 plasma  ~  3 :( uname -a
Linux plasma 4.19.69-1-MANJARO #1 SMP PREEMPT Thu Aug 29 08:51:46 UTC 2019 x86_64 GNU/Linux

 plasma  ~  sudo systemctl restart clightd.service
Job for clightd.service failed because the control process exited with error code.
See "systemctl status clightd.service" and "journalctl -xe" for details.

 plasma  ~  1 :( journalctl -xe 
-- The job identifier is 3945 and the job result is done.
Sep 18 13:19:53 plasma audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/syst>
Sep 18 13:19:53 plasma audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/syste>
Sep 18 13:19:53 plasma kernel: audit: type=1130 audit(1568796593.013:184): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd>
Sep 18 13:19:53 plasma kernel: audit: type=1131 audit(1568796593.013:185): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd>
Sep 18 13:19:53 plasma systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
-- Subject: A start job for unit clightd.service has begun execution
-- Defined-By: systemd
-- Support: https://forum.manjaro.org/c/technical-issues-and-assistance
-- 
-- A start job for unit clightd.service has begun execution.
-- 
-- The job identifier is 3945.
Sep 18 13:19:53 plasma clightd[1185]: /usr/lib/clightd/clightd: error while loading shared libraries: libddcutil.so.0: cannot open shared object file>
Sep 18 13:19:53 plasma systemd[1]: clightd.service: Main process exited, code=exited, status=127/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://forum.manjaro.org/c/technical-issues-and-assistance
-- 
-- An ExecStart= process belonging to unit clightd.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 127.
Sep 18 13:19:53 plasma systemd[1]: clightd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://forum.manjaro.org/c/technical-issues-and-assistance
-- 
-- The unit clightd.service has entered the 'failed' state with result 'exit-code'.
Sep 18 13:19:53 plasma systemd[1]: Failed to start Bus service to manage various screen related properties (gamma, dpms, backlight).
-- Subject: A start job for unit clightd.service has failed
-- Defined-By: systemd
-- Support: https://forum.manjaro.org/c/technical-issues-and-assistance
-- 
-- A start job for unit clightd.service has finished with a failure.
-- 
-- The job identifier is 3945 and the job result is failed.
Sep 18 13:19:53 plasma audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/syst>
Sep 18 13:19:53 plasma kernel: audit: type=1130 audit(1568796593.023:186): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=clightd comm="systemd>

 plasma  ~  clight --version 
clight: error while loading shared libraries: libgsl.so.23: cannot open shared object file: No such file or directory

4 aur/clightd 3.4-1 (+4 1.36%) (Installed)
    Bus interface to change screen brightness and capture frames from webcam.
3 aur/clight 3.1-3 (+6 2.32%) (Installed)

FedeDP, 2019-09-18:
It seems like both clightd and clight are having issues with their library dependencies.  
Can you try to rebuild both of them from aur?
Recently ddcutil was updated and its library soversion was bumped; moreover it seems libgsl too is giving some issues. I see that gsl was recently updated to 2.6-1 in Archlinux (and manjaro too!) and it bumped libgsl soversion.

RezaSh13, 2019-09-18:
> It seems like both clightd and clight are having issues with their library dependencies.
> Can you try to rebuild both of them from aur?
> Recently ddcutil was updated and its library soversion was [...]

The problem was solved; i rebuild both of them from aur.

closed 2019-09-18 by RezaSh13