FedeDP/Clight #102: [BUG] Some log messages reporting incorrect value

NICHOLAS85, 2019-10-03:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Software version: 4.0

Commit 338125b6d904483ae29d3c9755cd78fa67e2da5c (latest as of bug report)

**Used distribution:**
> Debian Testing

**Describe the bug**
Some log messages return incorrect values of what they're reporting on. See following log for examples:
<details>
<summary>clight.log</summary>

```sh
tail: /home/nicholas/.local/share/clight/clight.log: file truncated
Clight
* Software version:             4.0
* Global config dir:            /etc/default
* Global data dir:              /usr/share/clight
* Starting time:                Thu Oct  3 18:16:08 2019

Starting options:

### Generic ###
* Verbose (debugging):          Disabled
* Autocalibration:              Enabled
* Inhibit autocalibration:              Enabled
* Keyboard backlight:           Disabled
* Number of captures:           5
* Webcam device:                Unset
* Backlight path:               Unset
* Shutter threshold:            0.00
* Ambient gamma:                Disabled
* Screen contrib:               0.09
* Screen samples:               10
* User position:                Unset
* Daily screen temp:            6500
* Nightly screen temp:          4000
* User set sunrise:             Unset
* User set sunset:              Unset
* Event duration:               1800
* Dimmer backlight pct:         0.20

### Timeouts ###
* Daily timeouts:               AC 900  BATT 1800
* Nightly timeout:              AC 2700 BATT 5400
* Event timeouts:               AC 300  BATT 600
* Dimmer timeouts:              AC 45   BATT 20
* Dpms timeouts:                AC 900  BATT 300
* Screen timeouts:              AC 30   BATT -1

### Modules ###
* Backlight:            Enabled
* Gamma:                Disabled
* Dimmer:               Disabled
* Dpms:         Disabled
* Screen:               Enabled

### Smooth ###
* Bright smooth trans:          Enabled
* Gamma smooth trans:           Enabled
* Dimmer smooth trans:          ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:          0.01
* Gamma smooth steps:           50
* Dimmer smooth steps:          ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:                30
* Gamma smooth timeout:         300
* Dimmer smooth timeout:                ENTER: 30, EXIT: 30
* Gamma long transition:                Disabled

(I)[18:16:08]{main.c:134}       Clightd found, version: 4.0.
(I)[18:16:08]{main.c:164}       '/home/nicholas/.local/share/clight/modules.d/inhibit_bl_mod' loaded.
(I)[18:16:08]{upower.c:68}      Initial AC state: connected.
(W)[18:16:08]{interface.c:246}  Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[18:16:08]{interface.c:247}  Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(I)[18:16:08]{backlight.c:230}  Keyboard backlight calibration enabled.
(I)[18:16:09]{backlight.c:251}  Ambient brightness: 0.494 -> Backlight pct: 0.623.
(I)[18:16:15]{upower.c:38}      AC cable connected.
(I)[18:16:16]{backlight.c:251}  Ambient brightness: 0.458 -> Backlight pct: 0.370.
(I)[18:16:24]{upower.c:38}      AC cable disconnected.
(I)[18:16:24]{backlight.c:251}  Ambient brightness: 0.468 -> Backlight pct: 0.595.
```
</details>

Clights starts while AC power is connected, correctly reporting so in the log. However when the AC power is disconnected, a message is logged reporting `(I)[18:16:15]{upower.c:38}      AC cable connected.`, which is incorrect. The same happens when connecting the cable but the message now says AC cable disconnected.

A second example I see is Keyboard backlight calibration is disabled in the conf file, which is reported correctly in the variable summary at the top of the log. However a message appears stating `(I)[18:16:08]{backlight.c:230}  Keyboard backlight calibration enabled.`, which is incorrect. This message seems to stay the same regardless of what the Keyboard backlight calibration variable is set.

**Expected behavior**
To have the correct states logged to avoid any confusion in bug reports. Although they are logging incorrectly, I can see that the correct regression points are used for the current AC state. The correct keyboard backlight calibration setting is also used regardless of what the message says.

**To Reproduce**
Simply start clight and run `tail -f ${XDG_CONFIG_HOME:-$HOME/.local/share/}clight/clight.log` to view the log as you plug in or unplug your AC cable

Attached is a verbose log with me unplugging and plugging my AC cable in.
[clight.log](https://github.com/FedeDP/Clight/files/3688353/clight.log)

NICHOLAS85, 2019-10-03:
I realize I left the keyboard backlight calibration variable enabled in the verbose log, but it shows the same message regardless.

FedeDP, 2019-10-04:
Yes, this is super easy to fix(quite stupid bugs indeed!) I fill fix them asap, thanks for reporting!
I am currently moving my house, and I haven't got my pc with me yet, thus it can take some time to commit :)
As soon as my pc will be with me I wish to actually release clight and clightd 4.0 as last release was a year ago and huge work has been done on both of them, with lots of new features and bugfixes!

closed 2019-10-06 by FedeDP

NICHOLAS85, 2019-10-07:
@FedeDP Good luck with your move! I look forward to the 4.0 release, lots of great features come since the last one.

commit 357d958b (2019-10-06): Fixed a couple of small issues with log strings. Fixes #102