FedeDP/Clight #104: [FEATURE REQ] Send --verbose output to stdout

ro6, 2019-10-28:
**Is your feature request related to a problem? Please describe.**

I can't use `grep` on the `--verbose` output when trying to debug (eg `clight --verbose | grep gamma` produces no output)

**Describe the solution you'd like**

Use std I/O for `--verbose` output so I can `grep` it easily.

**Describe alternatives you've considered**

I tried redirecting `stderr`, but it doesn't seem to be going there either.

FedeDP, 2019-10-28:
Clight output is already sent to stdout. But you cannot grep a running process' output as far as i know.
You should grep its log file, eg:
`tail -f $XDG_DATA_HOME/clight/.clight.log | grep -i gamma`.

ro6, 2019-10-28:
Thanks, didn't realize that.

closed 2019-10-28 by ro6