FedeDP/Clight #105: [BUG] "Wrong number of no_smooth_dimmer_transition array elements."

ro6, 2019-10-28:
**Clight version the issue has been seen with:**
4.0
clight-git r517.21bd0ee-2

**Used distribution:**
5.3.5-arch1-1-ARCH

**Describe the bug**
```
clight --verbose
Wrong number of no_smooth_dimmer_transition array elements.
SCREEN forcefully disabled as Clightd was built without screen support.
Clightd found, version: 4.0.
...
```

Things seem to work otherwise, though I'm not sure yet if the `no_smooth_dimmer_transition` option is being used or not.

```
clight --verbose 
Wrong number of no_smooth_dimmer_transition array elements.
SCREEN forcefully disabled as Clightd was built without screen support.
Clightd found, version: 4.0.
Set timeout of 3s 0ns on fd 11.
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Failed to retrieve AC state; fallback to connected.
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
37.68 -122.44 loaded from cache file!
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
New location received: 37.68, -122.44.
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Set timeout of 0s 1ns on fd 23.
Emitting Location property
Normal transition to 6500 gamma temp started.
Next alarm due to: Mon Oct 28 17:44:00 2019
Set timeout of 31318s 0ns on fd 23.
AC curve: y = 0.080909 + 0.199152x + -0.010606x^2
BATT curve: y = 0.108881 + 0.137655x + -0.005711x^2
init_kbd_backlight(): The name org.freedesktop.UPower was not provided by any .service files
Keyboard backlight calibration unsupported.
Set timeout of 0s 1ns on fd 26.
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting Sunrise property
Average frames brightness: 0.761897.
Ambient brightness: 0.762 -> Backlight pct: 0.983.
Set timeout of 600s 0ns on fd 26.
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting Sunset property
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting DayTime property
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting Temp property
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting AmbientBr property
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
Emitting BlPct property
upower_check(): The name org.freedesktop.UPower was not provided by any .service files
```

FedeDP, 2019-10-29:
Hi!
Can you share your clight conf file (/etc/default/clight.conf) ?

FedeDP, 2019-11-24:
This should be fixed now that clight and clightd 4.0 have been released. Remember that conf file was changed and thus you should merge your local changes into new conf file!

closed 2019-11-24 by FedeDP

ro6, 2019-12-10:
@FedeDP Thanks for 4.0! Things are much better now. I didn't even need to tune the brightness curve from the new default config.