FedeDP/Clight #106: [BUG] Numerous issues

yochananmarqos, 2019-11-26:
**Clight version the issue has been seen with:**
> 4.0

**Used distribution:**
> Manjaro GNOME

**Describe the bug**
A few things:
* At the top of the log, it says:
```
(W)[10:32:14]{opts.c:194}	SCREEN forcefully disabled as Clightd was built without screen support.
```
Is there a missing Cmake flag in the PKGBUILD?

* When I start my laptop (Dell Inspiron 5575) on battery power, the `clight` process uses a lot of CPU and no log is generated. If I kill `clight`, restart the `clightd.service` and start `clight` again, everything is fine.

* When keyboard backlight calibration is on, the backlight does not come on at all in the dark. It kind of defeats the purpose of having a keyboard backlight.

* When dpms is enabled, the screen turns of while watching a full-screen video. I use SMPlayer.

**Expected behavior**
Self-explanatory.

**To Reproduce**
Normal use.

[clight.log](https://pastebin.com/5LB9KdLv)

FedeDP, 2019-11-26:
Hi! Thanks for reporting these issues!
> At the top of the log, it says:

Fixed in Clightd 4.0-2 (https://aur.archlinux.org/packages/clightd/). Thanks!

> When dpms is enabled, the screen turns of while watching a full-screen video. I use SMPlayer.

SMplayer uses mpv as backend. Unfortunately org.freedesktop.ScreenSaver support is broken in mpv at the moment. I have the same issue. I just right click on clight desktop file and select "Pause backlight calibration"; or you can use eg: vlc!
Hopefully it will be fixed soon. 

> When keyboard backlight calibration is on, the backlight does not come on at all in the dark. It kind of defeats the purpose of having a keyboard backlight.

This is weird! Can you report here output of: `busctl call org.freedesktop.UPower /org/freedesktop/UPower/KbdBacklight org.freedesktop.UPower.KbdBacklight GetMaxBrightness` ? Thanks!

> When I start my laptop (Dell Inspiron 5575) on battery power, the clight process uses a lot of CPU and no log is generated. If I kill clight, restart the clightd.service and start clight again, everything is fine. 

This is weird; i cannot reproduce it here. Your log seems fine.
Can you paste here a log from a clight fault run?

yochananmarqos, 2019-11-26:
> Fixed in Clightd 4.0-2 (https://aur.archlinux.org/packages/clightd/).
Aha! I thought there was a Cmake flag missing. 😄️ Thanks!

> output of: `busctl call org.freedesktop.UPower /org/freedesktop/UPower/KbdBacklight org.freedesktop.UPower.KbdBacklight GetMaxBrightness`
`i 2`

I re-enabled keyboard backlight calibration and I'll see if I can catch something abnormal in the log.

FedeDP, 2019-11-26:
Note how kbd backlight is set to (1.0 - backlight_level) * max_kbd_backlight (in your case 2): https://github.com/FedeDP/Clight/blob/master/src/modules/backlight.c#L268.
Moreover, it is rounded to nearest integer, thus it should be set to "2" on low ambient brightness, ie: your maximum!

Let me know about the other issue too! Thanks!

NICHOLAS85, 2019-11-26:
I’ve also experienced the issue with the clight process using a lot of cpu a couple times. I haven’t been able to pinpoint when this occurs or what causes it, I usually only notice once I feel my laptop warming up abnormally. Killing and starting the clight process fixes this. I’ll leave on the verbose setting and attempt to share the log the next time this occurs.

FedeDP, 2019-11-26:
Great, thanks for joining! I have been using clight devel for months and it didn't occur here, or I didn't notice...

NICHOLAS85, 2019-11-26:
Here’s a link to my current config file:
https://raw.githubusercontent.com/NICHOLAS85/dotfiles/4c9a2719a60ce691f4350fea735c10743830382e/.config/clight.conf

yochananmarqos, 2019-11-27:
Here's my [`clight.conf`](https://pastebin.com/qhb7zM06).

yochananmarqos, 2019-11-27:
I noticed a webcam error in my journal after rebooting:
```
Nov 26 20:44:36 inspirognome kernel: uvcvideo: Failed to query (SET_CUR) UVC control 10 on unit 2: -32 (exp. 2).
Nov 26 20:44:36 inspirognome kernel: uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
```

Only one error in the `clight.log`:
```
(D)[20:44:36]{bus.c:279}	get_screen_brightness(): Input/output error
```

FedeDP, 2019-11-27:
> Here’s a link to my current config file:

> Here's my clight.conf.

You both have dimmer dpms and gamma disabled. I will try with this config if i can reproduce the issue!

> I noticed a webcam error in my journal after rebooting:

That should not be an issue.

> Only one error in the clight.log:

This is interesting. It seems that XGetImage() fails here: https://github.com/FedeDP/Clightd/blob/master/src/modules/screen.c#L95.

yochananmarqos, 2019-11-28:
I can't reproduce the issue with the keyboard backlight now. Maybe it was a fluke or kernel 5.4 fixed it. It fixed not being able to resume from suspend, so you never know. ¯\\\_(ツ)_/¯

FedeDP, 2019-11-28:
Let's focus on high cpu bug then!

NICHOLAS85, 2019-11-30:
[clight.log](https://github.com/FedeDP/Clight/files/3906899/clight.log)
The high cpu bug has occurred again, attached is a log. Some info is I recently rebooted. Additionally, I built clightd with SCREEN support but this log states it was built without at the top. After restarting clight that message at the top dissapears.

FedeDP, 2019-11-30:
Mmmh I can't see anything wrong there. Is clightd.service enabled?

NICHOLAS85, 2019-11-30:
Yes, heres an output
```
$ systemctl status clightd
● clightd.service - Bus service to manage various screen related properties (gamma, dpms, backlight)
   Loaded: loaded (/lib/systemd/system/clightd.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-30 03:21:47 EST; 1h 58min ago
 Main PID: 12989 (clightd)
   Memory: 3.0M
   CGroup: /system.slice/clightd.service
           └─12989 /usr/lib/clightd/clightd

Nov 30 03:21:47 Nicks-Laptop systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Nov 30 03:21:47 Nicks-Laptop clightd[12989]: Registered Als sensor.
Nov 30 03:21:47 Nicks-Laptop clightd[12989]: (watch_displays_using_udev) Starting
Nov 30 03:21:47 Nicks-Laptop clightd[12989]: Registered Camera sensor.
Nov 30 03:21:47 Nicks-Laptop systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
```

FedeDP, 2019-11-30:
It seems disabled though:
`Loaded: loaded (/lib/systemd/system/clightd.service; **disabled**; vendor preset: enabled)`

Btw stil unable to reproduce it here unfortunately, even with your configs :(

yochananmarqos, 2019-11-30:
Here's a log where the keyboard backlight did not come on in a dark room and and it did not adjust the backlight at all.
[clight.log](https://del.dog/ymekikanex.txt)

FedeDP, 2019-11-30:
> (D)[20:03:08]{bus.c:279}	get_screen_brightness(): Input/output error
(D)[20:03:08]{timer.c:40}	Disarmed timerfd on fd 25.
(D)[20:03:09]{backlight.c:317}	Average frames brightness: 0.000000.
(I)[20:03:09]{backlight.c:252}	Ambient brightness: 0.000. Clogged capture detected.

It seems like Clightd is not working well here: SCREEN module is failing to retrieve screen brightness and webcam captures return 0.00000 ambient brightness.
Consequently, Clight treats 0.0000 value as clogged capture and keyboard backlight is not touched (neither is screen backlight).

I always though 0.0000 was an impossible number for ambient brightness...is your room *this* dark? I think Clightd is misbehaving here...

yochananmarqos, 2019-11-30:
Yes, it was very dark. I was watching videos at night. I just updated to kernel 5.4.1, I'll see if there's any change tonight.

yochananmarqos, 2019-12-01:
The log was the same tonight, but I caught it chewing on the CPU again. When it happens, it stops logging.

The `uvcvideo` errors in the journal have to have something to do with it as it wasn't happening before v4.0.

FedeDP, 2019-12-02:
> Yes, it was very dark. I was watching videos at night. I just updated to kernel 5.4.1, I'll see if there's any change tonight.

Still i think that 0.00000 is an unreachable value...the issue itself can be quickly fixed (i should just set shutter_threshold default value to < 0 when conf option is commented), i'd like to understand if it is a normal value (if you can reproduce the "0.00000" ambient brightness it would be great!). 
May be it is just you webcam's quirk :)

> The log was the same tonight, but I caught it chewing on the CPU again. When it happens, it stops logging.

Interesting! I was not able to reproduce it yet. 
Are you sure it is Clight and not Clightd that is using CPU? Did something happened before (eg: system was resumed from suspend...)?

> The uvcvideo errors in the journal have to have something to do with it as it wasn't happening before v4.0.

It is because Clightd is now always resetting webcam default values before capturing to avoid captures with different parameters!
It the webcam still when those errors appear in dmesg? Can you check with any application?

NICHOLAS85, 2019-12-02:
@FedeDP, sorry for the late response. It is true clightd was somehow disabled at some point but still functioning. I've re-enabled it for now and will continue to try to narrow down the environment causing clight to get stuck using high cpu. I can also confirm that the process is clight and not clightd which has this CPU problem. I believe I've noticed it happening most after wakes from hibernation, but not 100% of the time.

FedeDP, 2019-12-03:
>  I've re-enabled it for now and will continue to try to narrow down the environment causing clight to get stuck using high cpu.

Thank you very much!

> I can also confirm that the process is clight and not clightd which has this CPU problem.

Great information, thanks.

> I believe I've noticed it happening most after wakes from hibernation, but not 100% of the time.

That is unfortunate...my laptop does not support hibernation very well thus i would not be able to test it...

FedeDP, 2019-12-16:
Any news on this topic?
I am really willing to kill this bug! :)

yochananmarqos, 2019-12-16:
The high CPU issue hasn't resurfaced at all for me.

FedeDP, 2019-12-17:
That is great! But it still leaves us with some unresolved questions...

NICHOLAS85, 2019-12-24:
High CPU resurfaced once again, it had been quite a while since it last occured. I rebooted after updating many packages including my kernel and such. Noticed the high cpu shortly after the reboot.

Log attached: [clightlog.txt](https://github.com/FedeDP/Clight/files/3997416/clightlog.txt)

FedeDP, 2019-12-24:
Once more i can only see an unusual error about get_screen_brightness call.
Can you try to disable SCREEN module and see if issue does persist? 
Thank you and merry xmas!

NICHOLAS85, 2019-12-26:
I've set no_screen to true and I'll let you know if it occurs again. Merry Christmas!

FedeDP, 2019-12-26:
Great! Thank you very much for your efforts testing this bug!

Merry Xmas!!

FedeDP, 2020-01-07:
I recently pushed some fixes that hopefully will address this issue.
You can try to update clight-git and let me know if issue persists! Thanks!

Moreover,
> Here's a log where the keyboard backlight did not come on in a dark room and and it did not adjust the backlight at all.

Was fixed (now 0.0 is a correct and not discarded ambient brightness).

yochananmarqos, 2020-01-08:
@FedeDP Thanks, I'll try it out and let you know how it goes.

NICHOLAS85, 2020-01-08:
Happy New Year! Unfortunately, I have had the high cpu issue resurface after setting no_screen on the old build. Heres the new log: [clight.log](https://github.com/FedeDP/Clight/files/4036242/clight.log)
I will rebuild clight from source and reenable screen and see if the issue comes up again.

FedeDP, 2020-01-08:
That is unfortunate :(

> Heres the new log: clight.log

As always, it seems clean. 

>  I will rebuild clight from source and reenable screen and see if the issue comes up again.

Great, thank you! Happy new year!

yochananmarqos, 2020-01-09:
It's working great with the git packages, the fix worked.

FedeDP, 2020-01-09:
> It's working great with the git packages, the fix worked.

Never say never :D i still hold my fingers crossed!

FedeDP, 2020-01-20:
Any news on this issue? :) Is it still working fine?

NICHOLAS85, 2020-01-23:
So far no issues!

FedeDP, 2020-01-28:
I am gonna close this one :)
Reopen it if any issue arises once more!

NICHOLAS85, 2020-02-01:
Ah! I've got some unfortunate news, it's popped up again. It's been almost a month since it last happened. I believe I turned off verbose mode so the following is all I could save from the log before I restarted the program. 

<details>
<summary>tail -f ~/.local/share/clight/clight.log </summary>

```
(I)[19:48:03]{main.c:134}       Clightd found, version: 4.1.
(I)[19:48:03]{main.c:164}       '/home/nicholas/.local/share/clight/modules.d/inhibit_bl_mod' loaded.
(I)[19:48:03]{upower.c:24}      Initial AC state: connected.
(W)[19:48:03]{interface.c:246}  Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
(I)[19:48:03]{interface.c:247}  Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
(I)[19:48:07]{backlight.c:249}  Ambient brightness: 0.750 -> Backlight pct: 0.796.
(I)[20:03:09]{backlight.c:247}  Ambient brightness: 0.749 (-0.042 screen compensation) -> Backlight pct: 0.758.
(I)[20:18:10]{backlight.c:247}  Ambient brightness: 0.643 (-0.049 screen compensation) -> Backlight pct: 0.654.
(I)[20:33:10]{backlight.c:247}  Ambient brightness: 0.754 (-0.042 screen compensation) -> Backlight pct: 0.763.
(I)[20:48:11]{backlight.c:247}  Ambient brightness: 0.764 (-0.047 screen compensation) -> Backlight pct: 0.767.
```

</details>

FedeDP, 2020-02-01:
That is very unfortunate :(
Reopening!
Any more info? eg: uptime? It happened once more after a resume from hibernation?
Thank you!

FedeDP, 2020-02-02:
Wow it seems i finally manage to see this issue on my laptop! Right after a normal reboot.
I tried to attach with strace and this was the output:
{code}
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
epoll_wait(5, [{EPOLLIN|EPOLLHUP, {u32=1071955024, u64=94340028611664}}], 64, -1) = 1
^Cstrace: Process 726 detached
{code}
Ie: an infinite loop of epoll_wait... It seems like an fd is triggering an event that is not catched by clight. Going to further investigate!

FedeDP, 2020-02-03:
I just pushed a commit that adds a debug log in a couple of points that i think may lead to this bug, even if it would be really surprising if they would!

NICHOLAS85, 2020-02-03:
I've rebuilt clight, enabled verbose mode, and will be watching for this issue to occur again. Looks like we're getting closer to the core cause

FedeDP, 2020-02-03:
Great!

> Looks like we're getting closer to the core cause

Well, I assume the bug is either in BACKLIGHT, BUS, USERBUS or INTERFACE modules, as when running clight with other modules disabled, the bug was still present.
There is only one place in Clight where an event could theoretically go unread (thus looping generating the same event over and over), and that is when BACKLIGHT module gets "paused" by either an inhibition, or screen gets dimmed/dpmsed, or sensor goes missing.
In that case, it should gracefully stop polling deregistering its fd; i think the issue may lay in here. Ie: BACKLIGHT does not properly deregisters its fd; even if it sound unbelievable to me, that is my main target right now!
Let's hope the issue arises as quickly as possible!

FedeDP, 2020-02-04:
One more, most probable, idea:
While a message is waiting on BACKLIGHT queue (eg: DIMMER tells BACKLIGHT to pause as screen is dimmed). a timer event from its fd is triggered, and it is enqueued.
During next epoll call, we will receive dimmer message that tells BACKLIGHT to pause, causing it to deregister its fd and stop managing it; *but its timer event is still enqueued!* Thus it still generates events that nobody manages.

I have to understand whether this is a bug on clight or libmodule :)

FedeDP, 2020-04-14:
Has the bug resurfaced for anyone?

yochananmarqos, 2020-04-14:
"Not I", said the fly.

NICHOLAS85, 2020-04-14:
I have had this occur once. Unfortunately, it was while I was basically reinstalling my system and hadn't reconfigured Clight to use verbose logs and I didn't have time to note what was causing it. I remember I was compiling some software when I noticed.

FedeDP, 2020-04-14:
> I remember I was compiling some software when I noticed.

Very unfortunate; when did this happen? Ie: 2 months ago or a week ago? 
I never saw it occurring on my laptop again.

NICHOLAS85, 2020-04-15:
This happened in the last week.

finkrer, 2020-06-04:
What can I share to help resolve this?

FedeDP, 2020-06-04:
Unfortunately the bug is really hard to understand and consequently fix.
Right now clight git master is quite unstable; i will hopefully fix stability issues in the next days; then i will go back to this issue to fix it before 4.1 release.

It seems like a timed event is continuously arising and nobody manages it.
You can attach your verbose log file when it happens again, thank you!

FedeDP, 2020-06-06:
@finkrer Hi! In latest commits i tried to fix this issue; can you test with latest master?

Please note that i am not sure what is causing this issue; it is just a guess.
If the issue persists, it would be great if you could try to run clight with backlight module disabled to check if issue is caused by it.
Thank you!

finkrer, 2020-06-06:
I am now running the latest git version in verbose mode, if the issue occurs again I'll be sure to report it.

FedeDP, 2020-06-06:
Great! Thank you!

FedeDP, 2020-06-18:
Any news on this? :)

finkrer, 2020-06-18:
No news is good news in this case.

I didn't want to be too hasty in saying it's been fixed, but the issue hasn't reoccurred so far.
My laptop heats up a lot when it happens, so I would notice.

NICHOLAS85, 2020-06-18:
I have had this issue occur exactly once since the last time I reported experiencing a problem (over a month ago). Unfortunately, I was not in a position to collect any diagnostic data and forgot to keep tabs on what commit I was using. This is quite a rare issue. Is the log the only useful data I can provide when this happens?

FedeDP, 2020-06-19:
> This is quite a rare issue. Is the log the only useful data I can provide when this happens?

Unfortunately even log is quite useless...the only really useful output is strace's one (https://github.com/FedeDP/Clight/issues/106#issuecomment-581162209), but there are so many levels of possibilities that is practically impossible being sure to fix this.
Hopefully, latest clight-git has a proper fix (3963ffc33b9b54a1755e0f2fb696efd36163b791 commit), or at least it tries to fix what i guess can be our issue here.

FedeDP, 2020-06-19:
By the way thanks for your continuous testing!

[24 more replies]

closed 2021-02-25 by FedeDP