FedeDP/Clight #107: [BUG] clight: error while loading shared libraries: libmodule.so.5: cannot open shared object file: No such file or directory

AndyLuckert, 2019-12-05:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!
-> I've installed the latest version I suppose, as I did it following the instructions under the following link today (05-12-2019):
https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies

**Clight version the issue has been seen with:**
> When trying to obtain the version, it throws the same error, as when attempting to start the program in the first place:
```
(base) User:~$ clight --version
clight: error while loading shared libraries: libmodule.so.5: cannot open shared object file: No such file or directory
```

**Used distribution:**
> Linux Lubuntu 18.04

**Describe the bug**
When trying to start clight right after installation.
```
(base) User:~$ clight
clight: error while loading shared libraries: libmodule.so.5: cannot open shared object file: No such file or directory
```

**Expected behavior**
I expect the program "Clight" to start.

**To Reproduce**
Install Clight via the provided shell-script by user brianread108 under the following link of this GitHub:
https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies

Please attach a Clight verbose log too!
I have no log-file on my system yet, since I couldn't even start the application after installing it due to the above-mentioned error.

FedeDP, 2019-12-05:
Hi! Thanks for opening this issue!
Libmodule build was wrong in [that script](https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies).
I just fixed it, thank you very much! 

Please note that the script will install everything from git as it targets experienced users that are willing to help me test clight; if you don't mind being on -devel version, it is fine!
Otherwise it would be safer to install from latest release tags instead, by modifying the script to download and extract latest tarballs.

AndyLuckert, 2019-12-05:
Hey FedeDP,

thanks for the rapid response!
I have two follow-up questions on that:

1. Can I copy-past and use the script you've linked, since it is updated now?

2. As for using the "safer" installation, how would I exactly have to modify the script?
I assume you refer to the lines where you clone the git-repository content:
```
git clone https://github.com/FedeDP/Clight.git
git clone https://github.com/FedeDP/Clightd.git
git clone https://github.com/FedeDP/libmodule.git
git clone https://github.com/rockowitz/ddcutil.git
```

FedeDP, 2019-12-05:
1) yes, obviously!
2) I updated the script once more, to include code to build latest releases :)
Enjoy! https://github.com/FedeDP/Clight/wiki/Build#dealing-with-dependencies

FedeDP, 2019-12-10:
I'll close this issue, feel free to reopen it if you need more help!

closed 2019-12-10 by FedeDP