FedeDP/Clight #109: [BUG] Brightness is set to minimum after opening the lid on AC power.

shashurup, 2019-12-18:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.0

**Used distribution:**
> Arch linux

**Describe the bug**
My laptop is set up to not to go to sleep upon closing the lid on AC power. When I close the lid and open it after a while screen brightness is set to minimum. Clight log shows nothing at the time.

**Expected behavior**
I expect appropriate brightness correction by clight.

FedeDP, 2019-12-19:
Hi, thanks for opening the bug report!

> My laptop is set up to not to go to sleep upon closing the lid on AC power. When I close the lid and open it after a while screen brightness is set to minimum. Clight log shows nothing at the time.

May be clight does a capture while lid is closed and thus it sets minimum backlight?
You have to ways to solve this issue:
# Inhibit clight backlight autocalibration upon closing the lid and re-enabling it once lid gets opened
# As soon as lid is opened, do a quick backlight calibration

shashurup, 2019-12-19:
I'm going to check this version. However, as far as I remember, there was nothing in the log about capture took place.

FedeDP, 2020-01-05:
I forgot about one possible easy fix for a docked laptop (with lid closed): just set a small yet positive value for "shutter_threshold" conf value.

shashurup, 2020-01-11:
Yes, this seems to work. Thank you!

closed 2020-01-11 by shashurup