FedeDP/Clight #110: [FEATURE REQ] Use ambient sensors for auto backlight

mindrunner, 2019-12-21:
**Is your feature request related to a problem? Please describe.**
I have a computer with an ambient light sensor and would like to prefer using the sensor in favor of the webcam for the screen brightness control.

**Describe the solution you'd like**
It would be awesome to natively support those sensors instead of my glue-script furhter down. I think, this is important since most modern laptops should have such sensors.

**Describe alternatives you've considered**
There is a non-maintained script called autolight [1]. Which does the Job. I glued autolight together with clight, thanks to the awesome new module feature. See module code below.

```
#include <clight/public.h>
#include <stdlib.h>

CLIGHT_MODULE("INHIBIT_AUTOLIGHT");

static void init(void) {
    /* Subscribe to inhibit state */
    M_SUB(INHIBIT_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
        if (up->new) {
            INFO("We are now inhibited! Stopping autolight\n");
            system("pkill -SIGSTOP -f autolight");
            INFO("Setting backlight to 100.0\n");
            system("light -S 100");
        } else {
            INFO("We're not inhibited anymore. Resuming autolight.\n");
            system("pkill -SIGCONT -f autolight");
        }
        break;
      }
    default:
        break;
    }
}
```

[1] https://gitlab.com/craftyguy/autolight

FedeDP, 2019-12-22:
Hi!
Clightd should already support Als devices through Linux iio subsystem. Indeed it should even give higher priority to Als devices whenever found. 
If it is not the case, please fill a bug report on clightd! 
You can quickly test manually calling clightd capture method with busctl!

mindrunner, 2019-12-22:
Clightd clearly uses my camera for backlight control. How can I change the setting to use the sensor?

FedeDP, 2019-12-22:
Give https://github.com/FedeDP/Clightd/wiki/Sensors a read!
It should automatically use Als devices though, but you can force its usage with busctl by specifying /org/clightd/clightd/Sensor/Als object path.

mindrunner, 2019-12-22:
Does that mean I cannot configure the sensor in clight config file?

FedeDP, 2019-12-22:
Yes you can configure it through SensorName config.
SensorName is returned by clightd after a successful Capture call.
Normally, you should not need to configure it though as Als devices are checked before Camera devices.

mindrunner, 2019-12-22:
https://github.com/FedeDP/Clightd/issues/34

closed 2019-12-22 by mindrunner