FedeDP/Clight #111: [BUG]

bradbrad2x, 2020-01-12:
**Clight version the issue has been seen with:**
> 4.0

**Used distribution:**
> Arch Linux

**Describe the bug**
1. Auto back light only works with webcam and not als device.

2. When als device is not working, sometimes the screen will turn black and will be unable to turn back to normal without putting the computer to sleep and waking the computer. Moving the mouse does not do anything.

**Expected behavior**
Normal use of als device with auto backlight working properly.

**To Reproduce**
Comment/uncomment sensor_devname=video0

> [clight.log](https://pastebin.com/5WyeAdU3)

FedeDP, 2020-01-12:
Hi! Thanks for opening this issue.
I don't quite understand: is ALS capture working for you?

> clight.log

I can see from your log that a clogged capture is always detected, thus screen backlight is not changed. Your shutter threshold is
> Shutter threshold:        0.10

Are you in a very dark room?

bradbrad2x, 2020-01-12:
> I don't quite understand: is ALS capture working for you?

ALS capture does not work. It only works with the webcam.

> I can see from your log that a clogged capture is always detected, thus screen backlight is not changed. Your shutter threshold is
> 
> > Shutter threshold:        0.10
> 

Clogged capture only shows when I try to use ALS capture.

> Are you in a very dark room?

I was using the laptop with the lights on.

FedeDP, 2020-01-12:
Ok!

> ALS capture does not work. It only works with the webcam.

CAn you share output of:
`busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Als org.clightd.clightd.Sensor IsAvailable "s" ""`
and
`ls -ahl /sys/bus/iio/devices`
?

Btw Clightd is undergoing a full reworking for its als device support: https://github.com/FedeDP/Clightd/issues/34.

It should be working fine using clightd-git version; care to test?

bradbrad2x, 2020-01-12:
>`busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Als org.clightd.clightd.Sensor IsAvailable "s" ""`

`sb "/dev/iio:device0" true`

> `ls -ahl /sys/bus/iio/devices`

`total 0
drwxr-xr-x 2 root root 0 Jan 12 06:47 .
drwxr-xr-x 4 root root 0 Jan 12 06:47 ..
lrwxrwxrwx 1 root root 0 Jan 12 06:47 iio:device0 -> ../../../devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0`

> It should be working fine using clightd-git version; care to test?

I have switched to the clightd-git version and it still seems like the backlight still does not change.

FedeDP, 2020-01-12:
> I have switched to the clightd-git version and it still seems like the backlight does not change.

Did you reboot after?

What about `udevadm info -a /sys/bus/iio/devices/iio:device0` output?

bradbrad2x, 2020-01-12:
> Did you reboot after?
 Yeah I tried to reboot before checking to see if it works but no luck.
> What about `udevadm info -a /sys/bus/iio/devices/iio:device0` output?

`
Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0':
    KERNEL=="iio:device0"
    SUBSYSTEM=="iio"
    DRIVER==""
    ATTR{in_illuminance_raw}=="400"
    ATTR{name}=="acpi-als"
    ATTR{in_illuminance_input}=="400"

  looking at parent device '/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00':
    KERNELS=="ACPI0008:00"
    SUBSYSTEMS=="acpi"
    DRIVERS=="acpi_als"
    ATTRS{status}=="11"
    ATTRS{hid}=="ACPI0008"
    ATTRS{path}=="\_SB_.ALS_"

  looking at parent device '/devices/LNXSYSTM:00/LNXSYBUS:00':
    KERNELS=="LNXSYBUS:00"
    SUBSYSTEMS=="acpi"
    DRIVERS==""
    ATTRS{hid}=="LNXSYBUS"
    ATTRS{path}=="\_SB_"

  looking at parent device '/devices/LNXSYSTM:00':
    KERNELS=="LNXSYSTM:00"
    SUBSYSTEMS=="acpi"
    DRIVERS==""
    ATTRS{hid}=="LNXSYSTM"
    ATTRS{path}=="\"
`

FedeDP, 2020-01-12:
Ok, it seems fine.
So, what does `busctl call org.clightd.clightd /org/clightd/clightd/Sensor/Als org.clightd.clightd.Sensor Capture "sis" "" 5 ""` returns?

bradbrad2x, 2020-01-12:
`sis" "" 5 "" 
sad "/dev/iio:device0" 5 0.8 0.8 0.8 0.8 0.8
`

FedeDP, 2020-01-12:
That's great; it means clightd is actually working fine.
Can you share a new clight log with new clightd?

bradbrad2x, 2020-01-12:
[clight.log](https://pastebin.com/hJ0pjDMF)

FedeDP, 2020-01-12:
Isn't it working fine now?
> (I)[12:15:50]{backlight.c:247}  Ambient brightness: 0.800 (-0.058 screen compensation) -> Backlight pct: 0.906.

It seems like your Sensor is always reporting 0.8 (ie: 400 lux) value. In fact it reported the same in your above udev output:
>  ATTR{in_illuminance_input}=="400"

bradbrad2x, 2020-01-12:
> It seems like your Sensor is always reporting 0.8 (ie: 400 lux) value. In fact it reported the same in your above udev output:
> 
> > ATTR{in_illuminance_input}=="400"

The lux value doesn't seem to change even when I run clight in a dark room.

FedeDP, 2020-01-12:
Can you run `udevadm info -a /sys/bus/iio/devices/iio:device0` in 2 very different conditions and report output here?
Btw it seems like your sensor is reporting wrong values, ie:
>    ATTR{in_illuminance_input}=="400"

does not depend on clight; it is a sensor driver reading exposed by udev.

bradbrad2x, 2020-01-12:
With large amounts of light
`looking at device '/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0':
    KERNEL=="iio:device0"
    SUBSYSTEM=="iio"
    DRIVER==""
    ATTR{name}=="acpi-als"
    ATTR{in_illuminance_input}=="400"
    ATTR{in_illuminance_raw}=="400"
`

With low amount of light:

`  looking at device '/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0':
    KERNEL=="iio:device0"
    SUBSYSTEM=="iio"
    DRIVER==""
    ATTR{in_illuminance_input}=="400"
    ATTR{in_illuminance_raw}=="400"
    ATTR{name}=="acpi-als"
`

It outputs the same thing in both conditions.

FedeDP, 2020-01-12:
Unfortunately it seems like a bug in your ALS driver, or worse your ALS hardware is misbehaving :(

bradbrad2x, 2020-01-13:
I looked into it and it seems that the als device on my laptop is disabled by default. I just needed to make an ACPI call to enable the sensor `_SB.PCI1.LPCB.EC0.ALSC` and now it seems to work properly.

FedeDP, 2020-01-13:
Wow that's great, thanks! We can close this then :)
Added this information to Clightd wiki page: https://github.com/FedeDP/Clightd/wiki/Sensors#als-devices.

bradbrad2x, 2020-01-13:
Also to keep the als device enabled, I had to make a systemd service launch a script that does the acpi call at boot.

bradbrad2x, 2020-01-13:
Service:
>als-enable.service
>`[Unit]
>Description=Enables ALS device
>[Service]
>ExecStart=pathtoscript
>[Install]
>WantedBy=multi-user.target`

Script:
>als-enable
>`echo \\_SB.PCI0.LPCB.EC0.ALSC 0x1 > /proc/acpi/call`

closed 2020-01-14 by FedeDP