FedeDP/Clight #112: [FEATURE REQ] Granular keyboard backlight control

bradbrad2x, 2020-01-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. EG: I'm always frustrated when [...]

Sometimes my keyboard does not change the keyboard backlight when needed in my current day to day conditions.

**Describe the solution you'd like**

I want to be able to control when the keyboard backlight brightens and dims according to the ambient brightness. It would be similar to the backlight regression points already implemented.

FedeDP, 2020-01-16:
While this is quite easy to implement, usually keyboards have much less granular backlight control and thus it would be useless to have such feature.

Can you paste here output of:
`busctl call org.freedesktop.UPower /org/freedesktop/UPower/KbdBacklight org.freedesktop.UPower.KbdBacklight GetMaxBrightness`?

bradbrad2x, 2020-01-16:
> While this is quite easy to implement, usually keyboards have much less granular backlight control and thus it would be useless to have such feature.

Although keyboards usually have much less granular control, it would be nice to set when the keyboard backlight should turn on/off. For example, a scenario could be if I only want it to turn on if its really dark (<20% ambient).

> Can you paste here output of:
> `busctl call org.freedesktop.UPower /org/freedesktop/UPower/KbdBacklight org.freedesktop.UPower.KbdBacklight GetMaxBrightness`?

`i 3`

FedeDP, 2020-01-16:
> Although keyboards usually have much less granular control, it would be nice to set when the keyboard backlight should turn on/off. For example, a scenario could be if I only want it to turn on if its really dark (<20% ambient).

I see, seems a legit use case indeed! Thanks for your input :)

FedeDP, 2020-01-20:
What would you think about a threshold instead? Ie: if ambient brightness drops below the threshold, keyboard backlight will be managed just like now (ie: with a backlight level of `1.0 - ambient_br`.
On levels above this threshold, keyboard will be set to lowest value (ie: switched off).
It is much simpler and yet effective; having a 10-points curve for keyboards (that often have 3 backlight levels) seems unnecessarily complicated to me!

bradbrad2x, 2020-01-21:
> What would you think about a threshold instead? Ie: if ambient brightness drops below the threshold, keyboard backlight will be managed just like now (ie: with a backlight level of `1.0 - [...]

Yeah that would also work

FedeDP, 2020-04-27:
Sorry for disappearing for so long!
Anyway, I still think that a threshold is the way to go; it is very simple, with minimum complexity added and it still should do the trick.
I may wait for other user inputs on this topic!

FedeDP, 2020-04-30:
I've made available a threshold config for keyboard backlight; let me know if it works fine for you! Thank you very much for your testing!

closed 2020-05-05 by FedeDP