FedeDP/Clight #113: [FEATURE REQ] Easing out brightness based on time of day

Gert-dev, 2020-01-26:
**Is your feature request related to a problem? Please describe.**
Clight already supports a set of integers that map detected ambient lighting to screen brightness levels. I've been using Clight mostly comfortably for a few weeks now, but have noticed that, in the evening, my screen is usually a bit too bright, so I turned down the maximum brightness set by it. However, during the day, the screen is now a bit too dark.

**Describe the solution you'd like**
I'm wondering if it would be interesting to have the light to brightness map somehow correlate with the time of day, a bit like gamma.

I'm not sure if this is achievable or even desirable, as currently screen brightness mostly depends on the amount of light that surrounds you, be it from the sun, a light bulb, or something else, irrespective of the time of day.

Perhaps being able to configure some sort of "easing factor" for the evening - e.g. use the same intervals as during the day, but make it less bright with a configurable percentage, would do the trick. Maybe this factor could also vary depending on how late it is, a bit like gamma.

**Describe alternatives you've considered**
Maybe it's just me and I should just tweak the light to brightness map more and get used to it :smile: .

FedeDP, 2020-01-27:
Hi! Thanks for reporting this feature request!

> I'm not sure if this is achievable or even desirable, as currently screen brightness mostly depends on the amount of light that surrounds you, be it from the sun, a light bulb, or something else, irrespective of the time of day.

I agree with that. However, note that Clight actually supports up to 50 curve points to manage ambient-brightness to screen-backlight mapping. I think you should just be able to fine tune your current curve to perfectly match your desired screen backlight.
Eg: you can stretch lower part of curve by adding more points; something like:
`ac_backlight_regression_points = [ 0.0, 0.04, 0.07, 0.09, 0.12, 0.15, 0.19, 0.22, 0.26, 0.29, 0.33, 0.38, 0.45, 0.61, 0.74, 0.81, 0.88, 0.93, 0.97, 1.0 ];`

This way, on lower ambient brightness you should have much higher granularity and thus control.

Gert-dev, 2020-01-27:
Thanks for the information, I didn't know that.

I'll try to do some more customization. In any case having the brightness managed automatically is a large improvement already, so thanks for this project!

closed 2020-01-27 by Gert-dev

FedeDP, 2020-01-27:
You are very welcome :) thank you for using clight!