FedeDP/Clight #114: [FEATURE REQ]Could I stop clight at night?

realasking, 2020-02-26:
**Is your feature request related to a problem? Please describe.**
Yes. I don't know if it's a problem of my webcam. It's always too bright if using clight at night. I have configured clight.conf, but it would make too dark on daytime.

**Describe the solution you'd like**
Could you provide a group of settings to make users for choosing when clight worked and when not?
For example, if the 'latitude' and the 'longitude'  and a given light Y were set, auto adjustment of brightness would be turned off and was set to the given Y and can also be manully adjust at the time after sunset and before sunrise.   

**Describe alternatives you've considered**
No.You have done a great job. I love this software very much and haven't considered any other alternatives.

FedeDP, 2020-03-02:
Hi! Thanks for the very kind words :)

> It's always too bright if using clight at night.

This is weird: since clightd 4.0 any internal webcam setting is setted to default values before any capture, thus there should be no difference between them...

Anyway, you can disable automatic backlight whenever you want through [clight bus api](https://github.com/FedeDP/Clight/wiki/Api):

`busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" true`

Moreover, you can write a [custom module](https://github.com/FedeDP/Clight/wiki/Custom-Modules) to fit your own request :)
[Night-de-theme](https://github.com/FedeDP/Clight/wiki/Custom-Modules#night-de-theme) should be easily updated to fit your purpose:
```
#include <clight/public.h>

CLIGHT_MODULE("DAYTIME");

DECLARE_MSG(calib_req, NO_AUTOCALIB_REQ);

static void init(void) {
    /* Suscribe to daytime updates */
    M_SUB(DAYTIME_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case DAYTIME_UPD: {
        daytime_upd *up = (daytime_upd *)MSG_DATA();
        if (up->new == DAY) {
             calib_req.nocalib.new = false;
            INFO("We're now during the day!\n");
        } else {
             calib_req.nocalib.new = true;
            INFO("We're now during the night!\n");
        }
        M_PUB(&calib_req);
        break;
    }
    default:
        break;
    }
}
```

realasking, 2020-03-02:
Oh，it's awesome! The custom module is so cool. I don't know that before.
Thanks for your help. Now I have slightly modified the module and it works perfectly.

FedeDP, 2020-03-02:
Yes, I think they are great too! But I'm so bad at advertising features :)

realasking, 2020-03-03:
Oh, cmft~ This is a powerful tool for laptop users and there will be more and more people realized it.  I'll tell my friends about it, too. Don't be up set.

realasking, 2020-03-08:
Hi, I created a bash script for conky with this tool and your just click and paste your given command for adjusting the brightness. It's very comfortable to use. Thank you again ~
```bash
#!/bin/bash
#A Script for automatic brightness adjustment with conky
#Author: realasking
#March 08,2020
#Dependecies:
#            clight
#            clightd
#            sunrise
#            xbacklight
#            dbus
#            dig
#            curl

#Get IP address
IPADD=`curl http://members.3322.org/dyndns/getip 2>/dev/null`

#Validating IP address
#Method from : https://blog.csdn.net/JackLiu16/article/details/81709835
regex="\b(25[0-5]|2[0-4][0-9]|1[0-9][0-9]|[1-9][0-9]|[1-9])\.(25[0-5]|2[0-4][0-9]|1[0-9][0-9]|[1-9][0-9]|[0-9])\.(25[0-5]|2[0-4][0-9]|1[0-9][0-9]|[1-9][0-9]|[0-9])\.(25[0-5]|2[0-4][0-9]|1[0-9][0-9]|[1-9][0-9]|[1-9])\b"
CK=`echo $IPADDR | egrep $regex | wc -l`

#Get Latitude and Longitude
if [ $CK -eq 0 ];then
    #For Beijing
    latlon=" +39.9 +116.5"
else
    datain=`curl https://ipvigilante.com/${IPADD}`
    lat=`echo $bb|awk -F "latitude" '{print $2}'|cut -d"\"" -f3`
    lon=`echo $bb|awk -F "latitude" '{print $2}'|cut -d"\"" -f7`
    latlon=" ${lat} ${lon}"
fi

#Set light level
darknait="11"
don="14"

#Time Settings
a=`sun $latlon`
# Solve Sunrise time and Sunset time as minutes of a Day
hourise=`echo $a|cut -d":" -f1|awk '{print $3}'`
minrise=`echo $a|cut -d":" -f2|cut -d"," -f1`
Sunrise=`echo " ( $hourise * 60 ) + $minrise " |bc`
houset=`echo $a|cut -d"," -f2|awk '{print $2}'|cut -d":" -f1`
minset=`echo $a|cut -d"," -f2|awk '{print $2}'|cut -d":" -f2`
Sunset=`echo " ( $houset * 60 ) + $minset " |bc`

#Generate Adjusting times
Dawn=`echo " $Sunrise - 30 "|bc`
#Sunrise
#Sunset
DarkNight=`echo " ( $Sunset + 60 ) "|bc`

#get time
Current=`date +%H:%M`
houC=`echo $Current|cut -d":" -f1`
minC=`echo $Current|cut -d":" -f2`
Ctime=`echo " ( $houC * 60 ) + $minC " |bc`
echo $Ctime

#Adjust brightness
if [ $Ctime -lt $Sunrise ];then
    /usr/bin/busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b"  true
    if [ $Ctime -lt $Dawn ];then
	xbacklight -set "$darknait"
    else
	xbacklight -set "$don"
    fi
else
    if [ $Ctime -gt $Sunset ];then
	/usr/bin/busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b"  true
	if [ $Ctime -lt $DarkNight ];then
	    xbacklight -set "$don"
	else
	    xbacklight -set "$darknait"
	fi
    else
	/usr/bin/busctl --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b"  false
    fi
fi
```

FedeDP, 2020-03-09:
That's great! 

> This is a powerful tool for laptop users and there will be more and more people realized it. I'll tell my friends about it, too. Don't be up set.

:) Thank you once more for your very kind words!
I think we can close this one, can't we? 

Again, thanks for your time spent on clight!

closed 2020-04-09 by FedeDP