FedeDP/Clight #117: [SUPPORT] Runnig clight on Void Linux

linkert, 2020-04-09:
Sort of sorry to bother you with this but I would really like to get clight going on my system - seems almost too good to be true :) I'll stick to reddit if you are not comfortable with this kind of support.

First things first before I start rambling here are some information that might be relevant
```
@ ~: inxi -v 8
...
USB: Device-3: 1-1.6:5 info: Acer type: Video driver: uvcvideo interfaces: 2 rev: 2.0 speed: 480 Mb/s chip ID: 5986:02d2
...
Graphics:  Device-1: Intel 3rd Gen Core processor Graphics vendor: Lenovo driver: i915 v: kernel bus ID: 00:02.0
           chip ID: 8086:0166
           Display: server: X.Org 1.20.8 driver: modesetting alternate: fbdev,intel,vesa compositor: picom
           resolution: 1366x768~60Hz
           OpenGL: renderer: Mesa DRI Intel HD Graphics 4000 (IVB GT2) v: 4.2 Mesa 20.0.4 compat-v: 3.0 direct render: Yes
```

I run a Thinkpad X230 - Void Linux, which uses ```runit``` for its init service supervision. Clight is in the Void repos which is great but generally, not sure if it's the package maintainers or devs themself that include it, there is a runit service included even if there's no mention of support for other init systems other than systemd on the software projects git, website or man page. Clight is one of the packages that have no init service shipped with the package on Void.

Simply trying to run clight from my bspwm, alacritty X session:
 ```
@ ~: clight
Clightd found, version: 4.0.
BUS: Failed to connect to user bus
```

I asked on reddit, someone said - *"... to run dbus with X11 using something like dbus-run-session, then run clight from within the X11 session."*

So I check the man page for ```dbus-run-session``` it hurts my head and I go like this:
```
@ ~: dbus-run-session clight
Clightd found, version: 4.0.
Failed to init.
Failed to retrieve AC state; fallback to connected.
Next alarm due to: Thu Apr  9 19:15:00 2020
```

I guess it's a step forward. Any ideas on how to have it not fail init and retrieve AC state?

Here's the log in its entirety:
```
Clight
* Software version:		4.0
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Thu Apr  9 10:57:25 2020

Starting options:

### Generic ###
* Verbose (debugging):		Disabled
* Autocalibration:		Enabled
* Inhibit autocalibration:		Disabled
* Keyboard backlight:		Disabled
* Number of captures:		5
* Webcam device:		video0
* Backlight path:		intel_backlight
* Shutter threshold:		0.00
* Ambient gamma:		Disabled
* Screen contrib:		0.10
* Screen samples:		10
* User position:		59.29	18.05
* Daily screen temp:		6500
* Nightly screen temp:		4000
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900	BATT 300
* Screen timeouts:		AC 30	BATT -1

### Modules ###
* Backlight:		Enabled
* Gamma:		Enabled
* Dimmer:		Enabled
* Dpms:		Enabled
* Screen:		Enabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		ENTER: 30, EXIT: 30
* Gamma long transition:		Disabled

(I)[10:57:25]{main.c:134}	Clightd found, version: 4.0.
(W)[10:57:25]{upower.c:17}	Failed to init.
(I)[10:57:25]{upower.c:21}	Failed to retrieve AC state; fallback to connected.
(I)[10:57:25]{gamma.c:176}	Next alarm due to: Thu Apr  9 19:15:00 2020
```

FedeDP, 2020-04-09:
Hi! It's great to see such strong willing to use clight :) I feel honored!
You need UPower installed for upower module to work.
Btw from your latest log, it seems like clight has started fine; is that the only log you got? I can't see other modules doin' work there.

linkert, 2020-04-09:
Ah, nice!

Still starting through ```dbus-run-session clight```.

*GAMMA forcefully disabled as Clightd was built without gamma support.* - hm, is this actually true, meaning I have to build clight myself then?

```
(W)[15:01:54]{opts.c:189}	GAMMA forcefully disabled as Clightd was built without gamma support.
(W)[15:01:54]{opts.c:194}	SCREEN forcefully disabled as Clightd was built without screen support.

Clight
* Software version:		4.0
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Thu Apr  9 15:01:54 2020

Starting options:

### Generic ###
* Verbose (debugging):		Enabled
* Autocalibration:		Enabled
* Inhibit autocalibration:		Disabled
* Keyboard backlight:		Disabled
* Number of captures:		5
* Webcam device:		video0
* Backlight path:		intel_backlight
* Shutter threshold:		0.00
* Ambient gamma:		Enabled
* Screen contrib:		0.10
* Screen samples:		10
* User position:		59.29	18.05
* Daily screen temp:		6500
* Nightly screen temp:		3600
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800
* Dimmer backlight pct:		0.20

### Timeouts ###
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Dimmer timeouts:		AC 45	BATT 20
* Dpms timeouts:		AC 900	BATT 300
* Screen timeouts:		AC 30	BATT -1

### Modules ###
* Backlight:		Enabled
* Gamma:		Disabled
* Dimmer:		Enabled
* Dpms:		Enabled
* Screen:		Disabled

### Smooth ###
* Bright smooth trans:		Enabled
* Gamma smooth trans:		Enabled
* Dimmer smooth trans:		ENTER: Enabled, EXIT: Enabled
* Bright smooth steps:		0.05
* Gamma smooth steps:		50
* Dimmer smooth steps:		ENTER: 0.05, EXIT: 0.05
* Bright smooth timeout:		30
* Gamma smooth timeout:		300
* Dimmer smooth timeout:		ENTER: 30, EXIT: 30
* Gamma long transition:		Enabled

(I)[15:01:54]{main.c:134}	Clightd found, version: 4.0.
(I)[15:01:54]{upower.c:24}	Initial AC state: disconnected.
(D)[15:01:54]{my_math.c:61}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
(D)[15:01:54]{my_math.c:61}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
(D)[15:01:54]{bus.c:279}	init_kbd_backlight(): No such interface “org.freedesktop.UPower.KbdBacklight” on object at path /org/freedesktop/UPower/KbdBacklight
(D)[15:01:54]{backlight.c:229}	Keyboard backlight calibration unsupported.
(D)[15:01:54]{timer.c:38}	Set timeout of 0s 1ns on fd 27.
(D)[15:01:55]{bus.c:279}	capture_frames_brightness(): Operation not permitted
(D)[15:01:55]{timer.c:38}	Set timeout of 1200s 0ns on fd 27.
(D)[15:02:14]{display.c:35}	Entering dimmed state...
(D)[15:02:14]{display.c:74}	A lower than dimmer_pct backlight level is already set. Avoid changing it.
(D)[15:02:14]{interface.c:290}	Emitting DisplayState property
(D)[15:02:45]{display.c:50}	Leaving dimmed state...
(D)[15:02:45]{bus.c:279}	set_backlight_level(): Operation not permitted
(D)[15:02:45]{interface.c:290}	Emitting DisplayState property
```

FedeDP, 2020-04-09:
**It is working!!** 

> GAMMA forcefully disabled as Clightd was built without gamma support. 

It means that Clightd was built [without GAMMA support](https://github.com/FedeDP/Clightd/wiki/Build#build-time-switches).
But it seems like it was indeed built with GAMMA support (https://github.com/void-linux/void-packages/blob/master/srcpkgs/Clightd/template).
Thus, i think it is a bug in clight; it sometimes happen that Clight fails to parse Clightd response. 
As far as I know, it should be fixed in clight devel branch; you can most probably fix it by adding a small delay to clight start, eg: sleep 2 && ....

Is clightd.service equivalent for runit enabled though?

FedeDP, 2020-04-09:
Moreover:
> (D)[15:01:55]{bus.c:279}	capture_frames_brightness(): Operation not permitted

Which user is running clightd?

linkert, 2020-04-09:
> Is clightd.service equivalent for runit enabled though?

Nope as mentioned in my OP heh.. *"Clight is one of the packages that have no init service shipped with the package on Void."*.

But I checked out the package template file as well and it mentions some *i2c_clightd* - I ran it by cat and it simply instructs to put it in /etc/modules-load.d to load i2c-dv at boot - is this it maybe? I noticed word "i2c" swooping by during boot, I should probably do as instructed right?

```
@ ~: cat /usr/lib/modules-load.d/i2c_clightd.conf
# Load i2c-dev at boot; put this in /etc/modules-load.d
i2c-dev
```

> Which user is running clightd?

Just my regular user. I added a line by all the other startups in my bspwmrc ``` dbus-run-session clight```.

FedeDP, 2020-04-09:
> I did copy the i2c config file and directory to /etc/ rebooted and it was a kernel module that probably does something important? Same deal though, operation not permitted and GAMMA/SCREEN disabled.

The i2c kernel module is needed to support external monitor through ddcutil.

Clight does not ship any systemd service; i am talking about clight**d**! 
Clight**d** need to run as root as it is a system dbus service (just like geoclue or upower, for example); Clight does not need to run as root.

ahesford, 2020-04-09:
The Void Linux clightd package installs `/usr/share/dbus-1/system-services/org.clightd.clightd.service`, so `dbus` will launch `clightd` on demand when `clight` is started. However, there does not seem to be a timeout option that will allow `clightd` to die when it is no longer being used. This means that the auto-starting daemon effectively runs forever once it spawns.

Using the system service manager (in this case, `runit`) to manage the liftetime of `clightd` seems like a better option. I'll submit a PR with the Void package to add a `runit` service for `clightd`, making `clight` a little better behaved for Void.

linkert, 2020-04-10:
Thank you both for your patience and contributions, highly appreciated - looking forward to your PR and the runit service - that's something I'm more comfortable with :)

closed 2020-04-10 by linkert

FedeDP, 2020-04-10:
>  I'll submit a PR with the Void package to add a runit service for clightd, making clight a little better behaved for Void.

I will gladly ship it with next clight**d** version if you are willing to find any way to check from CMakeLists whether runit is installed.
I think possibily something like: https://github.com/FedeDP/Clightd/blob/master/CMakeLists.txt#L102 should do the trick.

ahesford, 2020-04-10:
> I will gladly ship it with next clight**d** version if you are willing to find any way to check from CMakeLists whether runit is installed.
> I think possibily something like: [...]

I'm not sure this is something that should be folded into Clightd; the Void package can handle the service. Void's `xbps-src` is well suited to installing these services for packaging. If you want to see how it's done, check [Void PR #20814](https://github.com/void-linux/void-packages/pull/20814)

FedeDP, 2020-04-10:
I see, thanks for your work anyway!