FedeDP/Clight #120: [BUG] Not leaving DPMS state once entered

ferki, 2020-04-24:
**Clight version the issue has been seen with:**
4.0

**Used distribution:**
Gentoo

**Describe the bug**
When clight enters DPMS state, the ambient brightness captures stop as expected/documented. But then it doesn't come back from DPMS state, essentially disabling backlight changes (both automatic and manual) until clight is restarted.

Disabling the DPMS module by uncommenting `no_dpms = true;` in the config avoids this problem as a workaround.

**Expected behavior**
I'd expect clight to be able to leave DPMS state, and resume automatic brightness calibration (perhaps even do an out-of-schedule capture right after leaving DPMS state).

**To Reproduce**
1. Start clight
2. See regular captures happen
3. Wait until it enters DPMS state

Verbose log:
[clight.log](https://github.com/FedeDP/Clight/files/4526314/clight.log)

ps.: I edited out location, next alarm time and its timeout.

FedeDP, 2020-04-24:
Hi, thanks for opening this issue.
Can you test and eventually reproduce it on devel branch? It has received lots of improvements and bugfixes.
Thank you!

ferki, 2020-04-24:
It seems to work correctly in the `devel` branch, thanks! :+1: Are there any plans for a new release already?

FedeDP, 2020-04-24:
Great!
Well, i had some plans but then life happened :)
I have still some small fixes in Clight and Clightd; unfortunately i am currently very busy working on libmodule on my spare time, thus i am not giving Clight/Clightd much attention.
I will hopefully switch context and get back working on them soonish (i'd say a couple of weeks at least).

ferki, 2020-04-24:
Alright, no worries! I just asked so I can better decide whether I should simply wait for the next release, or go ahead and package the git version ;)

Thanks for clight! :+1:

FedeDP, 2020-04-24:
Thank you very much for taking the time to package clight!!!

ferki, 2020-04-25:
Hmm, I seem to have this issue again with the `devel` branch.

I'm afraid my initial hypothesis about this might have been wrong, and it might not necessarily related to DPMS, after all. And that possibly made my testing strategy flawed too.

This morning I again had the following in at the end of the logs:

```
(D)[07:00:36]{validations.c:132}        Failed to validate display state request.
(D)[07:36:20]{validations.c:132}        Failed to validate display state request.
```

clight process is still running, periodic calibration is not happening, and manual call to calibration also doesn't work. After killing the process and calibrating/starting again, it seems to work.

@FedeDP: what would you prefer? Reopening this issue, or posting `devel` logs on a new issue as soon as I have a good one to share?

FedeDP, 2020-04-25:
Let's reopen this issue and please post new logs; 
> (D)[07:00:36]{validations.c:132}        Failed to validate display state request.
> (D)[07:36:20]{validations.c:132}        Failed to validate display state request.

This is weird and probably it is the root cause of the issue; clight internal display state is not properly resetted to "on", thus backlight module remains paused.

ferki, 2020-04-27:
So, I managed to capture two logs:
[clight_freeze_dpms.log](https://github.com/FedeDP/Clight/files/4541792/clight_freeze_dpms.log)
[clight_freeze_dpms_2.log](https://github.com/FedeDP/Clight/files/4541793/clight_freeze_dpms_2.log)

The pattern seems to be that after

```
(D)[14:45:10]{display.c:40}     Entering dpms state...
(D)[14:45:10]{interface.c:371}  Emitting DisplayState property
```

it can't do more measurements, and later (usually much later) it writes:

```
(D)[12:24:11]{validations.c:132}        Failed to validate display state request.
```

When the measurement are not happening anymore, but clight is still running, the `DisplayState` is the following:

```
$ busctl --user get-property org.clight.clight /org/clight/clight org.clight.clight DisplayState
i 2
```

I also ran clight with the DPMS module disabled, and it was working fine for ~40 hours at least. I hope all this new information helps to narrow it down.

FedeDP, 2020-04-27:
Ok, i just pushed a small change to fix this :)
Thanks for your testing!

FedeDP, 2020-04-27:
PS: obviously feel free to reopen if you think it is not fixed!

closed 2020-04-27 by FedeDP

ferki, 2020-04-28:
@FedeDP: it looks like working as expected now, thanks! It survived the the whole day with this patch :+1:

commit 5bb1f900 (2020-04-27): Make sure to reset DISPLAY_ON state from DPMS module when dimmer is disabled. Fixes #120.