FedeDP/Clight #121: [BUG] disabled module still enforces configuration options

ferki, 2020-04-24:
**Clight version the issue has been seen with:**
`devel` branch, commit id 1a6b6fd2d5a5d044289c4a3c8f5702ed25f2b0bf

**Used distribution:**
Gentoo

**Describe the bug**
While testing `devel` branch related to #120, I noticed that setting a shorter DPMS timeout than the dimmer timeout causes clight to say

> DPMS AC timeout: wrong value (<= dimmer timeout). Resetting default value.

 _even with the dimmer disabled_.

**Expected behavior**
I expected a disabled module wouldn't have any effect on the enabled modules' configuration (nice to have a message about that, though :+1: ).

**To Reproduce**
1. build clight from commit 1a6b6fd2d5a5d044289c4a3c8f5702ed25f2b0bf
1. edit devel config to
    - disable dimmer with uncommenting `disabled = true` in the dimmer section
    - set DPMS timeouts to `timeouts = [ 30, 30 ];` in the DPMS section (note that this is shorter than the default dimmer `timeouts = [ 45, 20 ];`)
 1. start clight with that config file and read the printed output

Verbose log:
[clight.log](https://github.com/FedeDP/Clight/files/4528392/clight.log)

FedeDP, 2020-04-24:
That is a tiny nice bug you caught here!
As it is super-easy to fix, it would be great if you could send a PR (if you feel obviously), otherwise i will look into it as soon as possible!
If you care to try, take a look at https://github.com/FedeDP/Clight/blob/devel/src/conf/opts.c#L380 :)

ferki, 2020-04-24:
Thanks for the pointer! If I go and hack on it myself, I'll try to check the same for other modules too, and I'll definitely share the patch in a PR :)

FedeDP, 2020-04-25:
Fixed by PR #122 , thank you very much!

PR #122 (2020-04-25): Don't reset DPMS settings if DIMMER is disabled
This PR is a quick attempt to fix #121. It makes sure that a disabled DIMMER module does not have a side effect on the settings of the DPMS module.

@FedeDP: hope it's a good enough fix like that, let me know if you had something else in mind!
commit 513dec9f (2020-04-25): Don't reset DPMS settings if DIMMER is disabled (fix #121)
closed 2020-04-25 by FedeDP