FedeDP/Clight #125: [BUG] Dimmer doesn't account for manually set brightness

NICHOLAS85, 2020-05-04:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Latest git commit as of now (04d9ebf2af990bb9c27f43c9eb0c70c09c4d227a)

**Used distribution:**
> Debian Testing

**Describe the bug**
When Clight runs and sets the brightness to a level below `dimmer_pct`, if I've since turned my brightness up manually due to new conditions and let my computer idle, dimmer will start but won't dim with the log message `A lower than dimmer_pct backlight level is already set. Avoid changing it.`. After this point, if I wake the computer out of dim, Clight will 'restore' the previous brightness level, which was the original level below `dimmer_pct`. This results in the screen getting darker when it's not supposed to.

**Expected behavior**
When dimming begins, it should use the current actual brightness in its logic, instead of the brightness last set by Clight. And when exiting it should restore to the actual brightness level set prior to dimming instead of the last brightness set by Clight.

**To Reproduce**
`tail -f ~/.local/share/clight/clight.log` to view when the dimmer starts and stops.
Enable dimmer and shorten its timer to wait less time. Cover your sensor and make Clight scan, setting the brightness to a level below `dimmer_pct`. Turn your brightness up and then let the computer idle until the dimmer kicks in. The brightness should not change but you can see in the log that the dimmer has started and hasn't changed the brightness with the message `A lower than dimmer_pct backlight level is already set. Avoid changing it.`. Now wake the computer and see the screen dim as the dimmer exits, restoring the original clight scan brightness.

Please attach a Clight verbose log too!

FedeDP, 2020-05-04:
Yes, Clight does not play well with external backlight changes. 
That is something that i planned to improve for next Clight release (4.2); but as it is bugging me sometimes too, i may have a look for 4.1 release.

FedeDP, 2020-05-05:
Unfortunately there is no easy solution; one thing i came up with is to let Clight handle your backlight increment/decrement, may be adding 2 new dbus methods; this has some PROs:
* Clight knows that backlight changes (obviously as it is now its job!)
* Clight can change backlight on every connected monitor
* Clight will change backlight following your smooth options

But at least one CONS:
* You cannot bind it to laptop fn key; at least i guess you cannot; i'll be happy if you can prove me wrong though :)

NICHOLAS85, 2020-05-05:
Thanks for looking into this issue! I am able to bind commands/scripts to my brightness keys so it may vary from setup to setup. I don’t want to push to get that functionality if it is beyond Clight’s scope, if you believe that to be a useful feature for Clight to have go for it. I currently use [Brillo](https://gitlab.com/cameronnemo/brillo) to control my brightness but if I’m able to replicate my increase/decrease behavior with Clight I’ll switch. They also implement a get brightness function which I assume is distro agnostic so it may be useful to look at how they do it. 

I was wondering if there might be a way to fix this using a custom module but I don’t think the modules can read the output of a `system();` call. You see any other ways of using modules to our advantage?

FedeDP, 2020-05-05:
> They also implement a get brightness function which I assume is distro agnostic so it may be useful to look at how they do it.

Clightd already has backlight getter methods; but my main concern here is:
1) What happens if a monitor is connected while dimmed? You have called getBrightness on monitors present before dimming
2) What happens if user manually changed backlight on just one monitor? (i don't know if Brillo supports external monitors)

Having Clight handling this for us would be much better even if i don't really like "forcing" users at using Clight; but i guess you can see Clight as a backlight manager tool thus it makes no sense that manual backlight changes are carried by a different software.

FedeDP, 2020-05-05:
> I don’t want to push to get that functionality if it is beyond Clight’s scope, if you believe that to be a useful feature for Clight to have go for it.

I think we should define Clight scope here; for me, it is a backlight manager that has other tools built in (GAMMA and DPMS for example), thus i can see it is ok to implement such a behaviour.

But i will check if udev allows to watch for backlight changes (that would make Clight fully interoperable).

NICHOLAS85, 2020-05-06:
>1. What happens if a monitor is connected while dimmed? You have called getBrightness on monitors present before dimming
>2. What happens if user manually changed backlight on just one monitor? (i don't know if Brillo supports external monitors)

That's true, I hadn't thought about multi-monitor setups. To provide my input on those problems:
1. I'd assume connecting a monitor would wake the computer from it's dimmed state and thus monitors present would restore their non-dimmed brightness and the new monitor would display at it's configured brightness to be managed by Clight from then on.
2. I think a solution would involve calling getBrightness on every connected device and allowing Clight to adjust each independently. ie, if a monitor is below it's dimmed_pct and doesn't need to be dimmed it isn't, but a bright laptop screen could still be dimmed. I assume this suggestion starts going into functionality coming with your `BACKLIGHT multiple-monitors curves` TODO for 4.2, more specifically the per-monitor support and configuration.

NICHOLAS85, 2020-05-07:
> ...After this point, if I wake the computer out of dim, Clight will 'restore' the previous brightness level, which was the original level below dimmer_pct. This results in the screen getting darker when it's not supposed to.

I think a simple solution to this problem would be to not 'restore' the brightness after waking from dim if `lower than dimmer_pct backlight level is already set.` was true prior to dimming.

NICHOLAS85, 2020-05-07:
I created a quick fix for my original issue here: https://github.com/FedeDP/Clight/compare/master...NICHOLAS85:dimfix

Basically skip restoring brightness if it wasn't touched while dimming. If that's adequate and you see no issues I can make a pull request.

FedeDP, 2020-05-07:
While it is a correct idea and an improvement (avoid a dbus call is always good), it would fix your own use-case but not the every use case/issue.
Anyway, you can make a PR and i'll review it, thank you!
I will probably introduce 2 new dbus methods to change backlight level by a step for now; it is the only way to let Clight really know about current backlight levels; then, in future versions i will surely add per-screen curves support and will improve dimmer behaviour on multi-screen.

FedeDP, 2020-05-07:
In 15a06181b0f7696b6433c9bf912597584d607366 I added IncBl and DecBl dbus methods (https://github.com/FedeDP/Clight/wiki/Api#bus-methods).

NICHOLAS85, 2020-05-07:
> While it is a correct idea and an improvement (avoid a dbus call is always good), it would fix your own use-case but not the every use case/issue.

This is true, and as you said in the future I'm sure a proper solution will come about when per-screen support and configuration are added. I'll send a PR with the fix.

NICHOLAS85, 2020-05-22:
Figured out a temporary solution for my case that I'd like to share here:
https://gist.github.com/NICHOLAS85/6bc25695d6fb1f14896aa4f4071e6580

This module basically saves your current bl when entering dimmed state and restores it after leaving dimmed state. The saved bl is actually a bit lower than what is intended as it saves you bl percentage while you're starting to dim.

That brings up a feature idea, an ability to specify if your module runs before, in parallel, or after, other modules (internal or custom). Unsure if that'd be useful or not.

FedeDP, 2020-05-24:
Great, it seems fine!

> That brings up a feature idea, an ability to specify if your module runs before, in parallel, or after, other modules (internal or custom). Unsure if that'd be useful or not.

That is a feature that should end up in libmodule; it is a nice idea though; i will probably experiment around it! As always thank you for your support :)

FedeDP, 2020-06-15:
I consider this issue fixed through IncBl and DecBl dbus methods exposed by Clight 4.1.
I may add a "BacklightChanged" signal in Clightd to better support external tools, at least on laptops (on laptops it is quite easy; while i think it is not possible for external monitors).

closed 2020-06-15 by FedeDP

FedeDP, 2020-06-18:
Moreover, latest clight [commit](https://github.com/FedeDP/Clight/commit/767e18434298fe6a579a1e7f94fdc93f0b47ed66) adds support for external backlight changes on laptops; it requires clightd 4.2 :)

NICHOLAS85, 2020-06-18:
That's great! That resolves my original issue as I mainly use Clight on my laptop. Thanks for implementing it!