FedeDP/Clight #126: [FEATURE REQ] Create backlight curve regression over time

NICHOLAS85, 2020-05-06:
**Is your feature request related to a problem? Please describe.**
Clight comes with default ac/bat curves which can be manually adjusted easily. Sometimes it can be inconvenient to constantly change this curve as you notice it being off of ideal at certain times.

**Describe the solution you'd like**
Add a learning feature to Clight. Simply enable a config option and Clight starts saving your BlPct and AmbientBr on each scan. The users current brightness will be saved with the current ambient brightness as well as if on ac or battery. After a week or so of collecting data, a curve can be formed and a new set of regression points can be suggested to use in your conf file.

**Describe alternatives you've considered**
I decided to share this feature request to get your input on the idea. I was actually thinking of creating a script that takes advantage of Clight's Bus API to get the your current Ambient Brightness and current backlight brightness and recreate the request stated above.

However I found that there isn't a way of running Calibrate without having it adjust the screen brightness and that the BlPct returned is not the real value but what Clight last set it as.

FedeDP, 2020-05-07:
I always thought about a clight "teaching" mode, that when started will just ask user to try different ambient brightness (eg: outside, bright room, normal room, dark room, night) and set a desired backlight; then finally present curve points.
What would you think about this?

> the BlPct returned is not the real value but what Clight last set it as.

You can read latest ambient brightness from org.clight.clight.AmbientBr!

NICHOLAS85, 2020-05-07:
That sounds like a useful feature! My idea was for passive learning over time while yours is more of an active initial approach/setup but it sounds like it'd work well.

I'll still look into creating some sort of script to have passive learning just to learn more about Clights API, it seems really flexible.

>You can read latest ambient brightness from org.clight.clight.AmbientBr!

Ah yes, that's useful. My issue was I wanted to create a script that could be as passive as possible. So I'd like to capture the user's current set brightness and grab the current ambient brightness at the same time as well as ac/bat data, at certain intervals. With how `org.clight.clight.AmbientBr` works it only gives me the last captured value, which may not equal what it is now. So if the ambiance has changed and a user has adjusted their brightness for the current one, I'd be saving a brightness value for the wrong ambiance. To work around this I could trigger a Clight capture, but this is not passive and actually adjusts the screen brightness like it's meant to.

FedeDP, 2020-05-08:
In a419de952e9d4576d65b86a3784940873dc9b867 i added an initial very simple implementation for a CLI wizard for clight.
It is very simple but it should work fine.

> To work around this I could trigger a Clight capture, but this is not passive and actually adjusts the screen brightness like it's meant to.

Following your user request (and because I needed it too for Wizard module) "Calibrate" dbus method has been renamed to "Capture" and it now takes 2 params that should fullfill your request, see here: https://github.com/FedeDP/Clight/wiki/Api#bus-methods.

Btw i will very gladly add your script to wiki if you don't mind to share it :)

NICHOLAS85, 2020-05-08:
Cool! I'll check it out and provide some feedback when I get the time. That capture feature modification is useful as well thanks for implementing it.

Thinking about this more I just realize I could create a module which subscribes to `AMBIENT_BR_UPD` and outputs data from there. From that point, all I'd need to do is write a script which does the regression and curve point presentation on that data, instead of creating something that runs 24/7.

I've looked into creating the custom module and found I can use `bl_upd` to get the old backlight value. Is there a way to acquire the new ambient brightness via a module as well?

NICHOLAS85, 2020-05-09:
I figured out how to get everything I need so far except an initial ac/bat state when the module loads (I figured out how to get it when the state changes) and the current backlight level pre-adjustment. I'm looking forward to sharing this once it's complete!

FedeDP, 2020-05-10:
That's great!
Well, that is an open issue (in my mind) that i planned to improve; i will think about a solution; unfortunately right now only state changes are published, while initial state is not.
I will let you know as soon as i fix this situation!

NICHOLAS85, 2020-05-11:
Alright, I've got a first draft of the module which will be used in conjunction with the script I've yet to write to analyze the saved data. This is actually my first time writing in `C` which may or may not be apparent in the code. If you have any suggestions/fixes I'll be happy to incorporate them!
https://gist.github.com/NICHOLAS85/00dc1303abbfbd4ccb483be38366b42a

On another note, are custom modules supposed to have access to the `DEBUG` macro? It doesn't seem to function correctly:
```
trendlog.c: In function ‘receive’:
trendlog.c:48:1: error: ‘conf’ undeclared (first use in this function)
   48 | DEBUG("Test");
      | ^~~~~
trendlog.c:48:1: note: each undeclared identifier is reported only once for each function it appears in
```

FedeDP, 2020-05-12:
> On another note, are custom modules supposed to have access to the DEBUG macro? It doesn't seem to function correctly:

Yes, nice catch thanks! I will fix it asap!

>  If you have any suggestions/fixes I'll be happy to incorporate them!

Well, just stylish but i'd use "static" on global variables; everything else seems ok! it is a great example of custom module!
Note that you use new Clight "IncBl/DecBl" to change your screen backlight, you will receive a BL_UPD message with newly set backlight. You may use that instead of calling Clightd directly.

Again, i will fix initial state publishing as soon as possible; i'll have to change the internal architecture for some modules, nothing huge but surely not a super-simple work.

Finally, i plan to add a BUS_REQ message in the future, to let Clight do dbus calls. Most probably 4.2. release cycle.

FedeDP, 2020-05-12:
> Yes, nice catch thanks! I will fix it asap!

Fixed in f19fac9da73f831407616c7af57c27358685753c

NICHOLAS85, 2020-05-12:
Thanks for the suggestions!
> Note that you use new Clight "IncBl/DecBl" to change your screen backlight, you will receive a BL_UPD message with newly set backlight. You may use that instead of calling Clightd directly.

I'll probably make this variable to let a user chose to use clightd if they don't manage their brightness with clight.`DEBUG();` also now works properly thanks!

FedeDP, 2020-05-14:
> Again, i will fix initial state publishing as soon as possible; i'll have to change the internal architecture for some modules, nothing huge but surely not a super-simple work.

Architectural change got into: 9a085dc376806e2e5e4b015d64643b4e5caedccd
As you can see, it is a quite huge update; let me know if it work flawlessly for you! In my tests, it worked fine.
Obviously, initial state messages are now published just like normal messages.

Moreover, i splitted DAYTIME module from GAMMA :)

NICHOLAS85, 2020-05-16:
Great! It’ll be awhile before I can test it myself but those are welcome changes.
<—off topic—>
Clight is honestly a great piece of software I’m surprised I don’t see it talked about more often. Looking into `libmodule` it seems like a great way to create modular projects, I might take up trying to convert https://github.com/NICHOLAS85/gebaar-libinput (my fork of https://github.com/Coffee2CodeNL/gebaar-libinput) into a libmodule using project, I think it could greatly benefit from it and it’d give me an opportunity to learn C/C++.

FedeDP, 2020-05-16:
> Great! It’ll be awhile before I can test it myself but those are welcome changes.

Let me know, thank you!

> Clight is honestly a great piece of software I’m surprised I don’t see it talked about more often. 

Please let me thank you from my heart for your very kind words!
Unfortunately I am not very good (neither i have the time) at PRs :)

To be honest, i am really pleased with current state: i have some great users that are really involved in my projects and give great feedback; indeed i think Clight has a more-than-fine audience right now.

I fear i'd struggle to keep up with these side projects if there was much more user involvement, leading to loss in my own satisfaction.
Obviously, seeing the "stars" counter increasing is very pleasing, and most probably that is just a stupid fear.

> Looking into libmodule it seems like a great way to create modular projects, I might take up trying to convert https://github.com/NICHOLAS85/gebaar-libinput (my fork of https://github.com/Coffee2CodeNL/gebaar-libinput) into a libmodule using project

That is very nice! Please note that libmodule is undergoing a huge list of improvements (and unfortunately API breaks) leading to libmodule 6.0.0 in its [deve](https://github.com/FedeDP/libmodule/tree/devel) branch.
I'd suggest to wait a couple of months (at least that would be my designed deadline for the release) or be ready to deal with some API breaks (nothing tremendous but still...) :)

Again, thank you very much for your involvement with Clight and for your incredibly sweet words!!

NICHOLAS85, 2020-05-21:
It's good to hear you're in a good place with your project! I'll delay looking into libmodule until 6.0.0 is pushed, thanks for the heads up.

I just updated and installed the latest version(7bd1b601257b54d3d6ffb8c6424c3557fac8cef3) and seem to be having issues. Backlight calibration seems to no longer be working. It doesn't seem to be turning my webcam on and no messages are appearing in my logs. The same thing is happening with gamma. No changes I send using dbus are being applied and no messages are appearing in my log relating to gamma module. I'm not sure if anything other modules are broken, heres my log:
[clight.log.txt](https://github.com/FedeDP/Clight/files/4660662/clight.log.txt)
Does it seem as if these modules failed to load correctly? I'm unsure as to how I can determine this to be the case.

FedeDP, 2020-05-21:
Nice catch thank you! It will be fixed soon :)
It is caused by your fixed sunrise/sunset times (and by fixed location too). I did not properly manage that case, sorry for the rather stupid bug!

FedeDP, 2020-05-21:
It should be now fixed in master! Thank you!

NICHOLAS85, 2020-05-21:
It looks like its working now, thanks!

FedeDP, 2020-06-24:
> I'd suggest to wait a couple of months (at least that would be my designed deadline for the release) or be ready to deal with some API breaks (nothing tremendous but still...) :)

Unfortunately it will probably take indefinitely more time...
Btw do you think we may consider this issue as resolved through wizard mode; moreover, i added your trendlog module to wiki: https://github.com/FedeDP/Clight/wiki/Custom-Modules#automatic-curve; have you got any comment on that? :)

NICHOLAS85, 2020-06-24:
Ah, that's alright life gets busy. And thanks for adding the module to the wiki! I haven't gotten around to creating a solution for calculating the regression based on the data without distributing some software with it which I don't think should be necessary, If I ever get around to figuring that out I'll share it here. I've updated the gist with my latest version of the module, hopefully it'll be useful to some of your users!

And I'll go ahead and close this issue.

closed 2020-06-24 by NICHOLAS85