FedeDP/Clight #128: [BUG] Clight seems to believe it's always daytime

NICHOLAS85, 2020-05-11:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Latest git version as of now (f97b5bc80c635b1163959d428e4bf3eb60497d3a);

**Used distribution:**
> Bedrock Linux, installed in Debian Testing stratum

**Describe the bug**
Clight seems to always think it is daytime regardless of what the actual time is. Running the following command at various times of day/night returns the same result:
```sh
$ busctl --user get-property org.clight.clight /org/clight/clight org.clight.clight DayTime
i 0
```

**Expected behavior**
Clight should be aware of the time like it used to be.

**To Reproduce**
Change night time values in config and see that Clight does not use them when night arrives.

FedeDP, 2020-05-11:
Thank you for opening the issue!
I noticed that too yesterday.  
I'm gonna try to understand what happened.  
Thank you very much!

FedeDP, 2020-05-11:
>  Change night time values in config and see that Clight does not use them when night arrives.

You mean to set fixes sunset times?

NICHOLAS85, 2020-05-11:
> >  Change night time values in config and see that Clight does not use them when night arrives.
> 
> 
> 
> You mean setting fixed sunset times? 

I meant changing any of the time dependent variables in the config file. For example the second value in `ac_timeouts` which corresponds with night. I’ve also tried setting different sunset times but it hasn’t seemed to have changed anything about the behavior.

FedeDP, 2020-05-12:
While i can reproduce your issue, it would be great if you can provide a verbose log!
The weird part is that GAMMA module has not been touched since ages (apart from some renamings and very small changes that do not seem related).

FedeDP, 2020-05-12:
I could no more reproduce the issue with latest clight. Is the issue still present for you? In that case, a debug log would be super helpful!

NICHOLAS85, 2020-05-12:
Sure, issue seems present as of f19fac9da73f831407616c7af57c27358685753c. I forced a sunset time before my current time as it's not dark yet in my TZ.
[clight.log](https://github.com/FedeDP/Clight/files/4618337/clight.log)

FedeDP, 2020-05-12:
Is gamma module enabled? Its configuration is not logged so I assume it is not enabled.

NICHOLAS85, 2020-05-12:
I'll attach my config file here: 
[clight.conf.txt](https://github.com/FedeDP/Clight/files/4618359/clight.conf.txt)

NICHOLAS85, 2020-05-12:
No it is not enabled.

FedeDP, 2020-05-12:
Well it is GAMMA module to tell clight when it is DAY or NIGHT :) (since forever!)

NICHOLAS85, 2020-05-12:
Ahh that explains it wow I didn't even realize I was just changing things in a disabled module. So to get day/night capabilities I need to enable the GAMMA module. Would I be able to make clight not actually adjust gamma?

closed 2020-05-12 by NICHOLAS85

FedeDP, 2020-05-12:
Well, you can set same temp on both day and night. But if you are externally managing gamma and prefer clight not to interfere, you are out of luck. I did think to split DAYTIME module from GAMMA. I may give another look at that.

NICHOLAS85, 2020-05-12:
Yea, I'm concerned it would interfere with external software if people wanted day/night capabilities without the gamma adjustment. However, I find for my setup I can probably replace Night Color (KDE's gamma adjusting solution) with clight. I just need to learn the command to set NightTemp through the dbus interface, (to temporarily disable gamma adjustment).