FedeDP/Clight #129: [BUG] Debian/Ubuntu libgsl-dev dependency change

NICHOLAS85, 2020-05-20:
**Used distribution:**
> Bedrock Linux, Debian Testing Stratum

**Describe the bug**
libgsl25 is going to be replacing libgsl23, and already has in Debian testing in the libgsl-dev package deps. Currently, the Debian cpack specification points to libgsl23, which creates a package that can't be installed if you have an up-to-date libgsl-dev package.

https://packages.debian.org/bullseye/libgsl-dev

**Expected behavior**
Prepare for this change upstream.

FedeDP, 2020-05-20:
Thanks for this alert!
I may even switch to libgsl-dev package, even if devel files will be useless (as runtime deps), as this will avoid future issues, i guess.

FedeDP, 2020-05-21:
It should now be fixed in master, using -dev packages for all library deps.

NICHOLAS85, 2020-05-21:
Fixed!

closed 2020-05-21 by NICHOLAS85