FedeDP/Clight #130: [BUG]  clight 4.0-1 doesn't compile

simony2222, 2020-05-22:
**Clight version the issue has been seen with:**
> clight 4.0-1

**Used distribution:**
> Arch Linux x86_64 (5.6.14-arch1-1)

**Describe the bug**
Clight doesn't compile (actualy I think it's Clightd, but both of them seemed liked). I will note that I have libmodule 5.0.0-1 installed, so my issue seems to be new compare to [this one](https://github.com/FedeDP/Clight/issues/103) for instance.

**Expected behavior**
For the Package to compile. ^^'

**To Reproduce**
Both pacmac and `yay -S clight` yielded the same result.

**Log**
Here is the full log and my inputs to reproduce what happened

`% yay -S clight 
:: There are 2 providers available for clight:
:: Repository AUR
    1) clight 2) clight-git 

Enter a number (default=1): 1
:: Checking for conflicts...
:: Checking for inner conflicts...
[Aur: 2]  clightd-4.0-2  clight-4.0-1

  2 clightd                                  (Build Files Exist)
  1 clight                                   (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> a
:: Deleting (1/2): /home/simon/.cache/yay/clightd
:: Deleting (2/2): /home/simon/.cache/yay/clight
:: Downloaded PKGBUILD (1/2): clightd
:: Downloaded PKGBUILD (2/2): clight
  2 clightd                                  (Build Files Exist)
  1 clight                                   (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> n
:: Parsing SRCINFO (1/2): clightd
:: Parsing SRCINFO (2/2): clight
==> Making package: clightd 4.0-2 (Fri 22 May 2020 10:15:33 PM UTC)
==> Retrieving sources...
  -> Downloading Clightd-4.0.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   119  100   119    0     0    428      0 --:--:-- --:--:-- --:--:--   429
100 36228    0 36228    0     0  34048      0 --:--:--  0:00:01 --:--:--  162k
==> Validating source files with sha256sums...
    Clightd-4.0.tar.gz ... Passed
==> Making package: clight 4.0-1 (Fri 22 May 2020 10:15:35 PM UTC)
==> Retrieving sources...
  -> Downloading Clight-4.0.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   118  100   118    0     0    220      0 --:--:-- --:--:-- --:--:--   220
100 69478    0 69478    0     0  51237      0 --:--:--  0:00:01 --:--:--  122k
==> Validating source files with sha256sums...
    Clight-4.0.tar.gz ... Passed
==> Making package: clightd 4.0-2 (Fri 22 May 2020 10:15:37 PM UTC)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found Clightd-4.0.tar.gz
==> Validating source files with sha256sums...
    Clightd-4.0.tar.gz ... Passed
==> Removing existing $srcdir/ directory...
[... 83 lines cut ...]
/sbin/ld: CMakeFiles/clightd.dir/src/modules/idle.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/idle.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/screen.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/screen.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensor.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensor.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensors/als.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensors/als.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensors/camera.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/sensors/camera.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/signal.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/modules/signal.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/utils/polkit.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/utils/polkit.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/utils/udev.c.o:(.bss+0x0): multiple definition of \`udev'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x0): first defined here
/sbin/ld: CMakeFiles/clightd.dir/src/utils/udev.c.o:(.bss+0x8): multiple definition of \`bus'; CMakeFiles/clightd.dir/src/main.c.o:(.bss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/clightd.dir/build.make:314: clightd] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/clightd.dir/all] Error 2
make: *** [Makefile:172: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
Error making: clightd`

(and I don't know why github won't format the thing proprely despite me espcaping all the "\`"...)

FedeDP, 2020-05-22:
Hi! Thanks for opening the issue!
Can you try to install clightd-git?
It seems like a Clightd bug that is now exposed with newest gcc (it was an hidden problem with older gcc I guess). If you're concerned about stability, note that clightd-git will soon be tagged as clightd 4.1 :)

simony2222, 2020-05-23:
`clightd-git` installed fine !
I guess I will stay with this one then switch to `clightd 4.1` once it's out.
Thank you very much !

FedeDP, 2020-05-23:
Great to hear!! You are welcome, thanks again for taking the time to open this issue!

closed 2020-05-23 by simony2222

FedeDP, 2020-05-24:
Hi @simony2222 !
Clightd 4.1 is now released and available on aur: https://aur.archlinux.org/packages/clightd/ .