FedeDP/Clight #131: [BUG] Segmentation fault, version 4.1

danjenson, 2020-05-24:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.1

**Used distribution:**
> Arch Linux

**Describe the bug**
Segmentation fault when running clight.

**Expected behavior**
No segmentation fault.

**To Reproduce**
Run clight daemon., then call `clight --verbose`

Please attach a Clight verbose log too!
```bash
λ clight --verbose
Clightd found, version: 4.1.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
init_kbd_backlight(): The name org.freedesktop.UPower was not provided by any .service files
Keyboard backlight calibration unsupported.
upower_check(): No route to host
UPower not present. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting AcState property
Set timeout of 0s 1ns on fd 41.
Emitting LidState property
Set timeout of 30s 0ns on fd 41.
Error loading from cache file.
37.67 -122.44 received from Geoclue2.
Loc distance: 37.67,-122.44 -> 91.00,181.00 : 3654.12 km.
New location received: 37.67, -122.44.
Set timeout of 0s 1ns on fd 42.
Emitting Location property
zsh: segmentation fault (core dumped)  clight --verbose
```

FedeDP, 2020-05-24:
Hi! Thanks for opening this issue!
Can you share your ~/.local/share/clight/clight.log too?
Moreover, please note that clight 4.1 changed conf file layour, you should probably update to it.

Thank you!

yochananmarqos, 2020-06-01:
Same here, I had been meaning to report it. At least I know it's not only me.

Apparently Clight is now afraid of the dark. Maybe I'm not the only one who watches videos with only natural light into the night?

Here's my log:

```
(W)[19:46:07]{config.c:297}	Config file /home/yochanan/.config/clight.conf not found.
(D)[19:46:07]{opts.c:456}	Inhibit module is not needed. Disabling.

Clight
* Software version:		4.1
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Thu May 28 19:46:07 2020

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Backlight path:		Unset
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		Unset
* Settings:		Unset

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

(I)[19:46:07]{main.c:139}	Clightd found, version: 4.2.
(D)[19:46:07]{backlight.c:368}	AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
(D)[19:46:07]{backlight.c:368}	BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
(I)[19:46:07]{upower.c:52}	AC cable disconnected.
(I)[19:46:07]{upower.c:63}	Laptop lid opened.
(D)[19:46:07]{interface.c:434}	Emitting AcState property
(D)[19:46:07]{timer.c:40}	Disarmed timerfd on fd 26.
(D)[19:46:07]{interface.c:434}	Emitting LidState property
(D)[19:46:07]{location.c:92}	40.57 -105.08 loaded from cache file.
(D)[19:46:07]{my_math.c:191}	Loc distance: 40.57,-105.08 -> 91.00,181.00 : 3434.86 km.
(I)[19:46:07]{location.c:60}	New location received: 40.57, -105.08.
(D)[19:46:07]{timer.c:38}	Set timeout of 0s 1ns on fd 27.
(D)[19:46:07]{interface.c:434}	Emitting Location property
(I)[19:46:07]{daytime.c:174}	Next alarm due to: Thu May 28 19:50:00 2020
(D)[19:46:07]{timer.c:38}	Set timeout of 233s 0ns on fd 27.
(D)[19:46:07]{interface.c:434}	Emitting Sunrise property
(D)[19:46:07]{timer.c:38}	Set timeout of 0s 1ns on fd 28.
(D)[19:46:08]{backlight.c:253}	Sensor '/dev/video0' is now available.
(D)[19:46:08]{backlight.c:422}	Resumed as a sensor is now available.
(D)[19:46:08]{interface.c:434}	Emitting Sunset property
(D)[19:46:08]{interface.c:434}	Emitting DayTime property
(D)[19:46:08]{interface.c:434}	Emitting NextEvent property
(D)[19:46:08]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.910857.
(I)[19:46:10]{backlight.c:293}	Ambient brightness: 0.911 -> Backlight pct: 0.786.
(D)[19:46:10]{timer.c:38}	Set timeout of 1200s 0ns on fd 28.
(D)[19:46:10]{interface.c:434}	Emitting SensorAvail property
(D)[19:46:10]{interface.c:434}	Emitting AmbientBr property
(D)[19:46:10]{interface.c:434}	Emitting BlPct property
(D)[19:50:00]{daytime.c:278}	Currently inside an event.
(I)[19:50:00]{daytime.c:174}	Next alarm due to: Thu May 28 20:20:00 2020
(D)[19:50:00]{timer.c:38}	Set timeout of 1800s 0ns on fd 27.
(D)[19:50:00]{timer.c:38}	Set timeout of 369s 0ns on fd 28.
(D)[19:50:00]{interface.c:434}	Emitting InEvent property
(D)[19:56:10]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.912542.
(I)[19:56:10]{backlight.c:293}	Ambient brightness: 0.913 -> Backlight pct: 0.787.
(D)[19:56:10]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[19:56:10]{interface.c:434}	Emitting AmbientBr property
(D)[19:56:10]{interface.c:434}	Emitting BlPct property
(D)[20:06:10]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.910601.
(I)[20:06:10]{backlight.c:293}	Ambient brightness: 0.911 -> Backlight pct: 0.786.
(D)[20:06:10]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[20:06:10]{interface.c:434}	Emitting AmbientBr property
(D)[20:06:10]{interface.c:434}	Emitting BlPct property
(D)[20:16:11]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.911379.
(I)[20:16:11]{backlight.c:293}	Ambient brightness: 0.911 -> Backlight pct: 0.786.
(D)[20:16:11]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[20:16:11]{interface.c:434}	Emitting AmbientBr property
(D)[20:16:11]{interface.c:434}	Emitting BlPct property
(D)[20:18:58]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.910911.
(I)[20:18:58]{backlight.c:293}	Ambient brightness: 0.911 -> Backlight pct: 0.786.
(D)[20:18:58]{interface.c:434}	Emitting AmbientBr property
(D)[20:18:58]{interface.c:434}	Emitting BlPct property
(D)[20:20:00]{daytime.c:278}	Currently inside an event.
(I)[20:20:00]{daytime.c:174}	Next alarm due to: Thu May 28 20:50:00 2020
(D)[20:20:00]{timer.c:38}	Set timeout of 1800s 0ns on fd 27.
(D)[20:20:00]{timer.c:38}	Set timeout of 0s 1ns on fd 28.
(D)[20:20:00]{interface.c:434}	Emitting DayTime property
(D)[20:20:01]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.911135.
(I)[20:20:01]{backlight.c:293}	Ambient brightness: 0.911 -> Backlight pct: 0.786.
(D)[20:20:01]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[20:20:01]{interface.c:434}	Emitting AmbientBr property
(D)[20:20:01]{interface.c:434}	Emitting BlPct property
(D)[20:30:01]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.910072.
(I)[20:30:01]{backlight.c:293}	Ambient brightness: 0.910 -> Backlight pct: 0.786.
(D)[20:30:01]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[20:30:01]{interface.c:434}	Emitting AmbientBr property
(D)[20:30:01]{interface.c:434}	Emitting BlPct property
(D)[20:40:02]{backlight.c:267}	Captured [5/5] from '/dev/video0'. Ambient brightness: 0.909106.
(I)[20:40:02]{backlight.c:293}	Ambient brightness: 0.909 -> Backlight pct: 0.786.
(D)[20:40:02]{timer.c:38}	Set timeout of 600s 0ns on fd 28.
(D)[20:40:02]{interface.c:434}	Emitting AmbientBr property
(D)[20:40:02]{interface.c:434}	Emitting BlPct property
```

FedeDP, 2020-06-01:
Thanks for your feedback! Can you share your conf file?
I will track it down asap!

yochananmarqos, 2020-06-01:
Oops, forgot to post it.

```
####################
# Clight conf file #
####################

## Verbose mode, useful to report bugs:
## run clight in verbose mode,
## then open issue on github attaching log
verbose = true;

###################
# INHIBITION TOOL #
########################################################
# Requires DIMMER or DPMS enabled. Disabled otherwise. #
########################################################
inhibit:
{
    ## Uncomment to let Clight manage "Docked" laptop state
    ## as an inhibition (pausing DPMS and dimmer).
    ## Note that "Docked" state is only checked when
    ## laptop lid is closed or opened.
    # inhibit_docked = true;
    
    ## Uncomment to let Clight manage inhibition requests
    ## suppressing org.freedesktop.PowerManagement too
    ## (ie: preventing your laptop to suspend/hibernate)
    # inhibit_pm = true;
};

##############################
# BACKLIGHT CALIBRATION TOOL #
##############################
backlight:
{
    ## Uncomment to disable
    # disabled = true;

    ## Uncomment to disable smooth transitions
    # no_smooth_transition = true;

    ## Transition step in percentage
    # trans_step = 0.05;

    ## Transition timeout in ms
    # trans_timeout = 30;

    ## Timeouts between captures during day/night/event on AC
    ## Set any of these to <= 0 to disable captures
    ## in the corresponding day time.
    # ac_timeouts = [ 600, 2700, 300 ];

    ## Timeouts between captures during day/night/event on BATT
    ## Set any of these to <= 0 to disable captures
    ## in the corresponding day time.
    # batt_timeouts = [ 1200, 5400, 600 ];

    ## Screen syspath to be use
    # screen_sysname = "intel_backlight";

    ## Set a threshold: if detected ambient brightness is below this threshold,
    ## capture will be discarded and no backlight change will be made.
    ## Very useful to discard captures with covered webcam.
    # shutter_threshold = 0.10;

    ## Disables automatic calibration for screen backlight.
    ## Then, it can only be manually triggered by bus api.
    # no_auto_calibration = true;

[... 176 lines cut ...]
    ##      dark ambient    -> lower monitor backlight level    -> lower monitor light contribution impact       #
    ##      high ambient    -> high monitor backlight level     -> higher monitor light contribution impact      #
    ## We can assume these 2 contributions to zero each other, letting us just compute screen-emitted brightness #
    ## and linearly multiplying it for our guessed ambient-brightness screen contribution value.                 #
    ##############################################################################################################

    ## Uncomment to disable screen-emitted brightness compensation support
    disabled = true;

    ## Screen timeouts on AC/on BATT.
    ## Set any of these to <= 0 to disable screen-emitted brightness compensation
    ## in the corresponding AC state.
    ## Disabled by default on BATT because it is quite an heavy operation,
    ## as it has to take a snapshot of your X desktop and compute its brightness.
    # timeouts = [ 30, -1 ];

    ## How much does your screen-emitted brightness affect
    ## ambient brightness, in your setup, in percentage.
    # contrib = 0.1;

    ## How many samples should be used to compute average 
    ## screen-emitted brightness.
    # num_samples = 10;
};
```

FedeDP, 2020-06-01:
Can you both confirm that Clight @ f19fac9da73f831407616c7af57c27358685753c did not show this bug?

danjenson, 2020-06-03:
I just built that commit and it worked fine for me, no segfault

yochananmarqos, 2020-06-03:
Same here, reverting to that commit seems to be working fine.

FedeDP, 2020-06-03:
Great! Just to be sure, can you try @ 7bd1b601257b54d3d6ffb8c6424c3557fac8cef3 too and confirm that bug is now appearing?

yochananmarqos, 2020-06-04:
I couldn't reproduce the issue with that commit.

FedeDP, 2020-06-04:
Wow that's great as it exonerates the biggest recent commit :)

What about at 6985c88f464e7c013b4212a44a348ee8e0a6b72a ? I am pretty sure the bug is now appearing!

yochananmarqos, 2020-06-04:
Nope, still can't reproduce it. I'm just covering the webcam with my hand and using the desktop shortcut to test. Maybe it it takes more time to occur. I'll test again later with the same conditions when it happened.

yochananmarqos, 2020-06-05:
I was able to reproduce the crash under the same original conditions.

FedeDP, 2020-06-05:
So it seems like 7bd1b60 introduced the bug unfortunately! 
Thank you anyway!
Just to clarify: are you sure it is segfaulting, or is it just crashing? (ie: is it leaving with SIGSEGV/SIGKILL or SIGABRT?) I was only able to get SIGABRTs, but never experienced a SIGSEGV...

FedeDP, 2020-06-05:
It would be really helpful if you could build clight in debug mode, and then run it through gdb, ie:
```
cd Clight/build
cmake -DCMAKE_BUILD_TYPE="Debug" ../
make
gdb ./clight
run
// ... wait on clight crashing
bt
```
and finally send me the backtrace.
Unfortunately i am not able to reproduce the SEGFAULT.
Thank you very much for your help!

yochananmarqos, 2020-06-07:
There was nothing useful. I waited until it crashed, ran `bt` and the following repeated with no end:

```
#83237 0x0000555555560a6d in ?? ()
#83238 0x0000555555560a6d in ?? ()
#83239 0x0000555555560a6d in ?? ()
#83240 0x0000555555560a6d in ?? ()
#83241 0x0000555555560a6d in ?? ()
#83242 0x0000555555560a6d in ?? ()
```

FedeDP, 2020-06-08:
:(
I don't really understand; it seems i cannot reproduce it; i tried with your config and it has been running for 20hrs. I will check when i go back home if it crashed in the meantime.

FedeDP, 2020-06-12:
I'd really like to fix this issue as it is, together with #106, the only issue i'd like to address before releasing Clight 4.1.
I really need a backtrace though, because i am not able to reproduce it; i am trying on my desktop pc at work right now but it seems it does not crash here neither.

yochananmarqos, 2020-06-12:
Why was I not able to get a good backtrace? Am I missing symbols? What I posted above was all I could get.

Apparently you won't be able to reproduce it unless you leave your laptop in a room with only natural light while the sun goes down.

FedeDP, 2020-06-12:
> Apparently you won't be able to reproduce it unless you leave your laptop in a room with only natural light while the sun goes down.

I tried that too!

> Why was I not able to get a good backtrace? Am I missing symbols? What I posted above was all I could get.

That was weird indeed; what does "file ./clight" tell?

FedeDP, 2020-06-12:
> Apparently you won't be able to reproduce it unless you leave your laptop in a room with only natural light while the sun goes down.

I tried that too!

> Why was I not able to get a good backtrace? Am I missing symbols? What I posted above was all I could get.

That was weird indeed; what does "file ./clight" tell? 
EDIT: what about `objdump --syms ./clight` too?

yochananmarqos, 2020-06-13:
I just tried to reproduce it again with the latest commit and could not. I'll try again when I have the opportunity.

yochananmarqos, 2020-06-13:
...and right before I was about to go to bed, it crashed while connected to power.

[`gdb clight` log](https://pastebin.com/AbnkPJCB)

This stands out:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7eb732e in ?? () from /usr/lib/libm.so.6
```

That's a `glibc` library.

After the segfault happens, `clight` immediately segfaults again even after restarting the `clightd` service.

> what does "file ./clight" tell?

```
$ file /usr/bin/clight 
/usr/bin/clight: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=be0e7bc46d9db92e5a60cc68750eac5647555caa, for GNU/Linux 3.2.0, stripped
```

> what about objdump --syms ./clight too?

```
$ objdump --sym /usr/bin/clight

/usr/bin/clight:     file format elf64-x86-64

SYMBOL TABLE:
no symbols
```

FedeDP, 2020-06-13:
So, it seems like you haven't got debug symbols.
Can you try to clean build folder before running the aforementioned commands?
Ie: 
```
cd Clight/build
rm -rf builds/*
cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Debug" \
        ..
make
gdb ./clight
run
// ... wait on clight crashing
bt
```

Big thanks for your patience btw, i am really appreciating it!!!

yochananmarqos, 2020-06-13:
Argh! I just realized why there are no debugging symbols. Stripping is default in `/etc/makepkg.conf`. I just did a clean build with `options=(!strip)` to my PKGBUILD. Now to wait for a crash.

FedeDP, 2020-06-13:
Great, that's right! Thank you very much! Waiting for clight to crash :)

yochananmarqos, 2020-06-14:
Gotcha!

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7eb732e in ?? () from /usr/lib/libm.so.6
(gdb) bt
#0  0x00007ffff7eb732e in ?? () from /usr/lib/libm.so.6
#1  0x0000555555569218 in calculate_sunrise_sunset (lat=lat@entry=40.3780022, 
    lng=lng@entry=-105.093803, tt=tt@entry=0x7fffff7ff4c0, event=event@entry=SUNSET, 
    tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/utils/my_math.c:124
#2  0x00005555555697be in calculate_sunset (lat=lat@entry=40.3780022, 
    lng=lng@entry=-105.093803, tt=tt@entry=0x7fffff7ff4c0, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/utils/my_math.c:179
#3  0x0000555555560809 in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:193
#4  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#5  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#6  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#7  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#8  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#9  0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#10 0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#11 0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
    at /home/yochanan/Documents/pkgbuilds/clight-git/src/clight/src/modules/daytime.c:200
#12 0x0000555555560a6d in get_next_events (now=now@entry=0x7fffffffdc28, 
    lat=40.3780022, lon=-105.093803, tomorrow=tomorrow@entry=true)
```
There was more, but it was all the same.

FedeDP, 2020-06-14:
Great!!
Can you try if this small test is crashing for you?
[test.tar.gz](https://github.com/FedeDP/Clight/files/4776164/test.tar.gz)

Extract it and build it with `gcc test.c -o test -lm` and run with `./test`.
Thank you!

yochananmarqos, 2020-06-14:
It didn't crash, it just exited normally.

```
$ gcc test.c -o test -lm
$ ./test 
$ 
```

FedeDP, 2020-06-14:
Can you try with new master branch? Thanks!

yochananmarqos, 2020-06-17:
No crashes for a few nights in a row with the latest commit. Either it's fixed or it's shy because it knows it's being watched. 😆

FedeDP, 2020-06-17:
>  it's shy because it knows it's being watched. laughing

This second one seems most likely :D  
Great!
@danjenson Are you able to test with latest clight too? Thank you both for your super support!

FedeDP, 2020-06-22:
I close this one; feel free to reopen if the issue appears again!

closed 2020-06-22 by FedeDP