FedeDP/Clight #132: [BUG] Brightness calibration turns off screen entirely

finkrer, 2020-06-08:
**Clight version the issue has been seen with:**
> Latest master

**Used distribution:**
> Arch Linux

**Describe the bug**
When Clight detects 0 ambient brightness, it likewise sets the backlight to 0. This turns off the screen entirely, which means I don't see anything and honestly it's really confusing and not obvious what happened and how to fix it.

**Expected behavior**
I expect the default settings to be such that Clight does not turn my screen off. I know there is a way to fix that in the settings, but I don't think anyone wants that behavior by default, although I'm not sure if it works differently on other laptops.

**To Reproduce**
Just block the webcam and trigger a calibration. Actually it wasn't blocked, which may be another bug, I guess. Seems to happen when the screen was turned off due to inactivity.

```
(D)[19:08:07]{display.c:43}     Entering dpms state...
(D)[19:08:07]{interface.c:434}  Emitting DisplayState property
(D)[20:56:48]{display.c:48}     Leaving dpms state...
(D)[20:56:48]{display.c:53}     Leaving dimmed state...
(D)[20:56:48]{validations.c:133}        Failed to validate display state request.
(D)[20:56:48]{interface.c:434}  Emitting DisplayState property
(D)[20:56:48]{interface.c:434}  Emitting BlPct property
(D)[20:56:48]{screen.c:193}     Average screen-emitted brightness: 0.070431.
(D)[20:56:48]{timer.c:38}       Set timeout of 30s 0ns on fd 40.
(D)[20:56:48]{interface.c:434}  Emitting ScreenComp property
(D)[20:56:48]{backlight.c:276}  Captured [5/5] from '/dev/video0'. Ambient brightness: 0.000000.
(I)[20:56:48]{backlight.c:300}  Ambient brightness: 0.000 (-0.070 screen compensation) -> Backlight pct: 0.000.
(D)[20:56:48]{timer.c:38}       Set timeout of 2700s 0ns on fd 42.
(D)[20:56:48]{interface.c:434}  Emitting AmbientBr property
(D)[20:56:48]{interface.c:434}  Emitting BlPct property
(D)[20:57:18]{screen.c:193}     Average screen-emitted brightness: 0.062549.
(D)[20:57:18]{timer.c:38}       Set timeout of 30s 0ns on fd 40.
(D)[20:57:18]{interface.c:434}  Emitting ScreenComp property
(D)[20:57:48]{screen.c:193}     Average screen-emitted brightness: 0.054667.
(D)[20:57:48]{timer.c:38}       Set timeout of 30s 0ns on fd 40.
(D)[20:57:48]{interface.c:434}  Emitting ScreenComp property
(D)[20:58:14]{backlight.c:276}  Captured [5/5] from '/dev/video0'. Ambient brightness: 0.716659.
(I)[20:58:14]{backlight.c:300}  Ambient brightness: 0.717 (-0.055 screen compensation) -> Backlight pct: 0.853.
(D)[20:58:14]{interface.c:434}  Emitting AmbientBr property
(D)[20:58:14]{interface.c:434}  Emitting BlPct property
(D)[20:58:18]{screen.c:193}     Average screen-emitted brightness: 0.048941.
(D)[20:58:18]{timer.c:38}       Set timeout of 30s 0ns on fd 40.
(D)[20:58:18]{interface.c:434}  Emitting ScreenComp property
```

FedeDP, 2020-06-08:
Hi! Thanks for reporting this issue.

> I expect the default settings to be such that Clight does not turn my screen off. I know there is a way to fix that in the settings, but I don't think anyone wants that behavior by default, although I'm not sure if it works differently on other laptops.

Well, most laptops do not switch off screen on 0.0 backlight (mine for example) afaik; this is why the default behaviour is not to discard any capture, and a "workaroud" option was added.

finkrer, 2020-06-08:
Alright, thank you for replying! I guess I can close this then.

I think it would be helpful to include some info about this somewhere, just in case someone wonders what's going on.

I assume you are talking about the shutter threshold? I think what is actually better is to change the curve so that it just doesn't have 0 as a possible value for backlight. I don't really want to discard low ambient brightness values.

FedeDP, 2020-06-08:
> I think it would be helpful to include some info about this somewhere, just in case someone wonders what's going on.

Yeah, sure! It is a nice idea! Before shutter_threshold was a thing, there was another option that explicitly targeted those cases.

> I assume you are talking about the shutter threshold

Yes I am :)

> I think what is actually better is to change the curve so that it just doesn't have 0 as a possible value for backlight. I don't really want to discard low ambient brightness values.

Yeah, that's fine too (and, as you said, probably better). Shutter threshold is mostly useful to discard covered captures.

NICHOLAS85, 2021-01-08:
Hey sorry to reopen an old issue but I've run into a small annoyance related to this:
I agree it's possible to change the curve so 0 is no longer a possible backlight value, and have done so myself. However, if a user has the Screen compensation module enabled, it is still possible for Clight to set a backlight level of 0. This usually occurs in lowlight situations where the pre-screen adjustment value is already low and after subtracting the screen value it hits 0.

FedeDP, 2021-01-09:
Np mate!
You are right indeed! What about using `shutter_threshold` in conjunction with your curve?
It should work the way you want indeed: https://github.com/FedeDP/Clight/blob/b44c1e10e04af9968febd5fb2cc86188c378e562/src/modules/backlight.c#L317
Let me know if it fixes your issue!

NICHOLAS85, 2021-01-09:
I find that `shutter_threshold` doesn't help in situations where I'd still like the backlight to dim. For example, suspend midday at 100% ambient brightness and then open at midnight with 0% ambient brightness. With `shutter_threshold` Clight would just leave the device at the previous (100%) brightness requiring me to manually bring it down to a reasonable level. I'd rather Clight bring it down to some lower limit.

FedeDP, 2021-01-16:
I see; mmh that is a pretty unfortunate issue! 
Well, i think i can eg: use min and max values from backlight curves (both on ac and on batt) as threshold for minimum and maximum value for backlight level; this way, if level is below minimum, we will set minimum.
What do you think?

NICHOLAS85, 2021-01-18:
That sounds like a potential solution!

FedeDP, 2021-01-21:
It should now be fixed in master!

closed 2021-01-22 by FedeDP

NICHOLAS85, 2021-01-23:
Looks like it's working as expected, thanks!