FedeDP/Clight #134: [BUG] Clight halts on waking up from DPMS standby

pranjlp, 2020-06-27:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.0.1 (Clightd 4.1)

**Used distribution:**
> Arch

**Describe the bug**
Wake up from a dpms standby, makes clight inactive. No new enteries on to the stdout/logfile.  The webcam flash doesnt blink.  The process exits in the process list.  The busctl command doesnt work either. The workaround is to delete the file to release the lock and start a new clight process.

**Expected behavior**
Clight should resume operation, after waking from standby from dpms

**To Reproduce**
Let the dpms kick in and wake up from it using input activity.  Try using busctl api to calibrate. 

Please attach a Clight verbose log too!
[clight.log](https://github.com/FedeDP/Clight/files/4840683/clight.log)

FedeDP, 2020-06-27:
Hi, thanks for opening this issue!
> (W)[20:04:31]{interface.c:239}	Failed to create org.clight.clight dbus interface: File exists
> (W)[20:04:31]{interface.c:258}	Failed to init.

These errors are weird, it seems there is already a process owning org.clight.clight dbus interface; did you by any chance run clight once more while it was already running?

pranjlp, 2020-06-27:
Didn't have that issue before. Added a more thorough list of steps to reproduce.

[clight_before.log](https://github.com/FedeDP/Clight/files/4841262/clight.log)
before-log taken at 03:00 at night. The monitor entered dpms at 23:00 and is stuck there

busctl calibrate Command executes successfully without any exit codes, but with no effect.
both clight and clightd is there on the process list

starting a new clight instance on the terminal, expectedly doesn't work
```
Resource temporarily unavailable
A lock is present on /home/ra-s_al_ghul/.local/share/clight/clight.log. Another clight instance running?
```
Now I remove the log file.  However the clight process doesn't die, and I still can't start a new process.

Manually killed the process. Started clight with

```
clight </dev/null &>/dev/null &
```
New log generated. Webcam shows activity.
[clight_after.log](https://github.com/FedeDP/Clight/files/4841292/clight.log)

My clight config
[clight.txt](https://github.com/FedeDP/Clight/files/4841295/clight.txt)

pranjlp, 2020-06-28:
Hi,
Turns out my clightd service was enabled. After disabling it, the issue seemed to have vanished for last 6-7 hours.

As a side effect though, GAMMA and SCREEN module are disabled, when clight is started post boot.  I guess this is a known bug, discussed in other issues. Let me know if this is resolved somewhere.

A workaround, I figured, is to start clightd manually with systemd post-login and then start clight using desktop file/via terminal. Everything seems to work perfectly then, with all modules enabled.

There are no relevent logs I could find in clightd journal as why those modules were disabled.

FedeDP, 2020-06-28:
It should be OK to have Clightd enabled (well, not only ok but suggested too!).
Can you try to update to clight-git package? (it is quite stable right now and very near 4.1 release).

Clight 4.1 has lots of improvements and fixes over 4.0 and, at least the "side effect" is known and fixed!
Let me know, and thank you very much for your time spent workarounding the issues with clight :)

pranjlp, 2020-06-28:
Thanks for responding to all the issues, and actively working on this.  Hope this ends up in official repos, sometime. Last version seemed robust enough. I'll be closing this.

closed 2020-06-28 by pranjlp

FedeDP, 2020-06-28:
> Thanks for responding to all the issues, and actively working on this. Hope this ends up in official repos, sometime.

Thank you for spending your time on it and for your kind words :)
It would be really great to have clight in arch repositories!