FedeDP/Clight #135: [FEATURE REQ] Add example usage of dbus API

DjebbZ, 2020-07-12:
**Is your feature request related to a problem? Please describe.**
I'd like to bind a shortcut to increase/decrease the backlight (using the dedicated laptop keys), but I can't find the right syntax. `dbus-send /org/clight/clight org.clight.clight.IncBl double:30.0` produces no output and no change, and I've tried int16/32/64:30, same result. Sorry I'm a total noob in the dbus world.

**Describe the solution you'd like**
1. A "copy-pastable" example of the command for my request
2. In exchange, I will add the usage documentation in the Wiki and make the various single letters for IN and OUT more clear in the same wiki page, I suppose they represent the types like int16,boolean etc but I'm not even sure. Never stumbled across this notation so I don't know what you're refering to. I'd like to make it simpler for everyone.

**Describe alternatives you've considered**
Well, using xbacklight even it doesn't provide all the features of Clight.

Thanks in advance!

FedeDP, 2020-07-12:
Hi! Thanks for opening this request.
I always use `busctl` command as i am familiar with it; with busctl, it is as easy as:
`busctl --user call org.clight.clight /org/clight/clight org.clight.clight DecBl "d" 0.05

DjebbZ, 2020-07-13:
Thanks for the prompt answer! Give a week or 2 and I'll do the change. What does the "d" stands for?

FedeDP, 2020-07-13:
You are warmly welcome :)

"d" stands for "Double"; have a look here for other types: https://dbus.freedesktop.org/doc/dbus-specification.html#basic-types ! Thank you!

stites, 2020-08-04:
:+1:

Completely new to interfacing with dbus -- it's probably worth noting that without `--user`, I came across `org.clightd.clightd` and ran into some trouble with the API.

One potentially nice feature, especially for newbies like me, might be to add this sort of information to a manpage, help flag, or possibly even adding `Help` to clightd's API (although my scenario might be a bit niche).

FedeDP, 2020-08-04:
> Completely new to interfacing with dbus -- it's probably worth noting that without --user in busctl, I came across org.clightd.clightd and ran into some trouble with the API.

Yes, "--system" (the default, if neither --user or --system are specified) means you want to talk with a system level service ([Clightd](https://github.com/FedeDP/Clightd) is one of those!) while "--user" means that you want to talk to an user level service, like Clight.

> One potentially nice feature, especially for newbies like me, might be to add this sort of information to a manpage, help flag, or possibly even adding Help to clightd's API (although my scenario might be a bit niche).

There is no need for that as DBus already provides it: `busctl --user tree org.clight.clight` and then `busctl --user introspect org.clight.clight $path`
where $path was found with busctl tree.

As you can see, you will have some columns in `introspect` command that describe each exposed API.

stites, 2020-08-05:
For my particular use-case, I was thinking something along the lines of:
```
$ busctl get-property org.clightd.clightd /org/clightd/clightd org.clightd.clightd Help
s "For help with Clightd's bus interface, see XXX docs. To interface with Clightd, please refer to the Clight Bus API (ie: run busctl with --user) -- detailed examples and usage can be found at YYY" 
```

But since clightd doesn't really have a bus api, possibly just the second part with `YYY` being whatever artifact closes this ticket.

FedeDP, 2020-08-05:
Hi!
Well, there is the online documentation; if you get familiar with dbus, it is pretty straight-forward to understand Clight dbus API using `introspect` dbus command.

> But since clightd doesn't really have a bus api, possibly just the second part with YYY being whatever artifact closes this ticket.

Clightd has a dbus API too, but it is exposed as a system services (thus run as root); that is indeed its purpose, quoting its readme:
> Clightd is a bus interface that lets you easily set screen brightness, gamma temperature and get ambient brightness through webcam frames capture or ALS devices.

You can find its doc here: https://github.com/FedeDP/Clightd/wiki/Api

FedeDP, 2020-09-14:
I'll close this one. Feel free to reopen it if you have any more questions!

closed 2020-09-14 by FedeDP