FedeDP/Clight #136: Toggle redshift gamma?

loki7990, 2020-07-22:
Hello,

I'm sorry if this has already been answered before but I can't seem to find an answer for my question. Is it possible to toggle the redshift gamma? As in, turn it on if its off, turn it off if its on? I'm not very familiar with using busctl so I wasn't sure how to go about doing that through the API wiki page.

FedeDP, 2020-07-22:
Hi!
There is no direct way to achieve that; in Clight there is no "gamma" on/off indeed; you have daily and nightly screen temperatures. So, in a sense, gamma is always ON (when enabled obviously).
You are looking for change Gamma {Day/Night}Temp; you can achieve that with:
```
# Change Night temp to day temp (6500 is default day temp), thus "turning off" gamma while on night
busctl --user set-property org.clight.clight /org/clight/clight/Conf /org/clight/clight/Conf/Gamma NightTemp "i" 6500

# Change Day temp to night temp (4000 is default night temp),  thus "turning on" gamma while on day
busctl --user set-property org.clight.clight /org/clight/clight/Conf /org/clight/clight/Conf/Gamma DayTemp "i" 4000
```

loki7990, 2020-07-22:
Thanks so much for the prompt response! That's exactly what I was looking for. However, I get the error message when I try to use either command:
```
Failed to set property DayTemp on interface /org/clight/clight/Conf/Gamma: Unknown interface /org/clight/clight/Conf/Gamma or property DayTemp
```

I'm using the git package on the AUR, so was there a change on this property recently?

Also, Is it possible to query what the current temperature is? I'd like to put these commands into a simple shell script

NICHOLAS85, 2020-07-22:
You are able to query what the current temperature is using the following info: https://github.com/FedeDP/Clight/wiki/Api#read-only-bus-properties. Once I get access to my install with Clight I can send the command which is used, if FedeDP doesn't respond before then.

I've created some clight [modules](https://github.com/FedeDP/Clight/wiki/Modules) that you may be interested in which deal with managing gamma. [1](https://github.com/NICHOLAS85/dotfiles/blob/897a3840322fe1e56995aaa3a7fca6218bc75431/.local/share/clight/inhibit_bl_mod.c), [2](https://github.com/NICHOLAS85/dotfiles/blob/xps_13_9365_refresh/.local/share/clight/inhibit_bl2.c), and [3](https://github.com/NICHOLAS85/dotfiles/blob/xps_13_9365_refresh/.local/share/clight/inhibit_bl3.c).
These modules all, when a video starts playing, disable keyboard backlight, pause backlight calibration, and turn off gamma correction.  These changes are reverted when the video stops playing. 1 manages redshift, 2 manages KDE's Night Color, and 3 manages Clights builtin gamma correction. Hope you can find them useful!

FedeDP, 2020-07-22:
> You are able to query what the current temperature is using the following info: https://github.com/FedeDP/Clight/wiki/Api#read-only-bus-properties. 

Yes, that is:
`busctl --user get-property org.clight.clight /org/clight/clight org.clight.clight Temp`

> However, I get the error message when I try to use either command:

Yeah, sorry; i could not test it :) 
These are correct ones:
```
# Change Night temp to day temp (6500 is default day temp), thus "turning off" gamma while on night
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma NightTemp "i" 6500

# Change Day temp to night temp (4000 is default night temp),  thus "turning on" gamma while on day
busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" 4000
```

loki7990, 2020-07-23:
Ah perfect, thank you! Those work! I had to make sure that I was setting the right temperature for day/ night time but otherwise that was pretty straight forward. One last question, is it possible to query whether the dimmer or autobacklight calibration is currently active?  Based on your examples, I thought that 
```
 busctl --user get-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight Disabled
```
and
```
 busctl --user get-property org.clight.clight /org/clight/clight/Conf/Dimmer org.clight.clight.Conf.Dimmer Disabled
```

would tell me that current state which would allow me to execute the appropriate command based on the ones given in the .desktop file. Both of those above commands fail with a similar message as before unfortunately

FedeDP, 2020-07-23:
> Both of those above commands fail with a similar message as before unfortunately

Yeah, since clight 4.1 disabled modules haven't got their path exported on bus; thus you should just check which object path are available on bus, through:
`busctl --user tree org.clight.clight`

loki7990, 2020-07-23:
Interesting, it seems that there is a Inhibit module that is not documented in the wiki. 
```
├─/ScreenSaver
└─/org
  ├─/org/clight
  │ └─/org/clight/clight
  │   └─/org/clight/clight/Conf
  │     ├─/org/clight/clight/Conf/Backlight
  │     ├─/org/clight/clight/Conf/Daytime
  │     ├─/org/clight/clight/Conf/Dimmer
  │     ├─/org/clight/clight/Conf/Gamma
  │     ├─/org/clight/clight/Conf/Inhibit***
  │     ├─/org/clight/clight/Conf/Kbd
  │     ├─/org/clight/clight/Conf/Screen
  │     └─/org/clight/clight/Conf/Sensor
  └─/org/freedesktop
    └─/org/freedesktop/ScreenSaver
```
How can I get the adjustable properties of a module? I'd like to see if I can tinker with it

FedeDP, 2020-07-23:
Whoops, nice catch thank you! I will update wiki asap!

> How can I get the adjustable properties of a module? I'd like to see if I can tinker with it

If you check with `busctl --user introspect org.clight.clight /org/clight/clight/Conf/Backlight` you can see that some properties have "writable" flag; these properties can be setted through `busctl --user set-property`!

FedeDP, 2020-07-23:
Wiki page should now be properly updated: https://github.com/FedeDP/Clight/wiki/Api .

loki7990, 2020-07-23:
Thanks for that! After messing around, I've managed to find a way to query whether the Dimmer is inhibited
```
busctl --user get-property org.clight.clight /org/clight/clight org.clight.clight Inhibited
```
Yet I haven't found a way to query whether the backlight autocalibration is inhibited or not. Is that possible? Clearly I can set NoAutoCalib, but I need a way to determine the current state of it in order to make a simple toggle script

FedeDP, 2020-07-23:
> Thanks for that! After messing around, I've managed to find a way to query whether the Dimmer is inhibited

Correct!!

> Yet I haven't found a way to query whether the backlight autocalibration is inhibited or not.

You should just check
`busctl --user get-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib`

loki7990, 2020-07-23:
Perfect! That worked! Thank you so much!

closed 2020-07-23 by loki7990