FedeDP/Clight #137: Clight automatic brightness control

blackapple113, 2020-07-22:
Hi,
I am a newbie with Linux and I wanted to automate my keyboard and screen backlight. I've setup clight on Arch and it works. But it recalibrates only if I start the app via Terminal or app-icon.

My question is if it is possible to automatically recalibrate every five seconds or so. Or do I need to set it up manually e.g. over systemd or cronjobs?

kind regards

FedeDP, 2020-07-23:
Clight does indeed continuously capture, but it uses a timeout to avoid wasting battery.  
Have a look at `/etc/default/clight.conf` file!

FedeDP, 2020-07-23:
Note that clight starts automatically at boot; its desktop file allows users to perform a "quick calibration"!

blackapple113, 2020-07-23:
Thanks for the fast response :)
I edited the `etc/default/clight.conf`-file and set up the `ac_capture_timeouts = [ 5, 5, 5 ];` as well as the `batt_capture_timeouts = [ 5, 5, 5];`. I assume that these are milliseconds (it is not commented there). I also setup the `backlight_trans_timeout = 30`.

But even when I restart the `clight.service` the backlight will not adjust accordingly every x milliseconds. It will only work when I start clight either over the terminal via `clight` or over the app-icon.

Did I miss something in the configuration? I attached the log-file output. What I also don't understand is why the timeouts in the log-file are not accordingly to my configuration.

Thank you in advance

[clight.log](https://github.com/FedeDP/Clight/files/4965908/clight.log)

FedeDP, 2020-07-23:
Timeouts are in seconds indeed!

> But even when I restart the clight.service the backlight will not adjust accordingly every x milliseconds. 

There is no `clight.service`!
What DE are you using? Clight should automatically start if your DE supports xdg desktop specification.
Otherwise, you should start it from your wm autostart script.
Have a look here please: https://github.com/FedeDP/Clight/wiki/Start .

Timeouts may vary because each capture will be taken exactly 5s (in your case) after the latter has finished. Since webcam captures are not instant, it means each clight calibration will happen every "5s + webcam capture time".

blackapple113, 2020-07-23:
> There is no clight.service!
I'm sorry I meant `clightd.service`.

I am using KDE Plasma. When I look for processes there are two clightd and one clight process running so i assume it is starting automatically. But calibration does not work automatically.

> Timeouts may vary because each capture will be taken exactly 5s (in your case) after the latter has finished. Since webcam captures are not instant, it means each clight calibration will happen every "5s + webcam capture time".

But why differs the log-file from my configuration file in view of the timeouts? Shouldn't those times be the same?

FedeDP, 2020-07-23:
> But why differs the log-file from my configuration file in view of the timeouts? Shouldn't those times be the same?

Oh, i did not notice the difference! I thought you were talking about:
> (I)[14:32:06]{backlight.c:249}	Ambient brightness: 0.044 -> Backlight pct: 0.056.
> (I)[14:32:13]{backlight.c:249}	Ambient brightness: 0.287 -> Backlight pct: 0.371.

(ie: 7s instead of 5s between captures!)

It seems like clight is failing to read config file; can you  attach your /etc/default/clight.conf here?

> But calibration does not work automatically.

I am using kde plasma too; clight automatically starts (you can try to reboot your system and check that clight is running.  
Probably clight is using its default timeouts that are much longer than yours (10mins between captures).

blackapple113, 2020-07-23:
> Probably clight is using its default timeouts that are much longer than yours (10mins between captures).

That could be possible. Do I need to copy the config file anywhere else for clight to recognize? 

[clight.txt](https://github.com/FedeDP/Clight/files/4966051/clight.txt)

FedeDP, 2020-07-23:
Conf file seems fine; can you try to manually issue a:
`killall clight && clight`
and then paste here its output? It should print if it finds any error loading conf file.

blackapple113, 2020-07-23:
```
Clightd found, version: 4.2.
Initial AC state: disconnected.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
Normal transition to 6500 gamma temp started.
Next alarm due to: Thu Jul 23 21:30:00 2020
Ambient brightness: 0.232 -> Backlight pct: 0.307.

```

After that nothing happens.

FedeDP, 2020-07-23:
Super weird; can you share `ls -l /etc/default/clight.conf`output?

blackapple113, 2020-07-23:
`-rw-r--r-- 1 root root 7338 23. Jul 14:28 /etc/default/clight.conf`

Maybe its a problem with root?

blackapple113, 2020-07-23:
Okay wow I found my failure.

> Both files are checked when clight starts, in this order: global -> user-local -> cmdline opts.

That means that the user-local config file is overwriting the global file and is so on the only relevant for configuring clight. 

Thank you very much for your time and effort!

closed 2020-07-23 by blackapple113

FedeDP, 2020-07-24:
> Thank you very much for your time and effort!

Nice that you fixed the issue!!