FedeDP/Clight #139: [BUG] GAMMA and SCREEN forcefully disabled if clightd is not already running

mirfatif, 2020-07-28:
**Clight version the issue has been seen with:**
> 4.0

**Used distribution:**
> Ubuntu 20.04

**Describe the bug**
After boot log states:
```
(W)[20:56:11]{opts.c:189}   GAMMA forcefully disabled as Clightd was built without gamma support.
(W)[20:56:11]{opts.c:194}   SCREEN forcefully disabled as Clightd was built without screen support.

...

### Modules ###
* Backlight:        Enabled
* Gamma:        Disabled
* Dimmer:       Enabled
* Dpms:     Enabled
* Screen:       Disabled
```
But I built `clightd` with `-DENABLE_GAMMA=1 -DENABLE_SCREEN=1`.

Once I kill and restart `clight` (not `clightd`), both GAMMA and SCREEN are ENABLED and work normally:
```
...

### Modules ###
* Backlight:        Enabled
* Gamma:        Enabled
* Dimmer:       Enabled
* Dpms:     Enabled
* Screen:       Enabled
```
And I think it's because `clightd` is already running in latter case. If I enable `clightd` service (`systemctl enable clightd`), and it's running on boot when `clight` calls it, the problem doesn't occur.

Not sure if issue belongs to `clightd`.

**Expected behavior**
GAMMA should be adjusted on boot.

**To Reproduce**
Stated above.

FedeDP, 2020-07-29:
Hi, thanks for opening the issue!
This is somewhat a weird issue as all clight does is calling an "Introspect" on Clightd and then checks its declared object paths; you can see here: https://github.com/FedeDP/Clight/blob/master/src/conf/opts.c#L223 .

Probably Clightd, being started in that moment, sees its answer cut.
I will try to investigate further anyway!

FedeDP, 2020-07-29:
This is what DBus.Introspect tells us about clightd in that moment:
```
<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node>
 <interface name="org.freedesktop.DBus.Peer">
  <method name="Ping"/>
  <method name="GetMachineId">
   <arg type="s" name="machine_uuid" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Introspectable">
  <method name="Introspect">
   <arg name="data" type="s" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Properties">
  <method name="Get">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="out" type="v"/>
  </method>
  <method name="GetAll">
   <arg name="interface" direction="in" type="s"/>
   <arg name="properties" direction="out" type="a{sv}"/>
  </method>
  <method name="Set">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="in" type="v"/>
  </method>
  <signal name="PropertiesChanged">
   <arg type="s" name="interface"/>
   <arg type="a{sv}" name="changed_properties"/>
   <arg type="as" name="invalidated_properties"/>
  </signal>
 </interface>
 <interface name="org.clightd.clightd">
  <property name="Version" type="s" access="read">
   <annotation name="org.freedesktop.DBus.Property.EmitsChangedSignal" value="const"/>
  </property>
 </interface>
 <node name="Dpms"/>
 <node name="Backlight"/>
 <node name="Idle"/>
</node>
```

FedeDP, 2020-07-29:
To me, it seems like a timing problem somewhere into sd-bus library; 
here output of sd_bus_message_dump() directly on Introspect method call:
```
MESSAGE "s" {
        STRING "<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node>
 <interface name="org.freedesktop.DBus.Peer">
  <method name="Ping"/>
  <method name="GetMachineId">
   <arg type="s" name="machine_uuid" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Introspectable">
  <method name="Introspect">
   <arg name="data" type="s" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Properties">
  <method name="Get">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="out" type="v"/>
  </method>
  <method name="GetAll">
   <arg name="interface" direction="in" type="s"/>
   <arg name="properties" direction="out" type="a{sv}"/>
  </method>
  <method name="Set">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="in" type="v"/>
  </method>
  <signal name="PropertiesChanged">
   <arg type="s" name="interface"/>
   <arg type="a{sv}" name="changed_properties"/>
   <arg type="as" name="invalidated_properties"/>
  </signal>
 </interface>
 <interface name="org.clightd.clightd">
  <property name="Version" type="s" access="read">
   <annotation name="org.freedesktop.DBus.Property.EmitsChangedSignal" value="const"/>
  </property>
 </interface>
 <node name="Backlight"/>
 <node name="Idle"/>
 <node name="Dpms"/>
</node>
";
};
```

Here, a small python test:
```
#!/usr/bin/env python

import dbus

bus = dbus.SystemBus()
obj = bus.get_object('org.clightd.clightd', '/org/clightd/clightd')
iface = dbus.Interface(obj, 'org.freedesktop.DBus.Introspectable')
xml_string = iface.Introspect()
print(xml_string)
```

(as you can see, it does the very same thing that clight does)
And here its output:
```
<!DOCTYPE node PUBLIC "-//freedesktop//DTD D-BUS Object Introspection 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/introspect.dtd">
<node>
 <interface name="org.freedesktop.DBus.Peer">
  <method name="Ping"/>
  <method name="GetMachineId">
   <arg type="s" name="machine_uuid" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Introspectable">
  <method name="Introspect">
   <arg name="data" type="s" direction="out"/>
  </method>
 </interface>
 <interface name="org.freedesktop.DBus.Properties">
  <method name="Get">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="out" type="v"/>
  </method>
  <method name="GetAll">
   <arg name="interface" direction="in" type="s"/>
   <arg name="properties" direction="out" type="a{sv}"/>
  </method>
  <method name="Set">
   <arg name="interface" direction="in" type="s"/>
   <arg name="property" direction="in" type="s"/>
   <arg name="value" direction="in" type="v"/>
  </method>
  <signal name="PropertiesChanged">
   <arg type="s" name="interface"/>
   <arg type="a{sv}" name="changed_properties"/>
   <arg type="as" name="invalidated_properties"/>
  </signal>
 </interface>
 <interface name="org.clightd.clightd">
  <property name="Version" type="s" access="read">
   <annotation name="org.freedesktop.DBus.Property.EmitsChangedSignal" value="const"/>
  </property>
 </interface>
 <node name="Dpms"/>
 <node name="Sensor"/>
 <node name="Screen"/>
 <node name="Idle"/>
 <node name="Backlight"/>
 <node name="Gamma"/>
</node>
```

They were both called while clightd was not running yet.

mirfatif, 2020-08-05:
@FedeDP thanks for the response. I'm a mechanical engineer, and the maximum detail I know about DBus is that it's some kind of IPC mechanism. So don't expect a sane response or a contribution from my side towards resolving the problem 😄 
Let me know if you need more logs / traces, or need to perform some tests etc. I'm available.

FedeDP, 2020-08-05:
Well, i can reproduce the issue; i will dive into `busctl tree` source code to understand if there is anything i am doing wrong!
Otherwise, it is sd-bus library fault and i cannot do much more!

Anyway, the preferred way, as stated in the wiki pages, is to enable clightd.service: https://github.com/FedeDP/Clight/wiki/Start

mirfatif, 2020-08-05:
> Anyway, the preferred way, as stated in the wiki pages, is to enable clightd.service: https://github.com/FedeDP/Clight/wiki/Start

Yeah already did that. Just wanted to point out the flaw in your beautiful piece of code.
You can close the issue at anytime if no more needed for follow-up.

FedeDP, 2020-08-05:
> Just wanted to point out the flaw in your beautiful piece of code.

Thanks for the great compliment :) 
I will dive into it a little more; hopefully i find the root cause!

mirfatif, 2020-08-09:
@FedeDP please look into this as well. After boot, when `clight` tries to enter dimmed state for the first time, it fails with the following error. Later on cursor movement or power event, it enters dimmed state and remains there forever, whatever happens. Once I kill `clight` and restart, it start functioning normally.

```
...
(D)[17:14:17]{bus.c:279}	get_screen_brightness(): Input/output error
...
```
At the same time `clightd` throws this message:
```
~$ journalctl -u clightd
...
Aug 09 17:14:17 irfan-Inspiron-5521 clightd[790]: Setting Pixel Format: Device or resource busy
```

This happens sometimes. I don't know exactly with what conditions, looks random.

Let me know if you need configuration or complete log.