FedeDP/Clight #141: [BUG] Parts of custom module no longer works

NICHOLAS85, 2020-08-17:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.1 (Git version acf04c2)

**Used distribution:**
> Bedrock Linux 0.7.18beta3 (Poki), Debian stratum

**Describe the bug**
I've created a custom module based on the inhibit_bl one presented in the wiki. It used to work flawlessly however now I'm experiencing issues. Specifically my screen temperature change is no longer changing and keyboard backlight level is not changing to 0%. 

**Expected behavior**
To change gamma to 6500 on video playing and 4200 when not (At night) and to turn keyboard backlight to 0% when playing and back when not. 

**To Reproduce**
Here's the module, may I get some assistance debugging this issue?
```c
#include <clight/public.h>

/**
 * Modified to not set brightness to 100% and toggle gamma correction
 **/

/**
 * Small example custom module for Clight.
 *
 * It just hooks on INHIBIT state updates and:
 * -> when entering INHIBIT state (eg: start watching a movie), disables automatic calibration, switch off kbd backlight, pause night color
 * -> when leaving INHIBIT state, triggers a new backlight calibration and re-enables automatic calibration, unpause night color
 **/

/*
 * Rename to: inhibit_bl3.c
 *
 * Build with: gcc -shared -fPIC inhibit_bl3.c -o inhibit_bl3 -Wno-unused
 *
 * Place inhibit_bl_mod in: $HOME/.local/share/clight/modules.d/ OR, globally, in /usr/share/clight/modules.d/
 */

CLIGHT_MODULE("INHIBIT_BL");

DECLARE_MSG(capture_req, CAPTURE_REQ);
DECLARE_MSG(calib_req, NO_AUTOCALIB_REQ);
DECLARE_MSG(kbd_bl_req, KBD_BL_REQ);
DECLARE_MSG(temp_req, TEMP_REQ);

static void init(void) {
    capture_req.capture.reset_timer = false; // avoid resetting clight internal BACKLIGHT timer
    kbd_bl_req.bl.new = 0.0;
    temp_req.temp.smooth = 1;
    //temp_req.temp.step = 50;
    //temp_req.temp.timeout = 100;
    temp_req.temp.daytime = NIGHT;
    
    /* Subscribe to inhibit state */
    M_SUB(INHIBIT_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case INHIBIT_UPD: {
        inhibit_upd *up = (inhibit_upd *)MSG_DATA();
        calib_req.nocalib.new = up->new;
        if (up->new) {
            INFO("Pausing autocalibration and night color.\n");
            M_PUB(&kbd_bl_req);  // set 0% kbd backlight
            temp_req.temp.new = 6500;
        } else {
            INFO("Doing a quick backlight calibration and unpausing night color.\n");
            M_PUB(&capture_req); // ask for a quick calibration
            temp_req.temp.new = 4200;
        }
        M_PUB(&temp_req);
        M_PUB(&calib_req); // stop or start backlight autocalibration
        break;
    }
    default:
        break;
    }
}

```

FedeDP, 2020-08-21:
Hi! 
Can you share a verbose log? 
I don't quite understand if gamma alone isn't changing, or keyboard bl too?

FedeDP, 2020-08-24:
Everything seems working fine from your log:
> (D)[03:56:32]{interface.c:595}	New ScreenSaver inhibition held by 'firefox': 'audio-playing'. Cookie: 1804289383.
(D)[03:56:32]{inhibit.c:62}	ScreenSaver inhibition grabbed.
(I)[03:56:32]{inhibit_bl3.c:48}	Pausing autocalibration and night color.
(D)[03:56:32]{interface.c:434}	Emitting Inhibited property
(I)[03:56:33]{gamma.c:139}	Normal transition to 6500 gamma temp started.
(I)[03:56:33]{backlight.c:365}	Backlight autocalibration disabled.
(D)[03:56:33]{interface.c:434}	Emitting KbdPct property
(D)[03:56:33]{interface.c:434}	Emitting Temp property

Note that when "Normal transition to X gamma temp started" message is printed, it means that Clight already asked Clightd to set X screen temp and it answered ok.

Can you share, in this situation (ie: as soon as your movie starts), output of:
`busctl --user introspect org.clight.clight /org/clight/clight`

Thanks!

FedeDP, 2020-08-24:
I just tried and your module works for me!

NICHOLAS85, 2020-08-24:
Heres three files with the output of that command:
Before playing any video: [before.txt](https://github.com/FedeDP/Clight/files/5116761/before.txt)
Just started playing video: [start.txt](https://github.com/FedeDP/Clight/files/5116755/start.txt)
A couple seconds later while playing video: [finish.txt](https://github.com/FedeDP/Clight/files/5116767/finish.txt)

FedeDP, 2020-08-24:
Did you also rebuild your inhibit_bl3.c custom module, just in case?

NICHOLAS85, 2020-08-24:
Just rebuilt it with the command `gcc -shared -fPIC inhibit_bl3.c -o inhibit_bl3 -Wno-unused` and tested it at both `$XDG_DATA_HOME/clight/modules.d/` and `/usr/share/clight/modules.d` with it being the only custom module loaded and I'm getting the same behavior. Strange issue!

FedeDP, 2020-08-24:
Mmh can you change your module init() func to:
`static void init(void) {
    capture_req.capture.reset_timer = false; // avoid resetting clight internal BACKLIGHT timer
    kbd_bl_req.bl.new = 0.0;
    temp_req.temp.smooth = -1;
    temp_req.temp.daytime = -1;
    
    /* Subscribe to inhibit state */
    M_SUB(INHIBIT_UPD);
}`

FedeDP, 2020-08-24:
About keyboard backlight: a bug was fixed in latest commit; care to test?

NICHOLAS85, 2020-08-24:
Your change from `NIGHT` to `-1` worked which is good. However, I did only want this to run at night because I explicitly set it back to a temp of `4200` once the video stops playing. From my understanding, if I were to leave this change I would see my screen gamma change to 4200 when I finish watching a video during the day as well.

Also, I just tried your latest commit and it doesn't seem to have any effect on the behavior the module is exhibiting. (The backlight stays lit)

NICHOLAS85, 2020-08-24:
Btw thank you very much for assisting me diagnose this issue with my module

FedeDP, 2020-08-24:
> Your change from NIGHT to -1 worked which is good. However, I did only want this to run at night because I explicitly set it back to a temp of 4200 once the video stops playing. From my [...]

Clear, you are right! 
Well, your log seems fine...i don't really understand!

> (I)[03:56:32]{inhibit_bl3.c:48}	Pausing autocalibration and night color.
(I)[03:56:33]{gamma.c:139}	Normal transition to 6500 gamma temp started.
(I)[03:56:33]{backlight.c:365}	Backlight autocalibration disabled.
(D)[03:56:33]{interface.c:434}	Emitting KbdPct property
(D)[03:56:33]{interface.c:434}	Emitting Temp property

As you can see clight is touching both autocalib, kbdpct and temp properties. These logs are only printed if everything went succesfully :(

NICHOLAS85, 2020-08-25:
Ahhh I think I know what the issue was. I've got a script that automates building and installing the deb based on the instructions in the wiki. The install lines were something along the lines of `sudo apt reinstall ./clightd*.deb` which expands to the following: `sudo apt reinstall ./clightd_4.1_amd64.deb ./clightd_4.2_amd64.deb ./clightd_4.3_amd64.deb` and so on as clightd updates. Simply put, apt was installing the wrong version of clightd and clight based on that input, I've got to fix my script!
I've installed the correct, latest, versions manually and the correct gamma adjustment is occurring like expected. However, I am still facing that keyboard backlight issue where it's not being set to 0.

FedeDP, 2020-08-26:
Great to hear!

> However, I am still facing that keyboard backlight issue where it's not being set to 0.

Are you in case using a similar script for clight? Which git version are you using?

NICHOLAS85, 2020-08-26:
I've ditched the script for now and installed this commit of a Clight this morning: c2ecac9
Seems to not adjust keyboard backlight.

FedeDP, 2020-09-08:
I finally had to time to lookup: indeed there was a stupid mistake in keyboard module!
You can update your clight-git and it should be fixed :) (moreover i am finally able to test keyboard backlight!)

closed 2020-09-08 by FedeDP

NICHOLAS85, 2020-09-08:
Ah there we go, latest commit fixed it! Thanks