FedeDP/Clight #143: [BUG]  The clight does not change the screen light

RezaSh13, 2020-08-18:
Hi 
The clight program does not change the screen light
clightd-git r300.b03e52b-1
clight-git r715.767e184-1 

```
```
~ >>> clight                                                           [130]
Wrong AC frames value. Resetting default value.
Wrong BATT frames value. Resetting default value.
Clightd found, version: 4.3.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
Not a laptop device. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Next alarm due to: Wed Aug 19 06:13:00 2020
```

~ >>> clight --verbose                                                      
Wrong AC frames value. Resetting default value.
Wrong BATT frames value. Resetting default value.
Clightd found, version: 4.3.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
Not a laptop device. Killing UPower module.
Failed to retrieve AC state; fallback to ON_AC and OPEN lid.
Emitting AcState property
Set timeout of 0s 1ns on fd 38.
Emitting LidState property
Set timeout of 30s 0ns on fd 38.
Set timeout of 0s 1ns on fd 39.
Next alarm due to: Wed Aug 19 06:13:00 2020
Set timeout of 18819s 0ns on fd 39.
Emitting Sunrise property
Emitting Sunset property
Emitting DayTime property
Emitting NextEvent property
Set timeout of 0s 1ns on fd 40.
Paused as no sensor is available.
Emitting SensorAvail property

```

```

~ >>> cat .config/clight.conf 
####################
# Clight conf file #
####################

## Verbose mode, useful to report bugs:
## run clight in verbose mode,
## then open issue on github attaching log
# verbose = true;

###################
# INHIBITION TOOL #
########################################################
# Requires DIMMER or DPMS enabled. Disabled otherwise. #
########################################################
inhibit:
{
    ## Uncomment to let Clight manage "Docked" laptop state
    ## as an inhibition (pausing DPMS and dimmer).
    ## Note that "Docked" state is only checked when
    ## laptop lid is closed or opened.
    # inhibit_docked = true;
    
[... 221 lines cut ...]
    ##      dark ambient    -> lower monitor backlight level    -> lower monitor light contribution impact       #
    ##      high ambient    -> high monitor backlight level     -> higher monitor light contribution impact      #
    ## We can assume these 2 contributions to zero each other, letting us just compute screen-emitted brightness #
    ## and linearly multiplying it for our guessed ambient-brightness screen contribution value.                 #
    ##############################################################################################################

    ## Uncomment to disable screen-emitted brightness compensation support
    # disabled = true;

    ## Screen timeouts on AC/on BATT.
    ## Set any of these to <= 0 to disable screen-emitted brightness compensation
    ## in the corresponding AC state.
    ## Disabled by default on BATT because it is quite an heavy operation,
    ## as it has to take a snapshot of your X desktop and compute its brightness.
    # timeouts = [ 30, -1 ];

    ## How much does your screen-emitted brightness affect
    ## ambient brightness, in your setup, in percentage.
    # contrib = 0.1;

    ## How many samples should be used to compute average 
    ## screen-emitted brightness.
    # num_samples = 10;
};
```

FedeDP, 2020-08-21:
Hi!
As you can see, your verbose log file is telling you the issue:

> Paused as no sensor is available.

What kind of sensor are you trying to use?

FedeDP, 2020-08-23:
Unfortunately Clight does not support MJPG capture; it only supports either GreyScale or YUYV422.
I will look into supporting MJPEG in Clightd.

RezaSh13, 2020-08-23:
> Unfortunately Clight does not support MJPG capture; it only supports either GreyScale or YUYV422.
> I will look into supporting MJPEG in Clightd.

So now what can I do?It used to work well :|

FedeDP, 2020-08-23:
That is weird! Did clight work fine? On which version?

RezaSh13, 2020-08-23:
> That is weird! Did clight work fine? On which version?

https://github.com/FedeDP/Clight/issues/52
https://github.com/FedeDP/Clight/issues/101
https://github.com/FedeDP/Clight/issues/108
https://github.com/FedeDP/Clight/issues/133

These are my issues

RezaSh13, 2020-08-23:
> That is weird! Did clight work fine? On which version?

This is not just my problem, my friend also has a similar problem, Need more information?

FedeDP, 2020-08-23:
Can your friend run `v4l2-ctl --list-formats` too? Thank you very much!

Clightd did not really bother to check supported video format until recent releases; may be it "worked" for you by using YUYV/Grayscale algorithm on a compressed video format (MJPEG)?
It worked in the sense that it returned some numbers; but it was obviously wrong.
You can see [here](https://github.com/FedeDP/Clightd/commit/0f164e8d385345be6a4512471bb1d05e021bd475#diff-2a4db2376a76772cd7f1b083fc0e0f8cR186) the commit that added the new check on supported video formats; it was part of clightd 4.0 release.

RezaSh13, 2020-08-23:
> Can your friend run `v4l2-ctl --list-formats` too? Thank you very much!
>
> Clightd did not really bother to check supported video format until recent releases; may be it "worked" for you by using [...]
```

ioctl: VIDIOC_ENUM_FMT
  Type: Video Capture

  [0]: 'YUYV' (YUYV 4:2:2)
  [1]: 'MJPG' (Motion-JPEG, compressed)
```

FedeDP, 2020-08-24:
Mmh it should work fine for your friend; can you attach a verbose log for him/her too?

FedeDP, 2020-08-24:
Please note that supported camera pixelformats are found here: https://github.com/FedeDP/Clightd/wiki/Sensors#camera.

RezaSh13, 2020-08-24:
> Mmh it should work fine for your friend; can you attach a verbose log for him/her too?

He said that the screen light does not change and Installed the git version (But he did not restart, only activated the service)

```
clight --verbose                                                            
SCREEN forcefully disabled as Clightd was built without screen support.
Wrong AC frames value. Resetting default value.
Wrong BATT frames value. Resetting default value.
Disabling DPMS in wayland environment.
Clightd found, version: 4.3.
AC curve: y = -0.024825 + 0.191641x + -0.008928x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
AC cable connected.
Laptop lid opened.
Emitting AcState property
Emitting LidState property
35.71 51.41 loaded from cache file.
Loc distance: 35.71,51.41 -> 91.00,181.00 : 3707.34 km.
New location received: 35.71, 51.41.
Set timeout of 0s 1ns on fd 34.
Emitting Location property
Next alarm due to: Mon Aug 24 19:12:00 2020
Set timeout of 4227s 0ns on fd 34.
Emitting Sunrise property
Emitting Sunset property
Set timeout of 0s 1ns on fd 35.
Sensor '/dev/video0' is now available.
Resumed as a sensor is now available.
Emitting DayTime property
Emitting NextEvent property
Emitting SensorAvail property
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.849598.
Ambient brightness: 0.850 -> Backlight pct: 0.959.
Set timeout of 600s 0ns on fd 35.
Emitting AmbientBr property
Emitting BlPct property
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
Backlight level updated: 0.20.
^CReceived 2. Leaving.
Latest location stored in cache file!
```

```
Wrong AC frames value. Resetting default value.
Wrong BATT frames value. Resetting default value.
Disabling DPMS in wayland environment.
Clightd found, version: 4.3.
Failed to create org.freedesktop.ScreenSaver dbus interface: Unknown error -1
Fallback at monitoring requests to org.freedesktop.ScreenSaver name owner.
AC cable disconnected.
Laptop lid opened.
New location received: 35.71, 51.41.
Next alarm due to: Mon Aug 24 19:12:00 2020
Ambient brightness: 0.100 -> Backlight pct: 0.135.
```

FedeDP, 2020-09-08:
Can he update his clight-git and test again?

FedeDP, 2020-10-20:
Hi! I have some good news for you: latest **clightd-git** does now support jpeg webcam captures, thus it should work for you :)
Let me know (i already tested mjpeg capture on my xps btw)!

RezaSh13, 2020-10-26:
> Hi! I have some good news for you: latest **clightd-git** does now support jpeg webcam captures, thus it should work for you :)
> Let me know (i already tested mjpeg capture on my xps btw)!

Hi! This is good news, But I can't test it :((((

closed 2020-10-26 by RezaSh13