FedeDP/Clight #144: [BUG] clight is unable to affect other X displays

loki7990, 2020-08-19:
**Clight version the issue has been seen with:**
> clight 4.1

**Used distribution:**
> Arch linux

**Describe the bug**
I recently started using bumblebee for my optimus laptop after switching from using nouveau/ intel drivers. Apparently, how bumblebee works is that it creates a separate X display (:8 by default) where the nvidia gpu does the rendering. I use intel-virtual-output to create a virtual display in order to use the hdmi port on my laptop to use an external monitor ( the hdmi port is hardwired to the nvidia card, so if I use the hdmi port, the nvidia card starts on display :8). Before I switched to bumblebee, clight 4.1 was able to change the brightness and gamma of the external display in addition to the main display. After the switch, clight was no longer able to affect the external monitor. So far, I've been able to get backlight to work again on the external monitor. It required an extra line added to the Device section in `/etc/bumblebee/xorg.conf.nvidia`. The line was `Option     "RegistryDwords"  "RMUseSwI2c=0x01; RMI2cSpeed=100"` as documented in the man page of ddcutil. It required me to restart the clightd.service after the virtual output was created. This left me with gamma not being adjust during the appropriate changes in the day. From what I have gathered, clight changes the gamma through a similar mechanism as `xrandr --output ... --gamma x:x:x`. On the virtual display created by intel-virtual-output (named VIRTUAL1), it had no effect to changes to the gamma when using the xrandr command or the appropriate busctl command. As a current work around, I've discovered that I can do `DISPLAY=:8 xrandr --output HDMI-0 --gamma 1:0.77:0.59` to get a similar gamma level as the night time temp of 4000 on the external monitor. 

**Expected behavior**
Ideally, clight would be able to detect other X displays besides :0 and apply the appropriate changes to the other displays. I know that in older issues such as https://github.com/FedeDP/Clight/issues/22#issuecomment-365243411 , the busctl command can also intake the DISPLAY and XAUTHORITY values. I've tried using `xauth add :8 . <hex key>` using the hex key from the :0 display, but I keep getting `Failed to parse ':8' as signed 32bit integer: Invalid argument` everytime I try to use a command such as `busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" ":8" "/home/$USER/.lyxauth" 4000` or some derivative of it

**To Reproduce**
Install bumblebee
Setup external display with intel-virtual-output
Try to apply gamma changes
See that the external monitor didn't change its gamma temperature
Be a little bit sad, but determined to get to the bottom of this
Give up, post an issue on github

FedeDP, 2020-08-21:
Hi, thanks for opening a super-detailed issue!
Did you try to run clight with DISPLAY env variable set? eg: `DISPLAY=:8 clight`; it should fix your issue!
Clight uses environment DISPLAY variable; and it is started in 0 display in your case (even if rendering happens on 8 display).

loki7990, 2020-08-23:
I apologize if this is beyond the scope of the project! I understand if you
don't want to rework the code in order to get this to work. I think I can
come up with a better workaround that would require me to kill clight and
then restart it for the external display to allow it to adjust the gamma
before killing it again to restart on the normal monitor. Not exactly
elegant but it would work I think

FedeDP, 2020-08-24:
Or you can just use clightd on :8 display:
`busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)" :8 $YOURXAUTHORITY 4500 false 0 0`
See here for better understanding: https://github.com/FedeDP/Clightd/wiki/Api#orgclightdclightdgamma.

Even better, you could wrote a [custom module](https://github.com/FedeDP/Clight/wiki/Custom-Modules) for clight that waits on TEMP_UPD messages and then calls clightd on :8 to keep temperatures automatically in sync!

loki7990, 2020-08-24:
The custom module sounds like it would be the best option! Unfortunately, I
don't know C for squat :( I'll use that command you gave for now. I assume
it works for both DayTemp and NightTemp?

FedeDP, 2020-08-24:
I'll give this a try (note that this is untested though!)
```
#include <clight/public.h>

CLIGHT_MODULE("SYNCTEMP");

static void init(void) {
    /* Suscribe to daytime updates */
    M_SUB(TEMP_UPD);
}

static void receive(const msg_t *msg, const void *userdata) {
    switch (MSG_TYPE()) {
    case TEMP_UPD: {
        temp_upd *up = (temp_upd *)MSG_DATA();
        char cmd[1024];
        snprintf(cmd, sizeof(cmd), "busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set \"ssi(buu)\" \":8\" \"$YOURXAUTHORITY\" %d %d %d %d", up->new, up->smooth, up->step, up->timeout);
        system(cmd);
        break;
    }
    default:
        break;
    }
}
```

> I assume it works for both DayTemp and NightTemp?

Yes, Clightd does know nothing about Day/Night; it just receives a temperature to be set, and set it.

loki7990, 2020-08-24:
Ah, thank you! I'm a bit uncertain of the install instructions for custom modules; If I understand correctly, I need to put the module as SyncTemp.c into the `src/module/` directory and then build the program with the command `gcc -shared -fPIC myMod.c -o myMod -Wno-unused`? Is there a way I can incorporate this into the PKGBUILD? I'd like to keep clight listed as an installed package. I looked at the PKGBUILD, and the relevant bit seems to be:
```
build() {
    cd "${srcdir}/${_gitname}/build"
    cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Release" \
        ..
    make
}
```
Do I want to put the command like so:
```
build() {
    cd "${srcdir}/${_gitname}/build"
    gcc -shared -fPIC SyncTemp.c -o SyncTemp -Wno-unused
    cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Release" \
        ..
    make
}
```
?

loki7990, 2020-08-24:
Wow, that's awesome! Unfortunately, it seems that only the external monitor changes gamma with the module loaded. This is true for the command `busctl --user set-property org.clight.clight /org/clight/clight/Conf/Gamma org.clight.clight.Conf.Gamma DayTemp "i" 4000` at least. I don't know C, but maybe another line of `snprintf(cmd, sizeof(cmd), "busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set \"ssi(buu)\" \":8\" \"$YOURXAUTHORITY\" %d %d %d %d", up->new, up->smooth, up->step, up->timeout);` but replace `:8` with `:0`?

FedeDP, 2020-08-24:
Wait; you needed to sync gamma from ":0" (that clight already manages) to ":8"; this is what the syncgamma module is doing!
Did I miss something? 

Btw did you replace "$YOURXAUTHORITY" with correct one, right?

loki7990, 2020-08-24:
Yes, that is what I need, the gamma changes to `:0` need to be applied to `:8`. I did miss "$YOURXAUTHORITY"; so I replaced it just now with "/home/barbarossa/.lyxauth" (the output of `echo $XAUTHORITY`). The behavior is the same :( Maybe I need to set it to "$XAUTHORITY"?

FedeDP, 2020-08-24:
Mmh ok; i don't understand what you meant here:
> it seems that only the external monitor changes gamma with the module loaded.

Is this the ":0" display? or ":8" one?

> so I replaced it just now with "/home/barbarossa/.lyxauth" 

That is ok! Did you rebuild the module and overwrite the old one in .local/share/clightd/modules.d/ ?

loki7990, 2020-08-24:
Sorry, the external monitor is the `:8` one. `:8` changes in gamma while `:0` (the primary monitor) does not. Before trying this custom module, it was the other way around: `:0` changed in gamma while `:8` did not. Yes, I did rebuild and replace the module as well as `pkill clight` followed by `setsid -f clight --conf-file=/home/barbarossa/.config/clight.conf &` Maybe I need to restart `clightd`?

FedeDP, 2020-08-24:
Ok i found out why! Clightd does not work on multiple Xdisplays at the same time. Thus, when you call clightd Gamma.Set on ":8", it overwrites ":0", forgetting about it.
Note that if you instead wait ":0" transition to desired temp to finish, you are freely able to change temperature on ":8".

I think you should for now default at calling 
`busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)" :8 $YOURXAUTHORITY 4500 false 0 0`

manually; at least until i manage to fix clightd :)

loki7990, 2020-08-24:
> EDIT: thanks for your super patient testing!

Not a problem! Thanks for taking the time to help figure this out with me
:) Let me know if there's anything I can try in the mean time!

FedeDP, 2020-10-20:
Hi @loki7990 ! Good news!
Latest clightd-git does not support multiple x-displays changes at the same time :) 
Care to test it? Thank you!

loki7990, 2020-10-21:
I have the latest clightd-git installed! Is there a quick and easy command to test with? My scripts are currently setup to work around the current problem and I don't want to wait till dusk to report back the results

FedeDP, 2020-10-21:
Hi!
You can call:
`busctl call org.clightd.clightd /org/clightd/clightd/Gamma org.clightd.clightd.Gamma Set "ssi(buu)"  ":$DISPLAY" $XAUTH 4500 1 50 25`
twice, with ":0" and ":8" in place of $DISPLAY, and your Xauthority file in place of $XAUTH.

You should then see both Xdisplays start smoothly going to a lower (hotter) temperature.

loki7990, 2020-10-21:
It worked!! Thank you so much!!

FedeDP, 2020-10-21:
Huge!! Care to post your final custom module? So i can store it in wiki pages (or maybe even ship it as a skeleton!)
Thank you very much for your testing!

closed 2020-10-21 by FedeDP