FedeDP/Clight #146: [BUG] Gamma not corrected after wake from suspend/hibernate

NICHOLAS85, 2020-08-27:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Clight Git: 4a319eee115e3ba15f69e3396164f77cb0bf942b
**Used distribution:**
> Bedrock Linux 0.7.18beta3 (Poki), Debian Testing stratum

**Describe the bug**
When I open my laptop in the day after having suspended-then-hibernated the night before, the gamma level does not change to daytime levels. In the log I can see Clight attempted to do so without error but the change isn't exhibited until I kill and restart Clight.

**Expected behavior**
Clight should be able to adjust the gamma of the screen after waking from suspend/hibernate.

**To Reproduce**
Set Clight to a different gamma level during the NIGHT state, suspend, and then wake laptop up during the DAY state.

Please attach a Clight verbose log too!
[clight.log.txt](https://github.com/FedeDP/Clight/files/5138574/clight.log.txt)

FedeDP, 2020-09-08:
I can say that it works fine for me.
Maybe you are having some weird script that, eg: on suspend switch to a tty and on resume, switches back to X server?
It once happened to me: suspend got broken in a kernel, and to workaround it i wrote that systemd sleep script.
But clight was too fast to execute its gamma temperature setting (it happened while session was on tty still) and it had no effect.

FedeDP, 2020-10-31:
I could reproduce the issue. As i thought, it seems like a timing issue: clight is too quick to ask clightd to change screen temperature, and at that moment X is still not resumed, thus clightd fails.
I will delay display temperature adjustments of 5s to avoid these issues.

NICHOLAS85, 2020-10-31:
Ah it's good to see a delay could potentially remedy this issue. The only software I am aware of that affected the X server besides whatever my base kde installation does was https://github.com/phillipberndt/autorandr. I disabled the software to attempt to see if the behavior improved but it seemed to be sporadic. Once the delay is upstreamed I'll let you know if it fixes the issue.

FedeDP, 2020-11-01:
The work is now in master. Note that you also need new clightd-master! 
Clightd has seens lots of improvements lately, nearing next release (that i plan before end of year, as i lately often do! I like xmas presents).

FedeDP, 2020-11-01:
I may let users customize the behaviour, eg: customizing the timeout for gammas between 0 (no timeout) and eg: 30s.

FedeDP, 2020-11-01:
Ok, i added a new conf option under GAMMA: `delay`.
This should do the trick! (for me too, as my XPS takes ages to resume from suspend causing the same issue!)

NICHOLAS85, 2020-11-02:
Updated Clight and Clightd to the latest commits and set the delay to 5s. I'll see if gamma properly adjusts in the morning! Also I see all the Wayland support that's coming, exciting to see you are able to implement some of Clights features for Wayland users.

NICHOLAS85, 2020-11-02:
Looks like it worked well today, I'll reopen if I run into any issues. Thanks!

FedeDP, 2020-11-02:
Great to hear!

NICHOLAS85, 2020-11-06:
Updating this, so far the delay has been working and I've had a normal screen temp after waking my laptop!

On another semi-related note: I've noticed that if I log out and log in while clight has set my gamma to a lower level (light at night), upon logging in the gamma resets to its default and does not correct. I can send a signal to change the gamma which works, clight just doesn't remember to change its gamma upon logging in.

FedeDP, 2020-11-06:
Mmmh when you logout clight should be stopped and restarted when you login again.
If that's not the case, please open a new issue! Thank you!

FedeDP, 2020-11-06:
Ok i found https://fedoraproject.org/wiki/Changes/KillUserProcesses_by_default.
I was not aware; on my distro (archlinux) the default is `KillUserProcesses=no` in /etc/default/clight.conf.
So, clight gets not killed upon user exit or system poweroff.
I will add this information to the wiki, with recommendation to turn on the flag.

NICHOLAS85, 2020-11-06:
Ahh, that might explain it. I'll try running with `KillUserProcesses=yes` for awhile and see if I I like the changed behavior on my system.

NICHOLAS85, 2020-11-07:
Looks like with 7e6bcd9180244defb614a67ca226347eb79c17a5, `KillUserProcesses=yes`, and an adequate gamma delay, it's all working flawlessly. Thanks again!

FedeDP, 2020-11-12:
It seems like other things fail if resume from suspend is quite slow, namely BACKLIGHT Calibration and SCREEN GetBrightness.
I may add instead an option to delay clight on lid open, eg: sleep 5s when lid gets opened, instead of this GAMMA only delay.
What do you think?

NICHOLAS85, 2020-11-12:
That might be a good idea, I have noticed BACKLIGHT and SCREEN errors in the log sometimes after wake. It would be great if Clight was able to operate after the desktop has completely set up, without a set timeout, but I see many programs run into the same issue which makes me think it's not an easy or possible task to do. Adding an overall delay seems like the next best thing.
A possible idea is also to let Clight attempt again a couple seconds later after a failed BACKLIGHT calibration or failed SCREEN GetBrightness. If it fails again stop trying like the current behavior.

NICHOLAS85, 2020-11-12:
Also I wish I could contribute more to the codebase directly but I lack the C knowledge needed to do so, I hope my feedback via issues is helpful!

FedeDP, 2020-11-12:
> I hope my feedback via issues is helpful!

Super helpful, thank you!!

Anyway, i'm looking for a way to actually know that user graphical session has been fully resumed from suspension/hibernation... seems like there is no way though...

> A possible idea is also to let Clight attempt again a couple seconds later after a failed BACKLIGHT calibration or failed SCREEN GetBrightness. If it fails again stop trying like the current behavior.

This is fine, even if i don't really like the complexity added by that!
But, unfortunately, it seems like LID_OPENED is not the first signal that clight receives upon wake-up; so my designed solution won't work.
I'd really like to work this out in an event-driven manner, just like rest of clight does...

FedeDP, 2020-11-28:
Hi! Now that #166 is merged, gamma "delay" options has been dropped in favour of a "resumedelay" option, that will indeed wait N seconds before resuming clight features after resume from sleep.
In my case, the new code does indeed work even without a resumedelay on my dell XPS as it now waits upon systemd `PrepareForSleep false` signal.

closed 2020-11-28 by FedeDP

NICHOLAS85, 2020-11-29:
Great! I've rebuilt and reinstalled Clight and will test it out without setting `resumedelay` and see if the new systemd PrepareForSleep code works as expected.

FedeDP, 2020-11-29:
Fingercrossed!

NICHOLAS85, 2020-12-04:
So far Clight has been working perfectly! Gamma is correctly being adjusted after suspend and from reading the logs it seems everything is pausing and resuming correctly. Didn't even need to use the `resumedelay` option. I also noticed you pushed an update to fix the geoclue spamming logs issue awhile ago and that also worked correctly, no more spam.