FedeDP/Clight #147: Clight don’t change backlight

lukas1760, 2020-09-11:
Hi, I tried to install and run clight builded in January on RPI4 running on Buster. Installation without problems. Everything seems to be working but backlight is without change. After that I tried same packages on RPi3 with Buster and backlight is changing properly. Any ideas? Thanks

FedeDP, 2020-09-13:
Hi! Can you show me any log from both rpi3 and rpi4? :) 
Anyway, i'd upgrade to latest clightd (4.2) and clight (4.1)!

lukas1760, 2020-09-13:
Hi! As you wish :)
I had problem with building of latest clight package so i decide to use older created package.
[clight-rpi4-5.4.51-v7l+.log](https://github.com/FedeDP/Clight/files/5214002/clight-rpi4-5.4.51-v7l%2B.log)
[clight-rpi3-4.19.97-v7+.log](https://github.com/FedeDP/Clight/files/5214003/clight-rpi3-4.19.97-v7%2B.log)

lukas1760, 2020-09-13:
Building of latest Clight stucks on:
```
pi@raspberrypi:~/build_test/Clight/build $ make
Scanning dependencies of target clight
[  3%] Building C object CMakeFiles/clight.dir/src/conf/config.c.o
[  7%] Building C object CMakeFiles/clight.dir/src/conf/opts.c.o
[ 11%] Building C object CMakeFiles/clight.dir/src/main.c.o
[ 15%] Building C object CMakeFiles/clight.dir/src/modules/backlight.c.o
[ 19%] Building C object CMakeFiles/clight.dir/src/modules/bus.c.o
/home/pi/build_test/Clight/src/modules/bus.c: In function \u2018_call\u2019:
/home/pi/build_test/Clight/src/modules/bus.c:84:9: error: used struct type value where scalar is required
     if (args_va) {
         ^~~~~~~
/home/pi/build_test/Clight/src/modules/bus.c: In function \u2018call\u2019:
/home/pi/build_test/Clight/src/modules/bus.c:125:33: error: incompatible type for argument 3 of \u2018_call\u2019
         r = _call(a, signature, NULL, NULL, a->reply_cb != NULL);
                                 ^~~~
/home/pi/build_test/Clight/src/modules/bus.c:69:68: note: expected \u2018va_list\u2019 {aka \u2018__va_list\u2019} but argument is of type \u2018void *\u2019
 static int _call(const bus_args *a, const char *signature, va_list args_va, const void **args_ptr, bool expect_reply) {
                                                            ~~~~~~~~^~~~~~~
make[2]: *** [CMakeFiles/clight.dir/build.make:115: CMakeFiles/clight.dir/src/modules/bus.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/clight.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

```

FedeDP, 2020-09-13:
You are right! Latest commit fixed the build for you :)

Btw both logs seems equal.
Only issue i see is that ambient brightness is always:
```
(D)[23:38:57]{backlight.c:320}	Captured ambient brightness: 0.044000.
(I)[23:38:57]{backlight.c:250}	Ambient brightness: 0.044 -> Backlight pct: 0.058.
```
in both cases...

lukas1760, 2020-09-13:
Build is done! But still not changing brightness :( 

> Only issue i see is that ambient brightness is always:

it's because i had wrong time in one of RPI.

lukas1760, 2020-09-14:
I found this on ddcutil page:

*ddcutil does not work on the Raspberry Pi 4.* The HDMI related chip has
changed, and drivers have not yet been revised to enable userspace access
to /dev/i2c-2. This does not appear to be a high priority for the Pi
developers. User bangom has explored the problem in depth, and does have a
workaround that entails modifying the HDMI cable. See ddcutil issue #97 on
github <https://github.com/rockowitz/ddcutil/issues/97>. For a detailed
discussion of the problem, and bangom's hardware workaround, see Issue
#3152 on the github raspberrypi/linux project
<https://github.com/raspberrypi/linux/issues/3152>.

So ddcutil is not able to work  on RPI4 at this moment. Maybe in the
future...Anyway...many thanks for your help ;-)

closed 2020-09-14 by lukas1760

FedeDP, 2020-09-14:
> EDIT: After kernel update to 5.4.64-xxx is finally working!!!

THAT IS HUGE! :)

Balaji-Rahunathan, 2020-10-04:
> > EDIT: After kernel update to 5.4.64-xxx is finally working!!!
> 
> THAT IS HUGE! :)

i updated to 5.4.69 but still it doesn't work

FedeDP, 2020-10-06:
@Balaji-Rahunathan feel free to open a new issue!