FedeDP/Clight #148: [FEATURE REQ] Disable Dimmer when Watching Videos

hagen-eriksen, 2020-09-14:
**Is your feature request related to a problem? Please describe.**
Screen turns black after timeout when watching videos (on a web browser for example or using VLC).

**Describe the solution you'd like**
Dimmer is disabled when watching a video (web browser, VLC, etc).

**Describe alternatives you've considered**
- Not using clight (but damn, clight is just too useful for that!)
- Writing a shell program that stops clight then waits. When the program is killed, it starts clight again (still requires manual intervention to start the clight inhibitor before watching a video)

NICHOLAS85, 2020-09-14:
You might be able to do what you want with a custom module. Checkout this one: https://github.com/FedeDP/Clight/wiki/Custom-Modules#inhibit-backlight

FedeDP, 2020-09-14:
Hi!
Clight does already support that, automatically: https://github.com/FedeDP/Clight/wiki/ScreenSaver-Specification

What DE are you using?

hagen-eriksen, 2020-09-14:
That is very interesting!

I use i3 (manjaro i3 community version). Thanks for pointing out that inhibiting clight is already implemented! However, right-clicking a `.desktop` file is not really a thing when using rofi as launcher...

So you mean that this is somehow related to the DE (or lake thereof in this case)? I am missing the signals altogether or are they captured by other programs before clight can process them?

FedeDP, 2020-09-14:
Better still: without a real DE, clight is free to work as org.freedesktop.ScreenSaver interface owner (on DEs instead, it has to continuously monitor dbus for calls with destination org.freedesktop.ScreenSaver to intercepth them).
Can you provide me a verbose log for clight?

hagen-eriksen, 2020-09-17:
Do you mean the log file?

```
(W)[11:09:53]{opts.c:421}       DPMS BATT timeout: wrong value (<= dimmer timeout). Resetting default value.

Clight
* Software version:             4.1
* Global config dir:            /etc/default
* Global data dir:              /usr/share/clight
* Starting time:                Mon Sep 14 11:09:53 2020

Starting options:

### GENERIC ###
* Verbose (debug):              Disabled

### BACKLIGHT ###
* Smooth trans:         Enabled
* Smooth steps:         0.05
* Smooth timeout:               30
* Daily timeouts:               AC 600  BATT 600
* Nightly timeout:              AC 2700 BATT 2700
* Event timeouts:               AC 300  BATT 300
* Backlight path:               Unset
* Shutter threshold:            0.10
* Autocalibration:              Enabled
* Pause on lid closed:          Enabled

### SENSOR ###
* Captures:             AC 5    BATT 5
* Device:               Unset
* Settings:             Unset

### GAMMA ###
* Smooth trans:         Enabled
* Smooth steps:         50
* Smooth timeout:               300
* Daily screen temp:            6500
* Nightly screen temp:          4000
* Long transition:              Disabled
* Ambient gamma:                Disabled

### DAYTIME ###
* User position:                46.16   6.12
* User set sunrise:             Unset
* User set sunset:              Unset
* Event duration:               1800

### DIMMER ###
* Smooth trans:         ENTER: Enabled, EXIT: Enabled
* Smooth steps:         ENTER: 0.05, EXIT: 0.05
* Smooth timeout:               ENTER: 30, EXIT: 30
* Timeouts:             AC 300  BATT 300
* Backlight pct:                0.20

### DPMS ###
* Timeouts:             AC 900  BATT 300

### SCREEN ###
* Timeouts:             AC 30   BATT -1
* Contrib:              0.10
* Samples:              10

### INHIBIT ###
* Docked:               Disabled
* PowerManagement:              Disabled

(I)[11:09:53]{main.c:134}       Clightd found, version: 4.2.
(I)[11:09:53]{upower.c:52}      AC cable connected.
(I)[11:09:53]{upower.c:63}      Laptop lid opened.
[... 225 lines cut ...]
(I)[20:35:22]{backlight.c:309}  Ambient brightness: 0.103 -> Backlight pct: 0.140.
(I)[21:20:24]{backlight.c:309}  Ambient brightness: 0.111 -> Backlight pct: 0.152.
(I)[22:05:25]{backlight.c:314}  Ambient brightness: 0.078 -> Clogged capture detected.
(I)[22:19:55]{upower.c:63}      Laptop lid closed.
(I)[05:59:59]{upower.c:63}      Laptop lid opened.
(I)[05:59:59]{backlight.c:314}  Ambient brightness: 0.000 -> Clogged capture detected.
(I)[06:45:00]{backlight.c:309}  Ambient brightness: 0.191 -> Backlight pct: 0.256.
(I)[06:46:00]{daytime.c:174}    Next alarm due to: Thu Sep 17 07:16:00 2020
(I)[06:46:00]{gamma.c:133}      Normal transition to 4000 gamma temp started.
(I)[06:50:01]{backlight.c:309}  Ambient brightness: 0.585 -> Backlight pct: 0.640.
(I)[06:55:02]{backlight.c:309}  Ambient brightness: 0.428 -> Backlight pct: 0.515.
(I)[07:00:03]{backlight.c:309}  Ambient brightness: 0.637 -> Backlight pct: 0.674.
(I)[07:05:04]{backlight.c:309}  Ambient brightness: 0.842 -> Backlight pct: 0.768.
(I)[07:10:05]{backlight.c:309}  Ambient brightness: 0.916 -> Backlight pct: 0.787.
(I)[07:15:06]{backlight.c:309}  Ambient brightness: 0.915 -> Backlight pct: 0.787.
(I)[07:16:00]{daytime.c:174}    Next alarm due to: Thu Sep 17 07:46:00 2020
(I)[07:16:00]{gamma.c:133}      Normal transition to 6500 gamma temp started.
(I)[07:16:01]{backlight.c:309}  Ambient brightness: 0.917 -> Backlight pct: 0.788.
(I)[07:21:02]{backlight.c:309}  Ambient brightness: 0.893 -> Backlight pct: 0.782.
(I)[07:46:00]{daytime.c:174}    Next alarm due to: Thu Sep 17 19:12:00 2020
(I)[07:46:00]{gamma.c:133}      Normal transition to 6500 gamma temp started.
(I)[07:51:47]{backlight.c:309}  Ambient brightness: 0.917 -> Backlight pct: 0.788.
(I)[08:01:48]{backlight.c:309}  Ambient brightness: 0.916 -> Backlight pct: 0.787.
(I)[08:11:48]{backlight.c:309}  Ambient brightness: 0.916 -> Backlight pct: 0.787.
```

FedeDP, 2020-09-20:
Thanks! Well i can't see any error in your log file.
We should then check if vlc/browser is sending an org.freedesktop.ScreenSaver Inhibit call.
Can you run `busctl --user monitor --match destination=org.freedesktop.ScreenSaver,interface=org.freedesktop.ScreenSaver` in your terminal, and then open up a video with vlc?
After some seconds, you should see in your terminal that a line similar to:
```
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=9
Sender=:1.68  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
UniqueName=:1.68
MESSAGE "ss" {
STRING "vlc";
STRING "Riproduzione di alcuni media.";
};
```
has appeared.

hagen-eriksen, 2020-09-20:
Thanks a lot for helping me out! I ran the command then opened videos on different websites with Firefox and Vivaldi then tried VLC and got nothing printed in the terminal.

Would you have any clue as to where to look to solve this issue? Am I missing some kind of deamon?
By the way, I run manjaro i3 (community edition), installed two weeks ago.

FedeDP, 2020-09-20:
Wow that is weird (why does it always go this way?? :D )
Let's try this one, to check if there is any dbus service providing org.freedesktop.ScreenSaver interface; we should see clight here:
`busctl --user tree | grep ScreenSaver`

I will investigate vlc and firefox code further; as [noted in the wiki](https://github.com/FedeDP/Clight/wiki/ScreenSaver-Specification#chromiumchrome-notes) chromium had a mis-implementation of screensaver inhibition in the past that caused screensaver inhibition to only be emitted on gnome, kde and xfce.

In any case, you can help me trying to run vlc from terminal with verbose option: `-vv` and paste here its full output after opening a video file.
Thanks!

hagen-eriksen, 2020-09-21:
Thanks for your quick reply, your help is very much appreciated!
I tried both `busctl --user tree | grep ScreenSaver` and `busctl --user tree | grep Inhibit` and got nothing in return.

I am quite unfamiliar with the inner workings of D-Bus. I read some high-level introduction to it but how can I add the `ScreenSaver` interface? Do I miss a deamon, is it something else that I should do?

NB: Not sure if this is related but I currently have a very partial support for automated power management. I basically only use clight (so not screenlocking after timeout, no sleep after timeout, etc.), though I want to add those shortly

hagen-eriksen, 2020-09-21:
Weird, I just ran the command again and got the following output:
```bash
├─/ScreenSaver
    └─/org/freedesktop/ScreenSaver
Service org.freedesktop.ScreenSaver:
├─/ScreenSaver
    └─/org/freedesktop/ScreenSaver
```

hagen-eriksen, 2020-09-21:
The full tree blocks of the above outputs were the following:
```
Service org.clight.clight:
├─/ScreenSaver
└─/org
  ├─/org/clight
  │ └─/org/clight/clight
  │   └─/org/clight/clight/Conf
  │     ├─/org/clight/clight/Conf/Backlight
  │     ├─/org/clight/clight/Conf/Daytime
  │     ├─/org/clight/clight/Conf/Dimmer
  │     ├─/org/clight/clight/Conf/Dpms
  │     ├─/org/clight/clight/Conf/Gamma
  │     ├─/org/clight/clight/Conf/Inhibit
  │     ├─/org/clight/clight/Conf/Screen
  │     └─/org/clight/clight/Conf/Sensor
  └─/org/freedesktop
    └─/org/freedesktop/ScreenSaver
```
And
```
Service org.freedesktop.ScreenSaver:
├─/ScreenSaver
└─/org
  ├─/org/clight
  │ └─/org/clight/clight
  │   └─/org/clight/clight/Conf
  │     ├─/org/clight/clight/Conf/Backlight
  │     ├─/org/clight/clight/Conf/Daytime
  │     ├─/org/clight/clight/Conf/Dimmer
  │     ├─/org/clight/clight/Conf/Dpms
  │     ├─/org/clight/clight/Conf/Gamma
  │     ├─/org/clight/clight/Conf/Inhibit
  │     ├─/org/clight/clight/Conf/Screen
  │     └─/org/clight/clight/Conf/Sensor
  └─/org/freedesktop
    └─/org/freedesktop/ScreenSaver
```

hagen-eriksen, 2020-09-21:
I tested the command you gave me above (`busctl --user monitor --match destination=org.freedesktop.ScreenSaver,interface=org.freedesktop.ScreenSaver`)

This time the output matched what you expected:
```
Monitoring bus message stream.
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=5
  Sender=:1.265001  Destination=org.freedesktop.ScreenSaver  Path=/org/freedesktop/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.265001
  MESSAGE "ss" {
          STRING "/usr/bin/vivaldi-stable";
          STRING "Video Wake Lock";
  };
```

The D-Bus events were fired both for videos on web browser and for vlc...

Now I am very confused... What did I do to fix the issue?

FedeDP, 2020-09-21:
> The full tree blocks of the above outputs were the following:

Ok, this means clight is exposing org.freedesktop.ScreenSaver interface fine. Did you start clight in between your first run and this one?

> This time the output matched what you expected:

Great! This means clight is now working fine!

If Clight was already started the first time you ran `busctl --user tree | grep ScreenSaver`, for some reason it did not actually bind org.freedekstop.ScreenSaver dbus service; it would be weird as clight does log any error though.

hagen-eriksen, 2020-09-29:
Sorry for not answering earlier, my computer's battery broke down so I had to fix this first...

Anyway, back to business. I guess I did something wrong when starting clight the first time. I now improved the i3 startup procedure in which I start clight so that it kills all previously running clight processes if any, then starts a new one (there might already be a running clight process if I am restarting i3 in place). It seemed to work fine when I restarted i3 before the battery stopped working.

Thanks a lot for your kind and very quick help @FedeDP ! It made me explore D-Bus and also understand even more (can you believe it) why clight is so powerful. I now send screen backlight commands from my keyboard to clight directly.

closed 2020-09-29 by hagen-eriksen

FedeDP, 2020-09-29:
> Thanks a lot for your kind and very quick help @FedeDP ! It made me explore D-Bus and also understand even more (can you believe it) why clight is so powerful. I now send screen backlight commands from my keyboard to clight directly.

You are warmly welcome!! It's great to hear you are enjoying clight!

hagen-eriksen, 2020-10-06:
I was wondering if I could abuse the help of this project... I still have issues with the `ScreenSaver` interface :sob: 

I monitored yesterday the calls to the interface when watching a long vide on Firefox. And firefox does indeed inhibit the ScreenSaver. But some program then sent the uninhibit request. How can I learn what program it is that sent the uninhibit request?

Here are the logs:
```
busctl --user monitor --match destination=org.freedesktop.ScreenSaver,interface=org.freedesktop.ScreenSaver
Monitoring bus message stream.
‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=611
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.126484
  MESSAGE "ss" {
          STRING "firefox";
          STRING "video-playing";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=612
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.126484
  MESSAGE "ss" {
          STRING "firefox";
          STRING "audio-playing";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=613
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=UnInhibit
  UniqueName=:1.126484
  MESSAGE "u" {
          UINT32 1129566413;
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=614
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=Inhibit
  UniqueName=:1.126484
  MESSAGE "ss" {
          STRING "firefox";
          STRING "audio-playing";
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=615
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=UnInhibit
  UniqueName=:1.126484
  MESSAGE "u" {
          UINT32 1129566413;
  };

‣ Type=method_call  Endian=l  Flags=0  Version=1 Cookie=616
  Sender=:1.126484  Destination=org.freedesktop.ScreenSaver  Path=/ScreenSaver  Interface=org.freedesktop.ScreenSaver  Member=UnInhibit
  UniqueName=:1.126484
  MESSAGE "u" {
          UINT32 1129566413;
  };
```

FedeDP, 2020-10-08:
> I was wondering if I could abuse the help of this project... I still have issues with the ScreenSaver interface sob

You absolutely can!
That is weird: something is triggering 3 UnInhibit with same uint cookie; let's find the stupid monkey!
You can take note of `Sender=:xxxx` and find it in `busctl --user list` :)

mirfatif, 2021-04-27:
@FedeDP dimmer is disabled/paused when watching videos. Is it possible to achieve the same for calibration?

FedeDP, 2021-04-28:
Hi @mirfatif! Yes, you can; you have to use a custom module though: https://github.com/FedeDP/Clight/wiki/Custom-Modules#inhibit-backlight 
Let me know if you need some help (that wiki page should cover everything is needed for custom modules though!)

mirfatif, 2021-04-28:
> Hi @mirfatif! Yes, you can; you have to use a custom module though: https://github.com/FedeDP/Clight/wiki/Custom-Modules#inhibit-backlight
> Let me know if you need some help (that wiki page should cover everything is needed for custom modules though!)

It works. Fantabulous! Thanks a lot.

FedeDP, 2021-05-01:
Hi @mirfatif 
note that, as this is an highly requested feature, i added back it into clight in latest -git commit!
You should look for "inhibit_bl" conf option ;) 
Let me know if everything works fine for you!