FedeDP/Clight #149: Failed to set property NoAutoCalib

Sonico98, 2020-09-20:
**Clightd version:**
> 4.3

**Used distribution:**
> Arch Linux, i3

**Describe the bug**
Trying to run the command `busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" true` results in the error

`Failed to set property NoAutoCalib on interface org.clight.clight.Conf: Unknown interface org.clight.clight.Conf or property NoAutoCalib.`

I've tried using the AUR packages `clight` and `clight-git` (and the clightd ones too), as well as cloning the repo and installing things manually

**Expected behavior**
The property NoAutoCalib exists, allowing to enable or disable automatic brightness calibration.

**To Reproduce**
Trying to execute the command above.

**Please attach a Clight verbose log too!**
I doubt this one will help but here it is:

> Clightd found, version: 4.3.
AC curve: y = 0.030769 + 0.004154x + 0.004476x^2
BATT curve: y = -0.010629 + 0.153844x + -0.007284x^2
AC cable connected.
Laptop lid opened.
Emitting AcState property
Set timeout of 0s 1ns on fd 36.
Emitting LidState property
Set timeout of 0s 1ns on fd 37.
Set timeout of 30s 0ns on fd 36.
Next alarm due to: Sun Sep 20 19:00:00 2020
Set timeout of 9485s 0ns on fd 37.
Emitting Sunrise property
Failed to validate inhibit request.
Emitting Sunset property
Emitting DayTime property
Normal transition to 6500 gamma temp started.
Emitting NextEvent property
Emitting Temp property
Set timeout of 0s 1ns on fd 38.
Sensor '/dev/video0' is now available.
Resumed as a sensor is now available.
Emitting SensorAvail property
Captured [5/5] from '/dev/video0'. Ambient brightness: 0.937179.
Ambient brightness: 0.937 -> Backlight pct: 0.463.
Set timeout of 600s 0ns on fd 38.
Emitting AmbientBr property
Emitting BlPct property
Backlight level updated: 0.46.
Set timeout of 30s 0ns on fd 36.
Set timeout of 30s 0ns on fd 36.

FedeDP, 2020-09-20:
Hi! Thanks for opening this issue.
NoAutoCalib property is indeed exposed on [org.clight.clight.Backlight](https://github.com/FedeDP/Clight/wiki/Api#orgclightclightconfbacklight) interface; you should change your command with:
`busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf/Backlight org.clight.clight.Conf.Backlight NoAutoCalib "b" true`

Sonico98, 2020-09-21:
Thanks for the answer! Indeed, using that command works. Could it be that the desktop file is wrong then? As what I did was copy the command from the file at `/usr/share/applications/clightc.desktop`

> [Desktop Action PauseCalib]
Exec=busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" true
Name=Pause backlight calibration

> [Desktop Action ResumeCalib]
Exec=busctl --expect-reply=false --user set-property org.clight.clight /org/clight/clight/Conf org.clight.clight.Conf NoAutoCalib "b" false
Name=Resume backlight calibration

Also, please excuse my lack of knowledge using dbus! This should've been easy to notice...

FedeDP, 2020-09-21:
> Could it be that the desktop file is wrong then?

Yeah it can surely be! Would you be so nice to open a PR to fix it? You can find the desktop file in Extra/desktop/clightc.desktop :)

> Also, please excuse my lack of knowledge using dbus! This should've been easy to notice...

Thank you for taking the time to open this issue and looking into it, indeed!

Sonico98, 2020-09-22:
> Yeah it can surely be! Would you be so nice to open a PR to fix it? You can find the desktop file in Extra/desktop/clightc.desktop :)

Done! Hope it's fine, this was my first time submitting a PR and was so nervous, I hope I did nothing wrong. Thanks for the help! I guess this can be closed now.

closed 2020-09-22 by Sonico98