FedeDP/Clight #151: [BUG] Failed to validate location request.

NICHOLAS85, 2020-10-01:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> Commit b637ca7

**Used distribution:**
> Bedrock Linux 0.7.18beta3 (Poki), Debian Testing stratum

**Describe the bug**
This is more of a question than a bug report. I've switched from manually setting a latitude and longitude to allowing Clight to receive location data from Geoclue. In my logs, however, I sometimes see around ~2 of the following messages per second (w,x,y, and z replaced with real coordinates):
```
(D)[13:49:07]{location.c:172}   w x received from Geoclue2.
(D)[13:49:07]{my_math.c:191}    Loc distance: w,x -> y,z : 5.73 km.
(D)[13:49:07]{validations.c:11} Failed to validate location request.
```
Is this expected behavior? Should I have any concerns about battery usage with Clight receiving so many events from Geoclue?

**To Reproduce**
Comment out latitude, longitude, sunrise, and sunset settings under daytime

Please attach a Clight verbose log too!

FedeDP, 2020-10-01:
The output is normal, it means that clight received a new location from geoclue that is "too near" your current clight location, thus it would be useless to compute daytime events (sunrise/sunset) again.

It is very weird that you are spammed... Which version of geoclue are you on? It is most probably not a clight fault this time :D

NICHOLAS85, 2020-10-01:
Ahh makes sense to ignore further computations, the "failed to validate" can come off as more serious by just looking at the logs.

Geoclue's version is `2.5.6-1`

closed 2020-10-01 by NICHOLAS85