FedeDP/Clight #152: Clight not working with Fedora 33

perceival, 2020-10-05:
**NOTE:** Please try to upgrade Clight if not running on latest release, before reporting a bug!

**Clight version the issue has been seen with:**
> 4.2.

**Used distribution:**
> Fedora 33 with GNOME 3.38

**Describe the bug**
I was using on the same laptop clight under Manjaro. Since migrated to Fedora 33 I am no longer able to get it working. Config files should have been default ones in both cases.
Tried in addition devname = "video0"; and devname = "video1"; options without success.
I would appreciate any hints.

Please attach a Clight verbose log too!
[clight.log](https://github.com/FedeDP/Clight/files/5329553/clight.log)
[journalctl.txt](https://github.com/FedeDP/Clight/files/5329555/journalctl.txt)

FedeDP, 2020-10-08:
Hi, thanks for opening this issue!
I see 
> (D)[21:03:38]{backlight.c:457}	Paused as no sensor is available.

Line in clight log file; moreover i see
> Oct 05 21:03:38 snail audit[6994]: AVC avc:  denied  { connectto } for  pid=6994 comm="clightd" path=002F746D702F2E5831312D756E69782F5830 scontext=system_u:system_r:init_t:s0 tcontext=unconfined_u:unconfined_r:xserver_t:s0-s0:c0.c1023 tclass=unix_stream_socket permissive=0
Oct 05 21:03:38 snail audit[6994]: AVC avc:  denied  { read write } for  pid=6994 comm="clightd" name="video0" dev="devtmpfs" ino=26880 scontext=system_u:system_r:init_t:s0 tcontext=system_u:object_r:v4l_device_t:s0 tclass=chr_file permissive=0

in clightd log; it seems like SELinux is denying Clightd access to video0 (user camera).
Are you running clightd as root user?

perceival, 2020-10-09:
Thank you for your quick response!

On one laptop that is connected to a TV set I guess dimming is not going to work.
On the other I have followed https://github.com/FedeDP/Clightd/wiki/SELinux instruction but still something is preventing dimming from working. Is Wayland a reason?

```
Clight
* Software version:		4.2
* Global config dir:		/etc/default
* Global data dir:		/usr/share/clight
* Starting time:		Fri Oct  9 12:12:09 2020

Starting options:

### GENERIC ###
* Verbose (debug):		Enabled

### BACKLIGHT ###
* Smooth trans:		Enabled
* Smooth steps:		0.05
* Smooth timeout:		30
* Daily timeouts:		AC 600	BATT 1200
* Nightly timeout:		AC 2700	BATT 5400
* Event timeouts:		AC 300	BATT 600
* Backlight path:		Unset
* Shutter threshold:		0.00
* Autocalibration:		Enabled
* Pause on lid closed:		Disabled

### SENSOR ###
* Captures:		AC 5	BATT 5
* Device:		video1
* Settings:		Unset

### KEYBOARD ###
* Dim:		Disabled
* Threshold:		1.00

### GAMMA ###
* Smooth trans:		Enabled
* Smooth steps:		50
* Smooth timeout:		300
* Daily screen temp:		6500
* Nightly screen temp:		4000
* Long transition:		Disabled
* Ambient gamma:		Disabled

### DAYTIME ###
* User position:		Unset
* User set sunrise:		Unset
* User set sunset:		Unset
* Event duration:		1800

### DIMMER ###
* Smooth trans:		ENTER: Enabled, EXIT: Enabled
* Smooth steps:		ENTER: 0.05, EXIT: 0.05
* Smooth timeout:		ENTER: 30, EXIT: 30
* Timeouts:		AC 45	BATT 20
* Backlight pct:		0.20

### SCREEN ###
* Timeouts:		AC 30	BATT -1
* Contrib:		0.10
* Samples:		10

### INHIBIT ###
* Docked:		Disabled
* PowerManagement:		Disabled

(I)[12:12:09]{main.c:134}	Clightd found, version: 4.3.
[... 41 lines cut ...]
Oct 09 12:12:04 snail audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Oct 09 12:12:04 snail systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Oct 09 12:12:04 snail systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
Oct 09 12:12:04 snail audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Oct 09 12:12:09 snail clightd[69263]: X Error of failed request:  BadMatch (invalid parameter attributes)
Oct 09 12:12:09 snail clightd[69263]:   Major opcode of failed request:  73 (X_GetImage)
Oct 09 12:12:09 snail clightd[69263]:   Serial number of failed request:  7
Oct 09 12:12:09 snail clightd[69263]:   Current serial number in output stream:  7
Oct 09 12:12:09 snail clightd[69263]: Registered 'Als' sensor.
Oct 09 12:12:09 snail clightd[69263]: Registered 'Camera' sensor.
Oct 09 12:12:09 snail clightd[69263]: Registered 'Custom' sensor.
Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Creating client 0
Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Adding inotify watch as first client was started.
Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Starting Client 0
Oct 09 12:12:09 snail systemd[1]: clightd.service: Main process exited, code=exited, status=1/FAILURE
Oct 09 12:12:09 snail systemd[1]: clightd.service: Failed with result 'exit-code'.
Oct 09 12:12:09 snail audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Oct 09 12:12:14 snail systemd[1]: clightd.service: Scheduled restart job, restart counter is at 1.
Oct 09 12:12:14 snail audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Oct 09 12:12:14 snail audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Oct 09 12:12:14 snail systemd[1]: Stopped Bus service to manage various screen related properties (gamma, dpms, backlight).
Oct 09 12:12:14 snail systemd[1]: Starting Bus service to manage various screen related properties (gamma, dpms, backlight)...
Oct 09 12:12:14 snail systemd[1]: Started Bus service to manage various screen related properties (gamma, dpms, backlight).
Oct 09 12:12:14 snail audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=clightd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```

FedeDP, 2020-10-11:
> On one laptop that is connected to a TV set I guess dimming is not going to work.

It depends whether your TV supports ddc/ci protocol :) 

 >  Is Wayland a reason?

Wayland is supported!
I've seen your clightd log file; it seems like Clightd is crashing here:
> Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Creating client 0
Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Adding inotify watch as first client was started.
Oct 09 12:12:09 snail clightd[69263]: [default]|IDLE|: Starting Client 0
Oct 09 12:12:09 snail systemd[1]: clightd.service: Main process exited, code=exited, status=1/FAILURE
Oct 09 12:12:09 snail systemd[1]: clightd.service: Failed with result 'exit-code'.

Can you try to build clightd in debug:
```
cmake \
        -G "Unix Makefiles" \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DCMAKE_INSTALL_LIBDIR=lib \
        -DCMAKE_BUILD_TYPE="Debug" \
        -DENABLE_DDC=1 -DENABLE_GAMMA=1 -DENABLE_DPMS=1 -DENABLE_SCREEN=1 \
        ..
make
```
and then manually stop clight, start clightd with gdb, and then finally clight? (on wayland)
Ie: from within clightd build folder:
* `killall clight`
* `sudo gdb ./clightd`
* open new terminal
* `clight`
* check other terminal
Wait for crash, and then run `bt` command in gdb console; finally send me gdb output since crash onwards (ie: crash output + bt output).
Thank you!!

>  Update: After switching to X11 dimming works.

At least we know it is a wayland issue :)

FedeDP, 2020-10-11:
I tested with wayland on kde plasma (my arch installation) and it works fine indeed!

perceival, 2020-10-11:
Probably the issue was caused by autostart of clightd. After issuing _sudo systemctl disable clightd_ command service is invoke by clight starting up and in such a setup there are no more crashes. Thank you!

perceival, 2020-10-12:
Enabling Clightd service is redundant and is causing service to crash. No need to _systemctl enable clighd.service_.

closed 2020-10-12 by perceival