FedeDP/Clight #153: [BUG] Clight doesn't start

coldspark29, 2020-10-15:
**Clight version the issue has been seen with:**
> r715.767e184-1

**Used distribution:**
> Manjaro

**Describe the bug**
``Clightd must be updated. Required version: 4.3.`` is shown although I have used the latest git version. The sensor is basically working when I use clight -w.

FedeDP, 2020-10-15:
Hi! Thanks for opening this bug report! 
Are you running clightd-git too?

coldspark29, 2020-10-15:
Yes, I now went back to the distro’s version. That one works, but I can‘t run the wizard there. With the git version running the wizard works.

FedeDP, 2020-10-17:
Did you restart your system once installed clight-git and clightd-git?

coldspark29, 2020-10-17:
No I obviously didn't because it works now :)

closed 2020-10-17 by coldspark29