FedeDP/Clight #155: [BUG] "Failed to validate location request" when using verbose ouput

coldspark29, 2020-10-16:
**Clight version the issue has been seen with:**
> 4.2

**Used distribution:**
> Manjaro

**Describe the bug**
When I run clight in the terminal I get the folowing messages very frequently. GeoClue2 seems to be installed and working. 
```
53.50 10.00 received from Geoclue2.
Loc distance: 53.50,10.00 -> 53.59,10.05 : 5.98 km.
Failed to validate location request.
```

NICHOLAS85, 2020-10-16:
I ended up seeing the same thing. Here's the issue I opened: https://github.com/FedeDP/Clight/issues/151
The failed to validate location request means the newly received location data isn't far enough away from the last distance Clight used. Once you've traveled enough, Clight recalculates the values used for gamma and sunset/time related stuff.

If you don't travel often, you can set a latitude and longitude manually in your conf to avoid the Geoclue spam in your logs.

coldspark29, 2020-10-17:
I don't want clight to set my gamma anyway. KDE has had a nightshift functionality implemented for some time now and it works great. Clight just gets in the way of this. Yesterday evening at dusk my screen got more blue instead of shifting to red, so I turned it of. I will turn of the location now as well.

closed 2020-10-17 by coldspark29