FedeDP/Clight #156: [BUG] Example for custom modules possible outdated

coldspark29, 2020-10-16:
**Clight version the issue has been seen with:**
> 4.2

**Used distribution:**
> Manjaro

**Describe the bug**
I wanted to build the nightmode module according to the Wiki. Also tried the nightmode.skel, but I get a segmentation fault. For getting the system() function to recognize I needed ``#include <stdlib.h>``` but I also get ``exptected unqualified-id``. Could it be that the PubSub commands have changed or am I missing another include?

[image]

FedeDP, 2020-10-17:
Hi, thanks for opening this bug report!
You are surely right about stdlib.h include, thanks for pointing it out!
But i don't get the other issue; i am able to successfully build the module with no problem.

coldspark29, 2020-10-17:
I tried to test it in the shell and it did not change the theme

coldspark29, 2020-10-17:
Everything works since I have got the git version to work :)

closed 2020-10-17 by coldspark29